espressif/arduino-esp32 #3318: App rollback

jakubtopic, 2019-10-04:
Hello, I use **v1.10.0** as an ESP-IDF component and I'd wish to use the app rollback for ensuring functionality after broken firmware update (`Update.h` is used to perform the update). But after enabling `CONFIG_APP_ROLLBACK_ENABLE` it does not simply work. After firmware update, the following snippet returns `ESP_OTA_IMG_VALID` instead of expected `ESP_OTA_IMG_PENDING_VERIFY`.
```
const esp_partition_t *cur_app = esp_ota_get_running_partition();
esp_ota_img_states_t ota_state;
esp_ota_get_state_partition(cur_app, &ota_state);
```
And after restarting without marking the image as valid, the same (updated) image will be loaded again. Not the original one - there is no rollback.

My partition table looks like this:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1E0000,
app1,     app,  ota_1,   0x1F0000,0x1E0000,
spiffs,   data, spiffs,  0x3D0000,0x30000,
```
If there is any relevant data missing, I'll be happy to add it.

Is there a way how to make it work? Is the problem with the combination of ESP-IDF/Arduino framework?

closed 2019-12-17 by stale[bot]

marcovannoord, 2020-09-16:
So i found your issue because i ran into the same issue.
It took me quite some time to find out that during the [initArduino() ](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L171) function that is called when CONFIG_AUTOSTART_ARDUINO is enabled in menuconfig, will call the (weakly defined) function verifyOta().  
By default, that function will always return true, and before your setup() function is called, it will change from ESP_OTA_IMG_PENDING_VERIFY to ESP_OTA_IMG_VALID.

To fix this:
- Either override the verifyOTA() with your own function that will do some checks, or 
- disable "Autostart Arduino setup and loop on boot" in menuconfig and then start it yourself (for example as in  https://github.com/platformio/platform-espressif32/blob/master/examples/espidf-arduino-blink/src/Blink.cpp )

Adrian-Samoticha, 2025-04-07:
> So i found your issue because i ran into the same issue. It took me quite some time to find out that during the [initArduino() [...]

For anyone stumbling upon this issue in the future, don’t miss the fact that the `verifyOta` and `verifyRollbackLater` functions are defined in a .c file. Therefore, if you wish to disable the automatic verification, do not forget to add `extern "C"` to your function override: 

```cpp
extern "C" bool verifyRollbackLater(){
    return true;
}
```