espressif/arduino-esp32 #3320: v1.0.4: AsyncUDP doesn't properly release resources when receiving packets (fix proposed in description)

ssilverman, 2019-10-04:
### Hardware:
Board: Adafruit ESP32 Feather
Core Installation version: 1.0.4
IDE name: Arduino 1.8.10
Flash Frequency: 80MHz
PSRAM enabled: don't know
Upload Speed: 921600
Computer OS: OSX 10.14.6

### Description:
This is the same problem as for v1.0.3 detailed here: #3287
A proposed fix is here: #3290
The reasoning behind the fix is here: #3288

In summary, once all the UDP buffers are used up, the program stops being able to accept UDP packets.

Basically, per the API, `AsyncUDP`'s callback should take a reference, but that doesn't work. Instead, the callback taking a copy (and thus calling the copy constructor) does work. The references to `pbuf` aren't tracked correctly and the current logic for handling both the pass-as-copy and pass-as-reference doesn't work for both cases. The fix in #3290 does work for both cases.

The proposed fix may not be the best way to write this, and I'm sure the design will be improved, but the fix works with the current code by re-balancing the copy constructor to call `pbuf_ref`.

A test sketch is below and the bash script for repeatedly sending 512-byte packets is here (change the IP address and packet size as you wish):
```bash
while true; do echo -n $(printf '.%.0s' {1..512}) > /dev/udp/192.168.1.9/8000; sleep 0.05; done
```

### Sketch:
```cpp
#include <AsyncUDP.h>
#include <Esp.h>
#include <WiFi.h>

constexpr char kAPName[] = "ChangeMe";
constexpr char kAPPassword[] = "ChangeMe";
constexpr bool isSoftAP = false;  // Change to true for SoftAP mode

AsyncUDP udp;

void setup() {
  Serial.begin(115200);
  while (!Serial && millis() < 4000) {
    // Wait for Serial
  }
  Serial.println("Starting.");

  if (isSoftAP) {
    Serial.println("Starting SoftAP...");
    if (WiFi.softAP(kAPName, kAPPassword)) {
      Serial.print("    IP: ");
      Serial.println(WiFi.softAPIP());
    } else {
      Serial.println("ERROR: Starting SoftAP!");
    }
  } else {
    if (WiFi.begin(kAPName, kAPPassword)) {
      while (!WiFi.isConnected()) {
        delay(500);
      }
      Serial.print("    IP: ");
      Serial.println(WiFi.localIP());
      Serial.print("    Subnet: ");
      Serial.println(WiFi.subnetMask());
      Serial.print("    Gateway: ");
      Serial.println(WiFi.gatewayIP());
    } else {
      Serial.println("    ERROR: Connecting to AP!");
    }
  }

  if (!udp.listen(8000)) {
    Serial.println("ERROR: Starting UDP server!");
  }
  udp.onPacket(onPacket);
}

int counter = 0;

// Using a copy works, but using a reference, as written, does not.
void onPacket(AsyncUDPPacket &packet) {
  Serial.printf("%d: %d\n", ++counter, packet.length());
}

void loop() {
  // Print some status every 5s
  Serial.printf("Free heap: %d\n", ESP.getFreeHeap());
  delay(5000);
}
```

ssilverman, 2019-10-05:
I posted this issue because I wasn’t sure if v1.0.4 is being tracked separately. Apologies if this is considered double posting. Please delete if it’s inappropriate.

atanisoft, 2019-10-05:
@ssilverman there is only one tracker and it is cumulative in nature. I'd suggest hop on gitter.im and discuss the issue with @me-no-dev and others there.

ssilverman, 2019-12-13:
There's still no explanation about why the proposed fix in #3290 doesn't work.

closed 2020-02-26 by stale[bot]