espressif/arduino-esp32 #3323: SPIFFS file.read fills buffer with errors

lagunax, 2019-10-05:
working with upng library (adaptation for esp32)
latest esp-idf+arduino-esp32 installed.
trying to use SPIFFS file.open -> file buffer have wrong content (checked on header only)
simptom:
git clone https://github.com/elanthis/upng
modifying part file loading into next code:
```
...in upng.h
	#include <Arduino.h>
	#include <FS.h>
	#include "SPIFFS.h"
... in upng.cpp
upng_t* upng_new_from_file(const char *filename)
{
	upng_t* upng;
	unsigned char *buffer;
	//FILE *file;
	File file;
	long size;

	upng = upng_new();
	if (upng == NULL) {
		return NULL;
	}

	//file = fopen(filename, "rb");
	file = SPIFFS.open(filename, "rb");
	if (file == NULL) {
		SET_ERROR(upng, UPNG_ENOTFOUND);
		return upng;
	}

	/* get filesize */
	//fseek(file, 0, SEEK_END);
	file.seek(0,SeekEnd);
	//size = ftell(file);
	size = file.position();
	//rewind(file);
	file.seek(0,SeekSet);

	/* read contents of the file into the vector */
	buffer = (unsigned char *)malloc((unsigned long)size);
	if (buffer == NULL) {
		//fclose(file);
		file.close();
		SET_ERROR(upng, UPNG_ENOMEM);
		return upng;
	}
	//fread(buffer, 1, (unsigned long)size, file);
	file.read(buffer,(size_t)size);
	//fclose(file);
	file.close();
...
```
(remarked are original code, after remarked - new added lines for esp32)
uploaded png file onto esp32 with header:
```
(decimal)                 137  80  78  71  13  10  26  10
(hexadecimal)           89  50  4e  47  0d  0a  1a  0a
```
checking buffer after "file.close();" 
```
	Serial.printf("\nfile: %i\nfile: %i\nfile: %i\nfile: %i\nfile: %i\nfile: %i\nfile: %i\nfile: %i", int((char*)buffer[0]), int((char*)buffer[1]), int((char*)buffer[2]), int((char*)buffer[3]), int((char*)buffer[4]), int((char*)buffer[5]), int((char*)buffer[6]), int((char*)buffer[7]));
```
and got this in terminal (make monitor):
```
file: 137
file: 80
file: 78
file: 71
file: 13
file: 13
file: 10
file: 26
```
also decoding of png do not works, becous header do not eq to standart.
What goin on?

lagunax, 2019-10-06:
as i see in buffer: byte 0x0d added before every 0x0A byte. hot to prevent this? ("carriage return before new line" - rule)

lbernstone, 2019-10-06:
CRLF issues immediately point at Windows.  How are you getting the files onto the device?

lagunax, 2019-10-06:
> CRLF issues immediately point at Windows. How are you getting the files onto the device?

im working on linux (debian stretch with rt-kernel (working with sound on same pc)) and get this issue this way:
1) inserted into project support for FtpServer (https://github.com/nailbuster/esp8266FTPServer)
2) uploaded file (put)
3) downloaded file and checked if it is same as original (geet and checked - all bytes are same)
4) git clone https://github.com/elanthis/upng
5) changed some file strings with written above
6) looking at console and found that upng do not found right header
7) after line "file.close()" inserted code that prints to serial all bytes from buffer and founds that somehow buffer have '0d' byte added before every '0a' byte, size of buffer larger then original file for count of those added bytes (if original file size is 266 bytes, file contains 3 bytes with '0a'-value, then buffer have 269 bytes size)

hardware is TTGO-T-Display esp32 board with screen module.

lagunax, 2019-10-06:
> CRLF issues immediately point at Windows. How are you getting the files onto the device?

full project here
https://github.com/lagunax/ESP32-AutoHome

lagunax, 2019-10-06:
now making temporary workaround for this issue with next code, but thinking that this is spiffs issue.
```
unsigned long Remove0Dbefor0A(void *buffer, unsigned long size){
	/*
	 * Function search for 0D 0A combination and removes 0D bytes in files where this
	 * combination will found.
	 *
	 * buffer - pointer to the buffer
	 * size - size of buffer
	 *
	 * Returns size of modified buffer, with this size you can realloc buffer memory
	 * with less size then it was before calling function.
	 *
	 */

	char *left=(char *)malloc(1);
	char *cur= (char *)malloc(1);

	for (unsigned int iByte=(size-1); iByte>1; iByte--){
		memcpy(cur,buffer+iByte,1);
		if (*cur==0x0A){
			memcpy(left,buffer+iByte-1,1);
			if (*left==0x0D){
				for (unsigned int pos=iByte-1; pos<(size-2); pos++) memcpy(buffer+pos,buffer+pos+1,1);
				iByte--;
				size--;
			}
		}
	}

	free(cur);
	free(left);
	return size;
}
```

atanisoft, 2019-10-06:
When you ftp put, are you in bin mode?

lbernstone, 2019-10-06:
Pretty easy to verify that it is not SPIFFS.read with [esp32-fs-plugin](https://github.com/me-no-dev/arduino-esp32fs-plugin).

lagunax, 2019-10-06:
my fault. ftp-client worked in text mode. week for searching this fault XD
issue is closed, sorry

closed 2019-10-06 by lagunax