espressif/arduino-esp32 #3324: ESP32-Solo-1 with Arduino IDE

bardiabarai, 2019-10-05:
Hello, Is there any way to use Arduino IDE for ESP32-SOLO-1 (the single core version of ESP32)?
By the way I checked https://github.com/lbernstone/arduino-esp32-solo, This doesn't work and it has some errors after compiling.

lbernstone, 2019-10-05:
Weird, there are no issues in that repository.  v1.0.4 uploaded today.

bardiabarai, 2019-10-06:
My Error Massage:

Arduino: 1.8.9 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, 80MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\bardi\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libesp32.a(cpu_start.o):(.iram1.literal+0x84): undefined reference to `esp_coex_adapter_register'

C:\Users\bardi\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libesp32.a(cpu_start.o):(.iram1.literal+0xac): undefined reference to `bootloader_common_set_flash_cs_timing'

C:\Users\bardi\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libesp32.a(cpu_start.o): In function `start_cpu0_default':

/home/larryb/esp32/esp-idf/components/esp32/cpu_start.c:380: undefined reference to `esp_coex_adapter_register'

C:\Users\bardi\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libesp32.a(cpu_start.o): In function `call_start_cpu0':

/home/larryb/esp32/esp-idf/components/esp32/cpu_start.c:158: undefined reference to `bootloader_common_set_flash_cs_timing'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\bardi\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

bfaliszek, 2019-11-27:
@lbernstone How to install this library?

nenaddrop, 2020-01-22:
@bfaliszek navigate to the board sdk directory (/Users/<username>/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/sdk). In this directory you will find include/config and lib subdirectory. Use the files provided by the https://github.com/lbernstone/arduino-esp32-solo (but check out the correct version from commits https://github.com/lbernstone/arduino-esp32-solo/commits/master). Replace the corresponding files in those subdirectories (but maybe keep the old ones by renaming them). Hope this helps.

closed 2020-04-05 by stale[bot]

lamloei, 2020-04-10:
copy and replace
C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools

andrew9404-bq, 2020-05-09:
Hi, I changed the files of https://github.com/lbernstone/arduino-esp32-solo in the direction that you said, but now it shows me this error. I don´t know what to do.

Arduino: 1.8.12 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, 80MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

In file included from C:\Users\andre\OneDrive\Documentos\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-bt.c:22:0:

C:\Users\andre\OneDrive\Documentos\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-bt.c: In function 'btStart':

C:\Users\andre\OneDrive\Documentos\Arduino\hardware\espressif\esp32/tools/sdk/include/bt/esp_bt.h:120:24: error: 'CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF' undeclared (first use in this function)

     .bt_sco_datapath = CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF,          \

                        ^

C:\Users\andre\OneDrive\Documentos\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-bt.c:35:38: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'

     esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();

                                      ^

C:\Users\andre\OneDrive\Documentos\Arduino\hardware\espressif\esp32/tools/sdk/include/bt/esp_bt.h:120:24: note: each undeclared identifier is reported only once for each function it appears in

     .bt_sco_datapath = CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF,          \

                        ^

C:\Users\andre\OneDrive\Documentos\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-bt.c:35:38: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'

     esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();

                                      ^

exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

lamloei, 2020-05-09:
https://medium.com/@lamloei.com/%E0%B8%A7%E0%B8%B4%E0%B8%98%E0%B8%B5%E0%B9%80%E0%B8%9E%E0%B8%B4%E0%B9%88%E0%B8%A1%E0%B8%9A%E0%B8%AD%E0%B8%A3%E0%B9%8C%E0%B8%94-node32solo-8324cb5e8f1d

lamloei, 2020-05-09:
..\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools

andrew9404-bq, 2020-05-17:
Thank you, it works!, The main problem was that I installed the ESP32 library whit the steps shown in https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md and it didn´t creat these folders AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools. To solve the problem I install the ESP32 library direct from the Arduino IDE and replace the files that you said.

sixtyfive, 2021-11-15:
Referencing #5183.

netll, 2025-04-27:
> Weird, there are no issues in that repository. v1.0.4 uploaded today.奇怪的是，该存储库中没有问题。v1.0.4 今天上传了。

It is now 2025, and the Arduino IDE is already version 2.3.5. There is no SDK folder in the ESP32 directory, and there is no sdkconfig.h file. What should I do now to use it?