espressif/arduino-esp32 #3325: WiFiMulti not working with hidden SSID

trlafleur, 2019-10-06:
In a quick test with the provided example for WiFiMulti, I was unable to connect to a hidden SSID...

If I use the standard          WiFi.begin(nSSID, nPassword);   version, all work fine and it can find the hidden SSID.

~~~
/*
 *  This sketch tries to Connect to the best AP based on a given list
 *
 */

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

void setup()
{
    Serial.begin(115200);
    delay(100);
    
    WiFi.disconnect(true);
    WiFi.mode(WIFI_STA);
    
    wifiMulti.addAP("hiddenssid", "12345");          // Hidden SSID, will not work
    wifiMulti.addAP("openssid", "12345");             // open SSID, work fine...

    Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.print("IP address: ");
        Serial.println(WiFi.localIP());
        Serial.print("SSID: ");
        Serial.println(WiFi.SSID());
        Serial.print("Channel: ");
        Serial.println(WiFi.channel());
    }
}

void loop()
{
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.print("WiFi not connected!  ");
        Serial.println(WiFi.status());
        delay(1000);
    }
}
~~~

~~~
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: lafleur
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 1 networks found
[D][WiFiMulti.cpp:151] run():        0: [1][82:2A:A8:D7:F3:34] RSF-Guest (-57) *
[E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 1 networks found
[D][WiFiMulti.cpp:151] run():        0: [1][82:2A:A8:D7:F3:34] RSF-Guest (-58) *
~~~

lbernstone, 2019-10-06:
Does it work if the hidden ssid is the only one in the list?

trlafleur, 2019-10-06:
No...

lbernstone, 2019-10-06:
What do you expect the behavior to be?  WiFiMulti picks the entry on the list with the highest rssi, which can only be determined from a scan.  It has no idea of "hidden", just that the signal strength of an AP is below reception.

atanisoft, 2019-10-06:
WiFi.scanNetworks() will not show hidden SSIDs by default, WiFiMulti doesn't pass the flag to enable hidden networks to be included. You can try changing [this line](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiMulti.cpp#L103) to:
```
scanResult = WiFi.scanNetworks(false, true);
```
And see if it works.

trlafleur, 2019-10-06:
If the hidden flag is set to true, Scan will return an entry in its scan
table for all AP it sees with RSSI, channel and other information...
what needs to happen in the WiFiMulti code is if scan returns an SSID of
empty, ie: its hidden, it should try a connection with all the "requested"
SSID to see if it can connect...

As a minimum, a note needs to be added that code will NOT work with hidden
SSID.. but again, more and more we are seeing hidden SSID everywhere---> so
this code should be fixed to support hidden SSID.

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

BararDiHoumes, 2021-11-14:
I were expecting fix for this as well, and after all this time none work on it ? Really ?

lucasssvaz, 2024-02-01:
Fixed by #9202

closed 2024-02-05 by me-no-dev