espressif/arduino-esp32 #3331: SDWebServer text editor folder tree not loading

bpor, 2019-10-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
I am trying to use the SDwebserver with the text editor. I only updated the ssid and password in the SDWebServer.ino. I loaded the "SdRoot" folder to a 4GB sd card. index and other site are loading ok but the text editor is not loading the folder tree and the file upload is not working. 

[image]

On the page console, I am getting net::ERR_CONNECTION_RESET at the GET instruction. If I just run the GET /list?dir=/ from console I am getting a response for the esp32.

[image]

Is anyone having the same issue. Tried it with various ESP32 modules and also used old versions of SDWebserver and I am getting the same problem

atanisoft, 2019-10-07:
Check the serial console output from the esp32, it likely is failing to mount the SD card.

bpor, 2019-10-07:
The SD card is mounting and the web pages are loading. 
The printDirectory function is never called. other web pages are loading. the only sdcardnotfound that i am getting is for the favicon.ico but I know that that is not on the sd card.

[image]

closed 2019-12-20 by stale[bot]