espressif/arduino-esp32 #3332: Strange bluetooth behavior with freertos queue

lagunax, 2019-10-07:
Does anybody know why this happens when i try to connect to ESP32 within bluetooth?
```
make monitor
Toolchain path: /usr/media/Programming/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /usr/media/Programming/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (11) boot: ESP-IDF v3.2.3-dirty 2nd stage bootloader
I (11) boot: compile time 20:45:05
I (20) boot: Enabling RNG early entropy source...
I (20) boot: SPI Speed      : 40MHz
I (20) boot: SPI Mode       : DIO
I (23) boot: SPI Flash Size : 4MB
I (27) boot: Partition Table:
I (31) boot: ## Label            Usage          Type ST Offset   Length
I (38) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (45) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (53) boot:  2 factory          factory app      00 00 00010000 00177000
I (60) boot:  3 storage          Unknown data     01 82 00187000 00100000
I (68) boot: End of partition table
I (72) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1a1ac (106924) map
I (117) esp_image: segment 1: paddr=0x0002a1d4 vaddr=0x3ffbdb60 size=0x027c0 ( 10176) load
I (121) esp_image: segment 2: paddr=0x0002c99c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /usr/media/Programming/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (124) esp_image: segment 3: paddr=0x0002cda4 vaddr=0x40080400 size=0x0326c ( 12908) load
I (138) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0xa544c (676940) map
0x400d0018: _flash_cache_start at ??:?

I (371) esp_image: segment 5: paddr=0x000d546c vaddr=0x4008366c size=0x0e08c ( 57484) load
0x4008366c: btpwr_pll_track at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:755

I (403) boot: Loaded app from partition at offset 0x10000
I (403) boot: Disabling RNG early entropy source...
I (403) cpu_start: Pro cpu up.
I (407) cpu_start: Starting app cpu, entry point is 0x40080f00
0x40080f00: call_start_cpu1 at /usr/media/Programming/esp/esp-idf/components/esp32/cpu_start.c:250

I (0) cpu_start: App cpu up.
I (417) heap_init: Initializing. RAM available for dynamic allocation:
I (424) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (430) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (436) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (442) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (449) heap_init: At 3FFC5DC0 len 0001A240 (104 KiB): DRAM
I (455) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (461) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (468) heap_init: At 400916F8 len 0000E908 (58 KiB): IRAM
I (474) cpu_start: Pro cpu start user code
I (45) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (630) BTDM_INIT: BT controller compile version [acab313]
I (1325) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1395) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0
[Bluetooth] started!
E (6775) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6775) task_wdt:  - IDLE1 (CPU 1)
E (6775) task_wdt: Tasks currently running:
E (6775) task_wdt: CPU 0: IDLE0
E (6775) task_wdt: CPU 1: loopTask
E (11775) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11775) task_wdt:  - IDLE1 (CPU 1)
E (11775) task_wdt: Tasks currently running:
E (11775) task_wdt: CPU 0: IDLE0
E (11775) task_wdt: CPU 1: loopTask
/usr/media/Programming/esp/esp-idf/components/freertos/queue.c:706 (xQueueCreateCountingSemaphore)- assert failed!
abort() was called at PC 0x4008c427 on core 0
0x4008c427: xQueueCreateCountingSemaphore at /usr/media/Programming/esp/esp-idf/components/freertos/queue.c:1996

Backtrace: 0x4008f043:0x3ffd3770 0x4008f1a9:0x3ffd3790 0x4008c427:0x3ffd37b0 0x400f4dc6:0x3ffd37d0 0x400f5094:0x3ffd37f0 0x40109dcf:0x3ffd3810 0x4010f55e:0x3ffd3830 0x40121da9:0x3ffd3850 0x40101f25:0x3ffd3870 0x401025fe:0x3ffd38b0
0x4008f043: invoke_abort at /usr/media/Programming/esp/esp-idf/components/esp32/panic.c:135

0x4008f1a9: abort at /usr/media/Programming/esp/esp-idf/components/esp32/panic.c:170

0x4008c427: xQueueCreateCountingSemaphore at /usr/media/Programming/esp/esp-idf/components/freertos/queue.c:1996

0x400f4dc6: osi_sem_new at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/osi/semaphore.c:31

0x400f5094: fixed_queue_new at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/osi/fixed_queue.c:56

0x40109dcf: l2cu_allocate_lcb at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_utils.c:1798

0x4010f55e: l2c_link_hci_conn_req at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_link.c:862

0x40121da9: btm_sec_conn_req at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_sec.c:5829

0x40101f25: btu_hcif_connection_request_evt at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_hcif.c:582
 (inlined by) btu_hcif_process_event at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_hcif.c:180

0x401025fe: btu_hci_msg_process at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:152
 (inlined by) btu_task_thread_handler at /usr/media/Programming/esp/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:229

Rebooting...
```

ESP32 (TTGO-T-Display)

lbernstone, 2019-10-07:
Not arduino.
https://github.com/espressif/esp-idf/issues

atanisoft, 2019-10-07:
No code to advise on, but it looks like your loopTask is monopolizing core 1 and preventing the idle task from executing which is triggering the watchdog timer.

lagunax, 2019-10-08:
> No code to advise on, but it looks like your loopTask is monopolizing core 1 and preventing the idle task from executing which is triggering the watchdog timer.

main.cpp
```
#include "System.h"

TaskHandle_t loopTaskHandle = NULL;
System *sys = new System();

#if CONFIG_AUTOSTART_ARDUINO

bool loopTaskWDTEnabled;

void setup() {
	  sys->tft->init();
	  sys->tft->setRotation(2);
	  sys->tft->fillScreen(TFT_BLACK);
	  sys->tft->setCursor(0, 0, 2);
	  sys->tft->setTextColor(TFT_WHITE,TFT_BLACK);
	  sys->tft->setTextSize(1);

	  sys->tft->println("starting...");
  Serial.begin(115200);
  sys->tft->println("serial");

  SPIFFS.begin(true);
  sys->tft->println("SPIFFS");

  sys->BTInit();
  sys->tft->println("BT");
  sys->tft->fillScreen(TFT_BLACK);

  sys->DrawPng();

}

void loop() {

	#ifdef _BLUETOOTH_SERIAL_H_
byte *buff=(byte *) malloc(32);
unsigned int color=0;
char pos=0;
int input;
	//Bluetooth module
		if (sys->BTavailable()) {
			input=sys->BTread();
			if ((input>=48)&&(input<=49)){
				*(buff+pos)=(char)input;
				pos++;
			}
			if ((pos>=32)|(input=='\n')){
				pos=0;
				sys->BTwrite((const unsigned char *)"\n",1);
				for(byte ipos=0;ipos<31;ipos++){
					color+=*(buff+pos)^pos;
				}
				Serial.printf("\ncolor=%i\n",color);
				for (int xx=70;xx<100;xx++){
					for (int yy=70;yy<100;yy++){
						sys->tft->drawPixel(xx,yy,color);
					}
				}

			}

			Serial.write(input);
					}
	#endif
}

void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
        if(loopTaskWDTEnabled){
            esp_task_wdt_reset();
        }
        loop();
    }
}

extern "C" void app_main()
{
    loopTaskWDTEnabled = false;
    initArduino();
    xTaskCreateUniversal(loopTask, "loopTask", 8192, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
}

#endif
```

System.h
```

#define SERVICE_FTP
#define SERVICE_WEB

#ifndef MAIN_SYSTEM_H_
	#define MAIN_SYSTEM_H_

	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "esp_task_wdt.h"
	#include <Arduino.h>
	#include <FS.h>
	#include <BluetoothSerial.h>
	#include <WiFiClientSecure.h>
	#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip

	#ifdef SERVICE_FTP
		#include "FtpServer.h"
	#endif

	#ifdef SERVICE_WEB
		#include "WebService.h"
	#endif

	#define TFT_GREY 0x5AEB // New colour

	class System {

		private:
			#ifdef _BLUETOOTH_SERIAL_H_
				#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
					#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
				#endif
				BluetoothSerial SerialBT;
			#endif

			#ifdef WiFiClientSecure_h
				WiFiClientSecure WiFi_client;
			#endif
		public:
			#ifdef _BLUETOOTH_SERIAL_H_
				#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
					#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
				#endif

				void BTInit();

				TFT_eSPI *tft;  // Invoke library

				bool BTbegin(String);
				int BTavailable(void);
				int BTpeek(void);
				bool BThasClient(void);
				int BTread(void);
				size_t BTwrite(uint8_t c);
				size_t BTwrite(const uint8_t *buffer, size_t size);
				void BTflush();
				void BTend(void);
				esp_err_t BTregister_callback(esp_spp_cb_t * callback);
			#endif

			#ifdef WiFiClientSecure_h
				void WFAPInit();
				IPAddress WFlocalIP();
			#endif
			#ifdef WEBSERVER_H
			#endif
			#ifdef FTP_SERVERESP_H
				Service::FTP::FtpServer *FTP_Server;
			#endif
				void DrawPng();

		System();
		virtual ~System();
	};

#endif /* MAIN_SYSTEM_H_ */
```

System.cpp
```
long file here is a part of this file about bluetooth

#ifdef _BLUETOOTH_SERIAL_H_
	void System::BTInit(){
		this->BTbegin("ESP32-Autohome"); //Bluetooth device name
		Serial.println("[Bluetooth] started!");
	}

	bool System::BTbegin(String localName=String()){//String localName){//
		return SerialBT.begin(localName);
	}

	int System::BTavailable(void){
		return SerialBT.available();
	}

	int System::BTpeek(void){
		return SerialBT.peek();
	}

	bool System::BThasClient(void){
		return SerialBT.hasClient();
	}

	int System::BTread(void){
		return SerialBT.read();
	}

	size_t System::BTwrite(uint8_t c){
		return SerialBT.write(c);
	}

	size_t System::BTwrite(const uint8_t *buffer, size_t size){
		return SerialBT.write(buffer, size);
	}

	void System::BTflush(){
		SerialBT.flush();
	}

	void System::BTend(void){
		SerialBT.end();
	}

	esp_err_t System::BTregister_callback(esp_spp_cb_t * callback){
		return SerialBT.register_callback(callback);
	}

#endif
```

now searching where it starts to be (looking git commits)

lagunax, 2019-10-08:
something wrong was with loop(), moved out some vars to globalspace and all gone

closed 2019-10-08 by lagunax