espressif/arduino-esp32 #3333: ESP32 TimerAlarmEnable() in ISR interrupt panic'ed error

yogeshwaran774, 2019-10-08:
In setup i am configuring the timer and whenever GPIO interrupt occurs it goes into ISR and enable the timer.

while enabling the timer inside the ISR eps32 interrupt panic'ed error is coming but if i do it outside the ISR then it is working fine can anyone help me to understand what is happening inside ISR? Below i attached small part of code
------code_start-----
hw_timer_t *uploadtimer = NULL;
portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
void setup()
{
pinmode(26,INPUT_PULLUP);
attachInterrupt(26,ISR,FALLING)
 upload_timer = timerBegin(0, 80, true);                  //timer 0, div 80
  timerAttachInterrupt(upload_timer, &check_upload, true);  //attach callback
  timerAlarmWrite(upload_timer, 10000, false); //set time in us
}

void IRAM_ATTR ISR()
{
  portENTER_CRITICAL_ISR(&mux);
    timerAlarmEnable(upload_timer);
  portEXIT_CRITICAL_ISR(&mux);
}

void check_upload()
{
//wifi connect functions
}

-------code_end--------

**ESP32  ERROR**
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4014f4d1  PS      : 0x00060034  A0      : 0x80081402  A1      : 0x3ffbe740  
A2      : 0x0032da6c  A3      : 0x00000000  A4      : 0x000003e8  A5      : 0x00000000  
A6      : 0x00002100  A7      : 0x3ffc1c6c  A8      : 0x800835e6  A9      : 0x00000000  
A10     : 0x0032da6c  A11     : 0x00000000  A12     : 0xb2c00000  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x3ffb8058  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4014f8f6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4014f4d1

Backtrace: 0x4014f4d1:0x3ffbe740 0x400813ff:0x3ffbe760 0x400d3269:0x3ffbe780 0x400814d9:0x3ffbe7a0 0x40081a01:0x3ffbe7c0 0x4014f8f3:0x3ffbc650 0x400e927b:0x3ffbc670 0x4008ace9:0x3ffbc690 0x40088de9:0x3ffbc6b0

Core 0 register dump:
PC      : 0x4014f8f6  PS      : 0x00060834  A0      : 0x800e927e  A1      : 0x3ffbc0d0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800eb4de  A9      : 0x3ffbc0a0  
A10     : 0x00000000  A11     : 0x400868b4  A12     : 0x00060120  A13     : 0x3ffbb650  
A14     : 0x00000000  A15     : 0x3ffbbdc0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4014f8f6:0x3ffbc0d0 0x400e927b:0x3ffbc0f0 0x4008ace9:0x3ffbc110 0x40088de9:0x3ffbc130

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

me-no-dev, 2019-10-08:
what's the critical section for? Also timerAlarmEnable is not marked as IRAM, so that is not good.

closed 2019-10-08 by me-no-dev

yogeshwaran774, 2019-10-16:
Sorry i did not understand what you are asking and the timerAlarmEnable is kept in IRAM function only right?