espressif/arduino-esp32 #3335: ESP32 BLEDevice init-deinit  cycle occur memory leak

Taeyang-Choi, 2019-10-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      NO
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
Hi
I'm working on a program that turns the wifi-bluetooth ble communication on and off with the esp32 module.

Then I checked a memory leak while iterating through BLEDevice :: init() deinit().

If you repeat the init()-deinit() operation about 45 times, you will reboot because of insufficient memory.

Below is the test arduino code, serial monitor window, and backtrace data when rebooted.

Please help me to solve this problem
Thank you.

### Sketch:  
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;
BLEService *pService = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);
  bleInit();
}

void bleInit() {
  Serial.println("init start");
  // Create the BLE Device
  BLEDevice::init("ESP32");
  
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  pService = pServer->createService(SERVICE_UUID);
  Serial.println("create Service");

  // Create a BLE Characteristic
[... 376 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c2e4  PS      : 0x00060330  A0      : 0x8010227d  A1      : 0x3ffc8560  
A2      : 0x00000000  A3      : 0x3ffffb58  A4      : 0x00000300  A5      : 0x00000000  
A6      : 0xc331914b  A7      : 0x8fccc5c9  A8      : 0x00000000  A9      : 0x3ffc8500  
A10     : 0x00000000  A11     : 0x00001800  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x00060f20  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x0000002f  

Backtrace: 0x4000c2e4:0x3ffc8560 0x4010227a:0x3ffc8570 0x400fd376:0x3ffc8590 0x400fa861:0x3ffc85c0 0x400d239e:0x3ffc8620 0x400d3a66:0x3ffc8660 0x400d1963:0x3ffc8680 0x400d1a20:0x3ffc86c0 0x400d6fd1:0x3ffc86e0 0x4008e089:0x3ffc8700

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

lbernstone, 2019-10-08:
How about with `BLEDevice.deinit(true)` ?
The definition of the method is `static void deinit(bool release_memory = false);`

Taeyang-Choi, 2019-10-08:
> How about with `BLEDevice.deinit(true)` ?
> The definition of the method is `static void deinit(bool release_memory = false);`

If you run BLEDevice.deinit (true)
The second calling init () function stops running.

chegewara, 2019-10-10:
Here is the problem, not in library (i had the same issue):
```
void bleInit() {
  Serial.println("init start");
  // Create the BLE Device
  BLEDevice::init("ESP32");
  
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  pService = pServer->createService(SERVICE_UUID);
  Serial.println("create Service");

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
                    );

  // Create a BLE Descriptor
  Serial.println("create Characteristic");
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();
  Serial.println("pService start");
  // Start advertising
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(false);
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter
  BLEDevice::startAdvertising();
  //Serial.println("Waiting a client connection to notify...");
  Serial.println("init complete");
}
```
Every time you call this function you create new services, characteristics, descriptors and MyServerCallbacks, but you never delete it.

Taeyang-Choi, 2019-10-15:
> Here is the problem, not in library (i had the same issue):
>
> ```
> void bleInit() {
>   Serial.println("init start");
>   // Create the BLE Device
>   BLEDevice::init("ESP32");
>
>   // Create the BLE [...]

Thank you for your answer.

However, I couldn't find a library that deletes the server, characteristics, descriptors and MyServerCallbacks.

I could only find the BLEServer.removeService (BLEService * service) function that deletes services.

Are there any other libraries that delete the server, characteristics, descriptors and MyServerCallbacks?

chegewara, 2019-10-15:
Its something you have to do in your code.

closed 2019-12-28 by stale[bot]

PyTechMike, 2020-06-22:
Hi! Have you solved this issue? I am stuck on the step of implementing something else to `loop` function

peter1a, 2020-10-01:
Any solution with that?

diegofreitas, 2021-01-22:
A new instance is created here

`pServer = BLEDevice::createServer();`

you need to delete the object from memory somewhere else like this;

`delete pServer;`

 you dont need a lib to do that