espressif/arduino-esp32 #3337: WiFi does not connect

fedyfausto, 2019-10-08:
Hello! i'm using the ai-thinker esp2-cam and i'm trying to connect to my 2.4 wifi with a simple ssid and password (wpa and wpa2 activated) but actually the WiFi.status return zero and it does not connect to the router :<
cpp```
void setup() {
  // Wi-Fi connection
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  Serial.print("Camera Stream Ready! Go to: http://");
  Serial.print(WiFi.localIP());

}
```
How can i fix?

ddwood, 2019-10-08:
I had a similar issue where the status would progress 0 (IDLE).. to some "fatal" status such as 6 /WL_DISCONNECTED. I also had some "Brownout" detection's.
Once you hit a fatal error, then you are stuck in the loop forever. I added a test for fatal status errors and if I got one I would call ESP.restart() (Defined in Esp.h)
I am guessing the main reason for my failed connections was due to the insufficient power supply on my PC's USB port. The restart call does not seem to be called after I moved to a better power source.
The while loop above makes the assumption that the connection will eventually occur, however there are reasons that this could fail. 

I am not sure about this, but I suspect that the main loop method should be checking the status from time to time as well.

fedyfausto, 2019-10-10:
i tried to move all in the main loop but it does not work :( i'm using the last version i think so do you suggest to try an older version?

ddwood, 2019-10-11:
I changed the while loop like this:
  // Start the WIFI radio and join the specified LAN
  wl_status_t wifiStatus = WiFi.begin(ssid, password);
  wl_status_t lastWifiStatus = wifiStatus;
  while (wifiStatus  != WL_CONNECTED) {
     if(wifiStatus != lastWifiStatus) {
       Serial.print("WiFi status is ");
       Serial.println(WlStatusAsString(wifiStatus));
       lastWifiStatus = wifiStatus;
     }
     if(WifiStatusIsFatal(wifiStatus)) 
     {
       delay(2000);
       Serial.println("..");
       ESP.restart();
     } else {
       delay(500);
       Serial.print("..");
     }
     wifiStatus = WiFi.status();
  }

The following helper functions were used in the while loop - put these above your setup method.

const char *WlStatusAsString(wl_status_t status)
{
   static const char * WiFiResultCodeStrings[] =
     { "WL_IDLE_STATUS",
       "WL_NO_SSID_AVAIL",
       "WL_SCAN_COMPLETED",
       "WL_CONNECTED",
       "WL_CONNECT_FAILED",
       "WL_CONNECTION_LOST",
       "WL_DISCONNECTED",
       "WL_NO_SHIELD255"
    };

  if(WL_NO_SHIELD == status)
  {
    return "WL_NO_SHIELD";
  }
  if(status < sizeof(WiFiResultCodeStrings)/sizeof(const char *))
  {
    return WiFiResultCodeStrings[(int)status];
  }
  return "WL_UNKNOWN_STATUS";
}

bool WifiStatusIsFatal(wl_status_t status)
{
   switch(status) {
     case WL_IDLE_STATUS:
     case WL_SCAN_COMPLETED:
     case WL_CONNECTED:
     case WL_DISCONNECTED:
       return false;
     default:
        return true;
   }
}

Don't forget to add this at the top of your file:

#include "Esp.h"

ddwood, 2019-10-15:
Simpler option is to put a limit on the loop count while waiting for the status to be WL_CONNECTED.
If it is not connected after so many loops the likelihood is that it will not be connecting.

So why reset the board? This should not be necessary, but like I said earlier, if my power supply / USB is not enough, then I was getting issues starting the WiFi component.

closed 2019-12-28 by stale[bot]