espressif/arduino-esp32 #3338: Crash in IRAM_ATTR task

JPMJPM, 2019-10-09:
Hi,

sendudp task send two or three UDP packets and then crash.

Any clue ?. Thanks.

Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

`uint8_t *bufferudp = (uint8_t*) calloc(sizeof(dma_buff[0]) / 2, sizeof(uint8_t));`
`static uint8_t numbuf = 0;`

`static void IRAM_ATTR i2s_isr_dmarb(void* arg) {`
`  I2S0.int_clr.val = I2S0.int_raw.val;`
`  BaseType_t xHigherPriorityTaskWoken = pdFALSE;`
`  xSemaphoreGiveFromISR(xSemaphore, &xHigherPriorityTaskWoken);`
 ` if( xHigherPriorityTaskWoken == pdTRUE) {`
`    portYIELD_FROM_ISR();`
`  }`
`}`

`static void IRAM_ATTR sendudp(void *pvParameter) {`
`  while (1) {`
`    if (xSemaphoreTake(xSemaphore, portMAX_DELAY) == pdTRUE) {`
`      for (uint16_t y1 = 0; y1 < sizeof(dma_buff[numbuf]) / 4; y1++) {`
`        bufferudp[y1 << 1] = (dma_buff[numbuf][y1] >> (2 << 3)) & 0xff;`
`        bufferudp[(y1 << 1) + 1] = (dma_buff[numbuf][y1] >> (4 << 3)) & 0xff;`
`      }`
`      // Send UDP datagram`
`      sendto(mysocket, bufferudp, bufferudp_size, 0, (struct sockaddr *)&remote_addr,`
`       sizeof(remote_addr));`
`      numbuf = !numbuf;`
`    }`
`  }`
`}`

`void video() {`

`esp_intr_alloc(ETS_I2S0_INTR_SOURCE,ESP_INTR_FLAG_INTRDISABLED | ESP_INTR_FLAG_LEVEL1 | ESP_INTR_FLAG_IRAM, &i2s_isr_dmarb, NULL, &i2s_intr_handle_dmarb);`

`xSemaphore = xSemaphoreCreateBinary();`
`xTaskCreatePinnedToCore(&sendudp, "sendudp", 2048, NULL, 11, NULL, 1);`

`}`

`void setup() {`

`  video();`

`}`

`void loop() {`
` if (mp3->isRunning()) {`
`   if (!mp3->loop()) {` 
`      mp3->stop(); `     
`    }`
`  }`
`  ArduinoOTA.handle();`
`}`

`Decoding stack results
0x4008da68: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008dc99: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400860e1: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 143
0x4008620d: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 171
0x400fd7b3: _vfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 860
0x400f9acd: fiprintf at ../../../.././newlib/libc/stdio/fiprintf.c line 50
0x400f855d: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 59
0x4008b16d: vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h line 105
0x40089d59: xQueueGenericSendFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1184
0x40085072: emac_post at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/ethernet/emac_main.c line 1050
0x4008516d: emac_process_intr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/ethernet/emac_main.c line 627
0x4008b2c1: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x40089c05: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 828
0x4011eeaa: sys_mutex_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c line 93
0x4010c3e4: lwip_recvfrom_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c line 3398
0x400dba92: WiFiUDP::parsePacket() at C:\Users\mainuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/lwip/lwip/sockets.h line 585
0x400dc982: ArduinoOTAClass::handle() at C:\Users\mainuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src\ArduinoOTA.cpp line 367
0x400d3727: loop() at C:\Users\mainuser\Desktop\esp32_camara_i2s/esp32_camara_i2s.ino line 611
0x400ddf91: loopTask(void*) at C:\Users\mainuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x4008a1c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

me-no-dev, 2019-10-09:
see how nice your issue looks now? please paste code snippets properly.

bertmelis, 2019-10-09:
Is `sendudp` also called from ISR? You're not using `xSemaphoreTakeFromISR`.
You should also include the first lines of the crash report: it gives the crash reason.

JPMJPM, 2019-10-09:
Problem solved.

bufferudp_size is 2000 bytes > MTU.
With bufferudp_size = 1400 bytes < MTU no more crashes.

Anyway, 2000 bytes works in IDF because LWIP "Enable fragment outgoing IP packets" option is seted.

Is there any way to configure this LWIP option in arduino-esp32 ?.

Regards.

closed 2019-10-10 by JPMJPM