espressif/arduino-esp32 #3340: task create to read i2c sensor

yo90bosses, 2019-10-09:
### Hardware:
Board:							nodemcu-32
Core Installation version:			1.0.3
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     ?
Upload Speed:					115200
Computer OS:                                         Windows 10

Hi. Im using a esp32 for a project that involves the esp32 reading data from a BNO055 over I2C. Problem was the communication takes 1.5 to 6ms. I tried putting the part of my code that gets the data from the BNO055 in a task on core 1 (xTaskCreatePinnedToCore). But the ESP32 immediatly crashed and gives this error:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5800
entry 0x4008069c
Starting Magnetometer...
E (1390) FreeRTOS: FreeRTOS Task "getBNODataCode" should not return, Aborting now!
abort() was called at PC 0x400882d7 on core 1

Backtrace: 0x4008c29c:0x3ffb46b0 0x4008c4c9:0x3ffb46d0 0x400882d7:0x3ffb46f0

Code: 

```cpp
void getBNODataCode( void * pvParameters) {

    sensors_event_t orientationData , angVelocityData , linearAccelData;
    bno.getEvent(&angVelocityData, Adafruit_BNO055::VECTOR_GYROSCOPE);
    bno.getEvent(&linearAccelData, Adafruit_BNO055::VECTOR_LINEARACCEL);

    imu::Quaternion qbuf = bno.getQuat();

    attitude.w = qbuf.w();
    attitude.x = qbuf.y();
    attitude.y = qbuf.x();
    attitude.z = -qbuf.z();

}
```

attitude and bno are global objects

lbernstone, 2019-10-09:
Tasks should not return.  Move everything to do with the sensor into this task, have it initialize the device and have any values that can be declared.  Then, put your reading bits into a while or for loop.
If this seems a bit much, then take a look at the [Ticker](https://github.com/espressif/arduino-esp32/tree/master/libraries/Ticker) library.

closed 2019-12-22 by stale[bot]

anuj-tweek, 2022-06-01:
Hi, is the problem resolved? if so please can you share how?
Thanks