espressif/arduino-esp32 #3342: problem with esp_sleep_get_ext1_wakeup_status()

atlask1, 2019-10-10:
### Hardware:
Board:							ESP32 Module Custom 
Core Installation/update date:			?11/jul/2017?
IDE name:					         Arduino IDE
Flash Frequency:					  80Mhz
PSRAM enabled:                                          no
Upload Speed:				           115200
Computer OS:                                            Ubuntu

Hi, I'm trying the 'deepsleep example' of Arduino fw with esp32, in particular the ext1 wakeup, using 2 pins (25, 26) as interrupt with external pulldown and this configuration `esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK,ESP_EXT1_WAKEUP_ANY_HIGH)` .
Problem is:
also with `#define BUTTON_PIN_BITMASK 0x6000000 //pin 25 and 26` ,sometimes don't return the correct pin, return 0. ESP32 wakeup from sleep and enter in the 'gpio read function' but return 0 also if pin 25 or 26 is pressed.
I've tryed to debug 'GPIO_reason' to understand what happen, but I really don't understand.
I post the sketch but it's in the example of Arduino fw.
`
#define BUTTON_PIN_BITMASK 0x6000000 // pins 25 and 26

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}
void print_GPIO_wake_up(){
  unsigned int GPIO_reason = esp_sleep_get_ext1_wakeup_status();
  unsigned int GPIO_reason_log = (log(GPIO_reason))/log(2);
  Serial.println("INSIDE GPIO WAKEUP");
  Serial.print("GPIO_REASON: ");
  Serial.println(GPIO_reason);
  Serial.print("GPIO_REASON_LOG: ");
  Serial.println(GPIO_reason_log);
}
void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();
  print_GPIO_wake_up();
  /*
  First we configure the wake up source
  We set our ESP32 to wake up for an external trigger.
  There are two types for ESP32, ext0 and ext1 .
  ext0 uses RTC_IO to wakeup thus requires RTC peripherals
  to be on while ext1 uses RTC Controller so doesnt need
  peripherals to be powered on.
  Note that using internal pullups/pulldowns also requires
  RTC peripherals to be turned on.
  */
 // esp_sleep_enable_ext0_wakeup(GPIO_NUM_33,1); //1 = High, 0 = Low

  //If you were to use ext1, you would use it like
  esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK,ESP_EXT1_WAKEUP_ANY_HIGH);

  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}`

this is debug when return the correct pin:
`rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
;T�U���2-hal-cpu.c:168] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Boot number: 11
Wakeup caused by external signal using RTC_CNTL
INSIDE GPIO WAKEUP
GPIO_REASON: 67108864
GPIO_REASON_LOG: 26
Going to sleep now
`
and this is debug when return 0:
`ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
�T�U���2-hal-cpu.c:168] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Boot number: 12
Wakeup caused by external signal using RTC_CNTL
INSIDE GPIO WAKEUP
GPIO_REASON: 0
GPIO_REASON_LOG: 2147483648
Going to sleep now
`
the same with pin 25.
what I see is that esp identify correctly the wake up source (RTC_CNTL), go inside funtion to define wich pin wake up but GPIO_reason is 0 so log cannot be one of 2 pins of bitmask.
You can see GPO_reason defined 'unsigned' just to try, because my first test with this behaviour return negative log number...
thanks

closed 2019-12-23 by stale[bot]

victalmeida, 2021-03-05:
This topic has ben closed, but just for Learn and studies reason.This problems happens because the macro uint64_t esp_sleep_get_ext1_wakeup_status(void) return a uint64_t value. So if you use  uint32_t to store the value and try to acces pins that cause wake . Than the   higher pins will overflow you variable and the value will be 0;

EaverX, 2021-07-04:
you need uint64_t GPIO_reason rather than unsigned int GPIO_reason

flusflas, 2021-07-25:
I'm facing the same problem. I've tried using pins 25 and 26, and also 32 and 33. Same behavior. I'm using `uint64_t` to store the value from `esp_sleep_get_ext1_wakeup_status()`, so that's not the problem. **Sometimes** I get the right value (the pin number that woke up the device), but most of the time `esp_sleep_get_ext1_wakeup_status()` returns 0, so I think I'm using the function correctly but there is some inconsistency with the returned value.

victalmeida, 2021-07-26:
Hi Flusflas.
Try to check all this steps.

I am not shure but it seems that esp32 doesnt know whats is going on, so first you need to be sure that everything is ok with esp, so then we check the code. physical problem can make esp lose reference, Check the pin Mask, chech the math function, and check the code.
So here a few step to check the health of esp32.

1 - Be sure your physical parts are ok. Low Level Voltage can cause a dumb esp32
2 - When you use a pinMask to Woke a esp32, esp only accept high sinal, So check with this HIGH sinal is stable for esp
2 - Check the Mask pin and check the mask is matching with the right pins.
3- Try to debug if zero came before math function converts mask Pins, or After.
4 - If you user some frameworks try to clear cached memory recompile and build again.
5 - check if this 0 occour with all pins or with a specific pin.