espressif/arduino-esp32 #3344: wildcard for httpd_register_uri_handler

Skyseb, 2019-10-10:
Hi.

I need to match /* wildcard in http server

in this example from idf sdk

https://github.com/espressif/esp-idf/blob/5c88c5996dbde6208e3bec05abc21ff6cd822d26/examples/protocols/http_server/file_serving/main/file_server.c

but in arduino:

httpd_config_t config = HTTPD_DEFAULT_CONFIG();

      /* Use the URI wildcard matching function in order to
      * allow the same handler to respond to multiple different
      * target URIs which match the wildcard scheme */
      
      config.uri_match_fn = httpd_uri_match_wildcard;

error: 'httpd_config_t {aka struct httpd_config}' has no member named 'uri_match_fn'

need update ?

lbernstone, 2019-10-10:
arduino-esp32 is based on IDF v3.2.  You will need to use examples from that release.

Skyseb, 2019-10-11:
OK understood.

I base my project on the CameraWebServer
not for stream, its take imgs and save its in sdcard every 1 min

then i need to access to view/download this images over http.

loking example "SDWebServer".. is simple

but I intended it to keep working with httpd, more taking into account that in version 3.3 it is already resolved.

Is it possible that "libesp_http_server.a" can be compiled to work with version 3.2?

How can I try to do this?

lbernstone, 2019-10-11:
There is an experimental branch for v3.3.  You are welcome to try it.

Skyseb, 2019-10-11:
Thank. Ill try it!!

closed 2019-10-11 by Skyseb