espressif/arduino-esp32 #3346: ESP Pico D4 - Wifi init causing boot loop

walkerdanny, 2019-10-11:
### Hardware:
Board:							Custom ESP32-PICO-D4 based board
Core Installation/update date:		11/10/2019 (today)
IDE name:						ESP IDF (Stable version updated today) or Arduino 1.8.10
Flash Frequency:					80MHz
PSRAM enabled:                                      No
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
I'm developing a custom board which uses the ESP32-PICO-D4. It will happily run simple Arduino sketches which send output over UART, but as soon as I enable the Wifi I get a boot loop, seemingly initiated by the RTC watchdog timer.

I've flashed the sketch using the ESP IDF and changed any settings in the config that I thought might make a difference, but nothing has changed (although it gave me a better level of debug output).

It's not out of the question that it's a hardware issue, but I can't see any obvious cold joints etc. so I'd rather exhaust the software options before soldering a new chip on.

The last line of the debug output occasionally gets as far as ```W (334) phy_init: fail``` before the debug stream turns into a garbage sequence of characters (as below) and the watchdog timer resets it and I get ```rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)```

Can anyone suggest a fix? Thanks.

### Minimal sketch: 
```cpp

#include "Arduino.h"
#include <WiFi.h>

void setup(){
  Serial.begin(115200);
  Serial.println("Trying to start the WiFi");
  WiFi.mode(WIFI_AP); // Sketch doesn't execute past this point
  Serial.printf("Starting AP test");
  WiFi.softAP("test");
}

void loop(){
  Serial.println("loop");
  delay(1000);
}
```
### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:8696
load:0x40078000,len:12396
load:0x40080400,len:7116
entry 0x400807a4
D (67) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (36) boot: ESP-IDF v3.3 2nd stage bootloader
I (36) boot: compile time 14:45:03
D (36) boot: Enabling RTCWDT(9000 ms)
I (39) boot: Enabling RNG early entropy source...
D (45) boot: magic e9
D (47) boot: segments 04
D (49) boot: spi_mode 02
D (52) boot: spi_speed 0f
D (55) boot: spi_size 02
I (57) boot: SPI Speed      : 80MHz
I (62) boot: SPI Mode       : DIO
I (66) boot: SPI Flash Size : 4MB
D (70) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (78) boot: mapped partition table 0x8000 at 0x3f408000
D (84) flash_parts: partition table verified, 3 entries
I (89) boot: Partition Table:
I (93) boot: ## Label            Usage          Type ST Offset   Length
D (100) boot: load partition table entry 0x3f408000
D (105) boot: type=1 subtype=2
I (108) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (116) boot: load partition table entry 0x3f408020
D (121) boot: type=1 subtype=1
I (124) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (131) boot: load partition table entry 0x3f408040
D (136) boot: type=0 subtype=0
I (139) boot:  2 factory          factory app      00 00 00010000 00100000
I (147) boot: End of partition table
D (151) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (157) esp_image: reading image header @ 0x10000
D (162) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (169) esp_image: image header: 0xe9 0x06 0x02 0x02 4008115c
I (175) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18424 ( 99364) map
D (184) esp_image: free data page_count 0x00000032
D (189) bootloader_flash: mmu set paddr=00010000 count=2 size=18424 src_addr=10020 src_addr_aligned=10000
D (227) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (228) esp_image: segment 1: paddr=0x0002844c vaddr=0x3ffb0000 size=0x030f8 ( 12536) load
D (232) esp_image: free data page_count 0x00000032
D (237) bootloader_flash: mmu set paddr=00020000 count=1 size=30f8 src_addr=2844c src_addr_aligned=20000
D (251) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (253) esp_image: segment 2: paddr=0x0002b54c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/Danny/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1779

D (262) esp_image: free data page_count 0x00000032
D (267) bootloader_flash: mmu set paddr=00020000 count=1 size=400 src_addr=2b54c src_addr_aligned=20000
D (277) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
I (283) esp_image: segment 3: paddr=0x0002b954 vaddr=0x40080400 size=0x046bc ( 18108) load
D (292) esp_image: free data page_count 0x00000032
D (297) bootloader_flash: mmu set paddr=00020000 count=2 size=46bc src_addr=2b954 src_addr_aligned=20000
D (313) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
I (314) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x695f0 (431600) map
0x400d0018: _stext at ??:?

D (322) esp_image: free data page_count 0x00000032
D (327) bootloader_flash: mmu set paddr=00030000 count=7 size=695f0 src_addr=30018 src_addr_aligned=30000
D (463) bootloader_flash: mmu set block paddr=0x00090000 (was 0xffffffff)
I (463) esp_image: segment 5: paddr=0x00099610 vaddr=0x40084abc size=0x0b568 ( 46440) load
0x40084abc: phy_get_fetx_delay at ??:?

D (467) esp_image: free data page_count 0x00000032
D (472) bootloader_flash: mmu set paddr=00090000 count=2 size=b568 src_addr=99610 src_addr_aligned=90000
D (498) bootloader_flash: mmu set block paddr=0x000a0000 (was 0xffffffff)
D (498) esp_image: Calculated hash: ab7a1c24fd6d85b90a0953cfaf1aad1f77895411dde11b733f54d7c8acc782d8
D (503) bootloader_flash: mmu set paddr=000a0000 count=1 size=20 src_addr=a4b80 src_addr_aligned=a0000
D (513) bootloader_flash: mmu set paddr=000a0000 count=1 size=20 src_addr=a4b80 src_addr_aligned=a0000
I (531) boot: Loaded app from partition at offset 0x10000
I (531) boot: Disabling RNG early entropy source...
D (533) boot: Mapping segment 0 as DROM
D (537) boot: Mapping segment 4 as IROM
D (541) boot: calling set_cache_and_start_app
D (546) boot: configure drom and irom and start
D (550) boot: start: 0x4008115c
0x4008115c: call_start_cpu0 at C:/Users/Danny/Desktop/esp-idf/components/esp32/cpu_start.c:122

I (554) cpu_start: Pro cpu up.
I (557) cpu_start: Application information:
I (562) cpu_start: Project name:     esp-idf
I (567) cpu_start: App version:      1
I (571) cpu_start: Compile time:     Oct 11 2019 14:45:00
I (578) cpu_start: ELF file SHA256:  81cdb286cb5b7083...
I (584) cpu_start: ESP-IDF:          v3.3
I (588) cpu_start: Starting app cpu, entry point is 0x40081108
0x40081108: call_start_cpu1 at C:/Users/Danny/Desktop/esp-idf/components/esp32/cpu_start.c:270

I (0) cpu_start: App cpu up.
I (599) heap_init: Initializing. RAM available for dynamic allocation:
I (606) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (612) heap_init: At 3FFB9158 len 00026EA8 (155 KiB): DRAM
I (618) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (624) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (631) heap_init: At 40090024 len 0000FFDC (63 KiB): IRAM
I (637) cpu_start: Pro cpu start user code
I (207) cpu_start: Chip Revision: 1
I (207) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Trying to start the WiFi
I (30) wifi: wifi driver task: 3ffc2d44, prio:23, stack:3584, core=0
I (272) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (272) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (302) wifi: wifi firmware version: aeed694
I (302) wifi: config NVS flash: enabled
I (302) wifi: config nano formating: disabled
I (302) wifi: Init dynamic tx buffer num: 32
I (302) wifi: Init data frame dynamic rx buffer num: 32
I (312) wifi: Init management frame dynamic rx buffer num: 32
I (312) wifi: Init management short buffer num: 32
I (322) wifi: Init static rx buffer size: 1600
I (322) wifi: Init static rx buffer num: 10
I (332) wifi: Init dynamic rx buffer num: 32
W (332) ph  ets Ju  8 201��.�j�A����X
        ����Um����.W��Ea@���蕌ꥥ������
��������%}z�U����3�V/iAb���ӫ�Ҳ�M����������0��Ȳ���<HLK���t����VZ��t��������t����TX��t����V������a�CUi�����,�3�A�Ҷ��#!�Eia𳶳�H&%k��2)���Ⱥ������q�ʮ���������t����������ݺ��f�������t�����\LML,'k�����5ˮ��>$&/  �څQ�����"$��C��¤��%��樚���.�n�I��3�A�¤��$/&&  i��@����3SY��+�ԁ�5�I
```

lbernstone, 2019-10-11:
Probably power issues.  Use a different USB port or a separate power supply.

walkerdanny, 2019-10-12:
Thanks for the response - I'm powering it from a bench top PSU that's good for 2A so I doubt it's a lack of current. I can try monitoring the supply voltage as the WiFi initialises to see if there's a significant ripple when I have access to a scope.

lbernstone, 2019-10-12:
I missed the custom part of your post. Definitely check the power.  I have found that 4.7uF bypass and 100nF decoupling capacitor are enough to stabilize.
Do you have a lot of external devices that might be drawing power through the MCU?  It would be suspicious that the timing is always as the wifi starts, but perhaps put in a long delay before your WiFi.begin and see if it something else on the rail causing the dip.

walkerdanny, 2019-10-23:
I had a poke with a scope - there's a 0.1V drop on the supply line during WiFi init, which I'm not convinced is a huge problem. No amount of capacitance seems to make a difference to how far it'll get through the initialisation process.

I'm thinking it's something clock related, as the reset reason is the RTC watchdog timer kicking in, and the serial output becomes nonsense that can't be decoded. I think next I'll have to have a look at the garbage output it gives me after the crash to see if it's at a different baud rate or something.

lbernstone, 2019-10-23:
It won't hurt to look at other avenues, but I'd say it is 95% likely a power issue.  You could also try dropping the cpu speed before you init the WiFi with `setCpuFrequencyMhz(80)` which will reduce your power consumption by ~40mA and might give you enough cold cranking amps to fire that solenoid :smile:.

closed 2020-01-05 by stale[bot]

TaXue-DuBa, 2022-02-01:
Hello, I also have this problem. The develop board created by myself. When I use the `WiFi.begin()`. My board also reboot....

hgatesCPV, 2022-05-26:
I designed a board for ESP32-PICO-V3-02 and am having a similar issue - constant reboot loop with 
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) 
as soon as I try to turn on Ethernet (wired, with W5500 chip). 

I have the ESP32-PICO-D4 eval board from Espressif and that one does not have the same problem. 

But it is strange because I can build code, load it, blink LEDs, send serial messages. But when I enable the Ethernet (via SPI) it enters a reboot loop.

hgatesCPV, 2022-05-26:
I tried putting a 1K pullup on pin 18 since I saw that in the reference schematic. But no luck.

lbernstone, 2022-05-26:
Your issue is different than the OP.  Open a new issue and fill out the necessary information for assistance.