espressif/arduino-esp32 #3347: HTTPClient issues above 1.0.2

u4mzu4, 2019-10-11:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							Olimex ESP32-EVB
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.10
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I try to read some temperatures from ESP8266 webservers via http. It's working properly until 1.0.3/1.0.4. 1.0.2 is the latest stable version for the code below.
Is there any bug in my original code, or there is some unexpected behavior after the code update?
Errors:
- InstrFetchProhibited
- IllegalInstruction

### Sketch:  (leave the backquotes for [code formatting]
```cpp

//Includes
#include <HTTPClient.h>

//Defines
#define TIMEOUT   5000  //5 sec
#define NROFTRANSM 2

//Global variables
const char* host[2]      = {"http://192.168.178.53/","http://192.168.178.56/"};
const char* ssid      = "";
const char* password  = "";
float actualTemperature;
float transData[NROFTRANSM];

//Init services
HTTPClient webclient;

void ReadTransmitter() 
{
  static float lastvalidtransTemp[NROFTRANSM];

  for (int i = 0; i <NROFTRANSM; i++) 
  {
    webclient.begin(host[i]);
    webclient.setConnectTimeout(500);
    if(HTTP_CODE_OK == webclient.GET()) 
    {
      transData[i] = webclient.getString().toFloat();
    }
    else
    {
      transData[i] = 0.0;
    }
    webclient.end();
    Serial.print(i);
    Serial.print(": ");
    Serial.println(transData[i]);
  }
}

void setup() {
  Serial.begin(115200);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.begin (ssid, password);
}

void loop() {
ReadTransmitter();
delay(10000);
}

### Debug Messages:
```
 Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x07c00061  PS      : 0x00060630  A0      : 0x800d2860  A1      : 0x3ffb1eb0  
A2      : 0x3ffc1074  A3      : 0x07c00061  A4      : 0x00000001  A5      : 0x3ffb1ed8  
A6      : 0x3ffc10a8  A7      : 0x40404040  A8      : 0x8015d362  A9      : 0x3ffb1ec0  
A10     : 0x3ffba878  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000014  EXCCAUSE: 0x00000014  
EXCVADDR: 0x07c00060  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000  
Backtrace: 0x07c00061:0x3ffb1eb0 0x400d285d:0x3ffb1ed0 0x400d2951:0x3ffb1f00 0x400d2a1f:0x3ffb1f40 0x400d17ae:0x3ffb1f60 0x400d18af:0x3ffb1f90 0x400d5ec1:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
0: 23.50
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Core 1 register dump:
PC      : 0x6107c000  PS      : 0x00060630  A0      : 0x800d1b76  A1      : 0x3ffb1ec0  
A2      : 0x3ffc1074  A3      : 0x6107c000  A4      : 0x3ffc15c8  A5      : 0xeefa5af9  
A6      : 0x05000000  A7      : 0x05000000  A8      : 0x8015d362  A9      : 0x3ffb1eb0  
A10     : 0x3ffba7c0  A11     : 0x00000000  A12     : 0x0000001c  A13     : 0x3ffba7e0  
A14     : 0x00ff0000  A15     : 0x00000001  SAR     : 0x00000014  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
Backtrace: 0x6107c000:0x3ffb1ec0 0x400d1b73:0x3ffb1ee0 0x400d1be5:0x3ffb1f00 0x400d2d2d:0x3ffb1f20 0x400d1793:0x3ffb1f60 0x400d18af:0x3ffb1f90 0x400d5ec1:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0
```

Decoding stack results
0x400d285d: HTTPClient::connect() at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 989
0x400d2951: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 552
0x400d2a1f: HTTPClient::GET() at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 482
0x400d17ae: ReadTransmitter() at C:\Users\Andrew\Documents\Arduino\sketch_oct11a/sketch_oct11a.ino line 26
0x400d18af: loop() at C:\Users\Andrew\Documents\Arduino\sketch_oct11a/sketch_oct11a.ino line 62
0x400d5ec1: loopTask(void*) at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40088b9d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Decoding stack results
0x400d1b73: HTTPClient::disconnect(bool) at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 359
0x400d1be5: HTTPClient::end() at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 347
0x400d2d2d: HTTPClient::begin(String) at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 216
0x400d1793: ReadTransmitter() at C:\Users\Andrew\Documents\Arduino\sketch_oct11a/sketch_oct11a.ino line 24
0x400d18af: loop() at C:\Users\Andrew\Documents\Arduino\sketch_oct11a/sketch_oct11a.ino line 62
0x400d5ec1: loopTask(void*) at C:\Users\Andrew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40088b9d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

u4mzu4, 2019-10-12:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.178.42, MASK: 255.255.255.0, GW: 192.168.178.1
[V][HTTPClient.cpp:235] beginInternal(): url: http://192.168.178.53/
[D][HTTPClient.cpp:276] beginInternal(): host: 192.168.178.53 port: 80 url: /
[D][HTTPClient.cpp:1025] connect():  connected to 192.168.178.53:80
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Type: text/plain'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Access-Control-Allow-Origin: *'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Length: 5'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1161] handleHeaderResponse(): size: 5
[D][HTTPClient.cpp:1295] writeToStreamDataBlock(): connection closed or file end (written: 5).
[D][HTTPClient.cpp:370] disconnect(): tcp stop

[D][HTTPClient.cpp:383] disconnect(): tcp is closed

30
[D][HTTPClient.cpp:214] begin(): mix up of new and deprecated api
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060630  A0      : 0x800d1ec7  A1      : 0x3ffb1ed0  
A2      : 0x3ffc114c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060820  A7      : 0x00000000  A8      : 0x8015e8a6  A9      : 0x3ffb1e70  
A10     : 0x3ffba878  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x00000001  
A14     : 0x00060823  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffb1ed0 0x400d1ec4:0x3ffb1ef0 0x400d1f95:0x3ffb1f10 0x400d348e:0x3ffb1f30 0x400d1abd:0x3ffb1f70 0x400d77d5:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0

u4mzu4, 2019-10-19:
@Jeroen88 , without #2867 the sketch above is working properly. How can I improve it to working on core 1.0.4 as well?

Jeroen88, 2019-10-21:
Could you try

```
void loop() {
  const char* host = "http://192.168.178.53/";
  float transData;
  
  {
    HTTPClient webclient;
    webclient.begin(host);
    if(HTTP_CODE_OK == webclient.GET()) 
    {
      transData = webclient.getString().toFloat();
    }
    webclient.end();
  }
  Serial.println(transData);
  delay(10000);
}
```

u4mzu4, 2019-10-21:
Tested and works, thank you! It looks like a workaround instead of solution, but I can live with it. :-) 
Thanks again!

closed 2019-10-21 by u4mzu4

Jeroen88, 2019-10-21:
You are welcome :) 

You are using deprecated functions, you'd better use the new functions that pass a client (like begin(client,...)). This is indeed a workaround because it is for a deprecated function.

dronecz, 2019-11-07:
Hello @Jeroen88 , 

could you please elaborate what you mean by this: 

>  You are using deprecated functions, you'd better use the new functions that pass a client (like begin(client,...)). 

I have similar problem as @u4mzu4 and I do not use just **.begin()** in my code. 

Here is example of my code: 
```
void heartBeat() {
  if (Debug.isActive(Debug.ANY)) {
    if (wifiStatus == WL_CONNECTED) {
      String heartMessage = "";
      heartMessage += "{\"configVersion\":";
      heartMessage += configVersion;
      heartMessage += "}";
      char heartMessageChar[heartMessage.length() + 1];
      heartMessage.toCharArray(heartMessageChar, heartMessage.length() + 1);
      http.begin("https://fabman.io/api/v1/bridge/heartbeat", root_ca );  //Specify destination for HTTP request
      http.addHeader("Content-Type", "application/json");             //Specify content-type header
      http.addHeader("Accept", "application/json");
      http.addHeader("Authorization", API_key);
      int httpResponseCode = http.POST(heartMessageChar);   //Send the actual POST request
      if (Debug.isActive(Debug.ANY)) {
        Debug.print("Heartbeat message: ");
        Debug.println(heartMessageChar);
        payload = http.getString();
        Debug.print("JSON from server:" + (payload));
        //Serial.println(payload);
        Debug.println();
        if (httpResponseCode < 0) {
          online = 0;
          connection();
        }
        else if (httpResponseCode > 0) {
          Debug.print("Responde code: ");
          Debug.println(httpResponseCode);   //Print return code
        } else {
          Debug.print("Error on sending request: ");
          Debug.println(httpResponseCode);
        }
      }
      get_configVersion();
      http.end();  //Free resources
    }
  }
}
```

Here is log when it crash: 
```

[V][HTTPClient.cpp:235] beginInternal(): url: https://fabman.io/api/v1/bridge/heartbeat
[D][HTTPClient.cpp:276] beginInternal(): host: fabman.io port: 443 url: /api/v1/bridge/heartbeat
[D][HTTPClient.cpp:991] connect(): already connected, reusing connection
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Server: nginx/1.15.3'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Date: Wed, 06 Nov 2019 21:40:58 GMT'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Type: application/json; charset=utf-8'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Length: 15'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'cache-control: no-cache'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Security-Policy: default-src 'self' ; script-src 'self' *.braintreegateway.com https://js.stripe.com https://beacon-v2.helpscout.net https://www.google-analytics.com ; style-src 'self' 'unsafe-inline' https://fonts.googleapis.com ; font-src 'self' https://fonts.gstatic.com ; img-src 'self' data: https://beaconapi.helpscout.net https://d3hb14vkzrxvla.cloudfront.net https://d33v4339jhl8k0.cloudfront.net https://www.google-analytics.com https://stats.g.doubleclick.net https://www.google.com; object-src 'self' https://d3hb14vkzrxvla.cloudfront.net ; connect-src 'self' *.braintreegateway.com https://api.stripe.com https://beaconapi.helpscout.net https://chatapi.helpscout.net wss://ws-helpscout.pusher.com https://sockjs-helpscout.pusher.com https://d3hb14vkzrxvla.cloudfront.net https://www.google-analytics.com https://stats.g.doubleclick.net https://www.google.com; child-src 'self' *.braintreegateway.com https://js.stripe.com ; frame-src https://js.stripe.com https://hooks.stripe.com ; report-uri /api/v1/csp-violations'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1161] handleHeaderResponse(): size: 15
Heartbeat message: {"configVersion":2253657963}

[D][HTTPClient.cpp:1295] writeToStreamDataBlock(): connection closed or file end (written: 15).
[D][HTTPClient.cpp:370] disconnect(): tcp stop

[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
JSON from server:{"config":null}

Responde code: 200

Config version payload: {"config":null}

currentConfigVersion which is read: 0

No need to change configVersion in NVS

[D][HTTPClient.cpp:383] disconnect(): tcp is closed

[D][HTTPClient.cpp:188] begin(): mix up of new and deprecated api
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4017c7b4  PS      : 0x00060a30  A0      : 0x800dc827  A1      : 0x3ffb1e10  
A2      : 0x3ffc1890  A3      : 0xae300100  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x8008517c  A9      : 0x3ffb1db0  
A10     : 0x3ffce8fc  A11     : 0x00060c23  A12     : 0x00060c20  A13     : 0x00000001  
A14     : 0x00060c23  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xae300110  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x4017c7b4:0x3ffb1e10 0x400dc824:0x3ffb1e30 0x400dc8f5:0x3ffb1e50 0x400ddd59:0x3ffb1e70 0x400d61e9:0x3ffb1eb0 0x400df0f0:0x3ffb1f70 0x400d84e4:0x3ffb1f90 0x400ef54d:0x3ffb1fb0 0x40088ddd:0x3ffb1fd0
```
Many thanks for your answer.

u4mzu4, 2019-11-07:
Hello,
have you tried to move the 
HTTPClient http;
line inside the function heartBeat()?

Jeroen88, 2019-11-07:
@dronecz I am almost sure that I ported the HTTPClient library from the ESP8266 and that I left the deprecated functions out for the ESP32. I think someone brought them back in which was a bad idea.

You'd better use the functions [here](https://github.com/espressif/arduino-esp32/blob/91e095f5a76ec35f06c86689a87892425395b8db/libraries/HTTPClient/src/HTTPClient.h#L137) and none of the functions between #ifdef HTTPCLIENT_1_1_COMPATIBLE (like [here](https://github.com/espressif/arduino-esp32/blob/91e095f5a76ec35f06c86689a87892425395b8db/libraries/HTTPClient/src/HTTPClient.h#L140))

dronecz, 2019-11-07:
Hello @u4mzu4 , 

no I did not and to be honest I do not want to as I use this definition in some other functions as well. 
What I do not understand is, that is working normally for some time and it using this call `[HTTPClient.cpp:235] beginInternal(): ` and after that it use out of nowhere (at least from log) this call `[HTTPClient.cpp:188] begin():`. 

I had no issue before I upgraded to 1.0.4 . 

to @Jeroen88 
so I should call it like this: `http.begin( *http, url );` and if I want to use SSL I should migrate to [WiFiClientSecure ](https://github.com/espressif/arduino-esp32/blob/91e095f5a76ec35f06c86689a87892425395b8db/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino#L103)library?    

Thanks

u4mzu4, 2019-11-09:
Hello @dronecz ,

function is coming from ESP8266, but the examples are not updated (yet), so I've checked the 8266 samples for the "new" syntax. Here you are:
''
void loop() {
    WiFiClient client;
    HTTPClient http;
    Serial.print("[HTTP] begin...\n");
    if (http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html")) {  // HTTP

roy-zahor, 2019-11-11:
I have the same issue, and it seems that using the "new" API solves the problem.

Thanks @u4mzu4 @Jeroen88

dronecz, 2019-11-11:
Hello @u4mzu4 

I changed my code to this to use HTTPS: 
```
WiFiClientSecure client;
HTTPClient http;
void hearbeat() {
if (http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html")) {  // HTTPS
}
```
and so far so good. I hope this will solve my issues. 

Many thanks to all for your help. 

> void loop() {
>     WiFiClient client;
>     HTTPClient http;
>     Serial.print("[HTTP] begin...\n");
>     if (http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html")) {  // HTTP
> ```

I

H3ndrik-Jan, 2019-11-12:
> You'd better use the functions [here](https://github.com/espressif/arduino-esp32/blob/91e095f5a76ec35f06c86689a87892425395b8db/libraries/HTTPClient/src/HTTPClient.h#L137) and none of the functions [...]

Hi, I encountered the same issues @u4mzu4 started this thread with, and following @Jeroen88 s advice (as in the quote) solved my problem!
Much thanks!

Jeroen88, 2019-11-12:
You're welcome :)

If anyone is interested, the new api can be used with a secure WiFi client too. Just prepare the security on the client and then pass it to http client. If I recall correctly esp8266 has some examples

bemar76, 2020-04-15:
Solved also my problem. Thank you