espressif/arduino-esp32 #3348: CameraWebServer: How can I save settings?

heikoh81, 2019-10-11:
In CameraWebServer:

How can I save settings applied in the Webinterface, e.h. HMirror=On & resolution, so that they are restored on power loss?
(If power fails and comes back, I’d like to have my settings to be the same than before the power loss).
I can’t find a save button. Is there a way to hard code my own “defaults” in the script files, if yes, where?

me-no-dev, 2019-10-14:
you need to do this in the firmware and not in the web page. Have a look at the `Preferences` library included here and then plug that in the code to save and restore then. You save when you change something from the interface and you restore on boot after you init the camera.

closed 2019-10-14 by me-no-dev

heikoh81, 2019-10-14:
Sorry, I don't understand your last sentence.
How can I save when I change somethin from the interface?

Can you point me to the right piece of source code where I can hardcode my preferences for Hmirror and Resolution?

me-no-dev, 2019-10-14:
did you even read the example code? :)
settings for m5stack wide mirror+flip: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino#L83-L84
settings for resolution: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino#L80
And really... the main settings: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino#L28-L58

Everything is there in the example. It is an example. You have to do some work on your own :)

heikoh81, 2019-10-14:
Thanks for your answer. I tried to read it...

So in "s->set_framesize(s, FRAMESIZE_QVGA);",
"FRAMESIZE_QVGA" is the default value, and if I change that, recompile and reflash, then the Webinterface will load with my own default value?

On my test system, Hmirror is turned off by default.
However,
"s->set_hmirror(s, 1);"
implies that it is 1=on by default?

me-no-dev, 2019-10-14:
It sounds like you are not familiar with the programming language :)
The code is pretty basic. If you have issues understanding it, I will suggest you poke arround and see what happens and get to learn a bit. This tracker is to collect issues and bugs with the code and not really to help users with their code. We have a gitter channel (link in the readme). You can drop by and someone might have time and will to help out :)

rajthuvanoor, 2021-12-07:
See this video on how to set default resolution: https://youtu.be/p32ms9NaxBE. For HMirror just add this line below "s->set_framesize(s, FRAMESIZE_QVGA);" :  `s->set_hmirror(s, 1);     // 0 = disable , 1 = enable
`