espressif/arduino-esp32 #3350: CPU Freq Setting Chip ESP32-D0WDQ6

f0rlive3D, 2019-10-11:
Hi everyone, I was looking for help. I got an ESP32-WROOM-32 nodemcu module that uses the ESP32-D0WDQ6 chip according to the data sheet. The data sheet also mentions that the CPU frequency is adjustable from 80MHz to 240MHz.

My question is how do I change the frequency of the chip to 240MHz?

It is currently running at 160MHz.

I was trying but I did not succeed. Can somebody help me?

From already thank you very much.

stickbreaker, 2019-10-11:
Some of the ESP32 chips are frequency limited because of yield testing.  They are downgraded by the factory.  The maximum frequency is burned into fuses.  

If your module does not respond when you change frequency I would suggest you verify your module with your supplier.

https://github.com/espressif/arduino-esp32/blob/79e4339582ca0891a6e6e35220c90db613c0254d/cores/esp32/esp32-hal-cpu.h#L38
```c++
//function takes the following frequencies as valid values:
//  240, 160, 80    <<< For all XTAL types
//  40, 20, 10      <<< For 40MHz XTAL
//  26, 13          <<< For 26MHz XTAL
//  24, 12          <<< For 24MHz XTAL
bool setCpuFrequencyMhz(uint32_t cpu_freq_mhz);
```
chuck.

lbernstone, 2019-10-11:
You can also do this in the Arduino IDE.  Select the "ESP32 Dev Module" as your board.  You will then have a CPU Frequency option in tools.
[image]

me-no-dev, 2019-10-14:
If you set `Core Debug Level` to something other than `None` and you get [this error message](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-cpu.c#L159) then your ESP is not rated for 240MHz

javfernandez, 2019-10-15:
@me-no-dev please, can you tell me how to set de Core Debug Level? I just cant find the define declaration

atanisoft, 2019-10-15:
@javfernandez set board type to "ESP32 Dev Module" as shown in @lbernstone image above and the Core Debug Level menu option will be available.

javfernandez, 2019-10-15:
@atanisoft thanks!!

javfernandez, 2019-10-15:
Hello, I bougth a NodemCPU ESP32 in china and when I download a code with the Arduino IDE, it reset continuosly. Using the esptools I discover that this chip has a 160Mhz CPU Freq and Arduino IDE or Atom only compile for 240Mhz, thats why Im trying to change the CPU Frequency to 240.

As ESP continously resets with arduino IDE I can not see the message if CPU Freq has changed or not.

I can program the ESP at 160MHz only with espressif IDE, but I could not change the CPU Freq either

Some advise??

lbernstone, 2019-10-15:
You can use the ESP32 dev module board type with a NodeMCU32.  This will allow you to set the CPU frequency.  LOOK AT THE PICTURE I POSTED.

javfernandez, 2019-10-15:
@lbernstone I tried that, but as Arduino IDE libraries are compiled for 240Mhz only, the ESP32 continuosly reset after loading the sketch

me-no-dev, 2019-10-16:
on 1.0.4 all bootloaders are compiled for 160MHz. Frequency is switched afterwards. If your chip can not do 240, a message should come on to tell you that (debug)

closed 2019-12-29 by stale[bot]