espressif/arduino-esp32 #3351: Using arduino-esp32 as ESP-IDF 4.x component in Eclipse

Sterpa, 2019-10-13:
Dear All,
We have good [instruction](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md) and additional issues ([1](https://github.com/espressif/arduino-esp32/issues/1449)) ([2](https://github.com/espressif/arduino-esp32/issues/1840)) that explaned how to use arduino-esp32 as ESP-IDF 3.x (GNU) component, include external librares and build project.
But it seems not work on ESP-IDF 4.x (CMake)...
I try to use [instruction](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md) for last ESP-IDF 4.
I clone arduino-esp32 repository to "components" subdirrectory and run `idf.py menuconfig` (not `make menuconfig`)
and oops... there is no any options for Arduino, no "Autostart Arduino setup and loop on boot"!

Q2: What's wrong with `idf.py menuconfig` on ESP-IDF 4.x, where is Arduino option?

Next point - all of our instruction and issues based on cloning arduino-esp repository in to project subdir "components". But now Eclipse have built-in Arduino plugin, and we could add ESP32 board with just adding [package_esp32_dev_index.json](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) url in Arduino settings in Eclipse.
This will install arduino-esp32 repository to default folder `C:\Users\UserName\.arduinocdt`.
All external libraries will be in `C:\Users\UserName\.arduinocdt\libraries`
All build-in libraries will be in `C:\Users\UserName\.arduinocdt\packages\esp32\hardware\esp32\1.0.4-rc1\libraries`

Q2: Is it possible to use `C:\Users\UserName\.arduinocdt` arduino-esp32 repository in ESP-IDF project, without cloning it in to project directory?

I tryed to change COMPONENT_PATH in this issue like 
`COMPONENT_PATH += C:\Users\UserName\.arduinocdt` but it is not worked...

Thanks a lot for ideas!

lbernstone, 2019-10-13:
The cmake environment is not yet set up to allow arduino as a component in IDF 4.0.  This will likely arrive first in https://github.com/espressif/esp32-arduino-lib-builder.
For your second question, use a filesystem that allows symbolic links.  I link arduino and other libraries all the time, saving a lot of space.  Compilation is also 3x faster in linux compared to Windows.

Sterpa, 2019-11-15:
Well...
I try to use esp-idf-v3.3 + arduino-esp32(branch/idf-release/v3.3) It works well, but only in GNU make mode, and only in command prompt...
In Eclipse i got horrible errors with ([failing to find all header paths](https://github.com/espressif/esp-idf/issues/17)).
Using  esp-idf-v4 ippossible, because:
1. arduino-esp32 support maximum v.3.3
2. idf-eclipse-plugin don't support esp-idf-v3.3 (no cmake setup script)
3. arduino-esp32 have no release for idf-v3.3, so, i cannot install arduino-esp32-v3.3 in eclipse with package_esp32_index.json (you only have it for v.3.2)

This is hell circle...

Do you plane to provide package_esp32_index.json for arduino-esp32-v3.3 ?

me-no-dev, 2019-11-21:
soon :)

closed 2020-02-03 by stale[bot]

Ben79543, 2020-05-05:
Hi Espressif.
I also would like a way to use arduino with esp-idf 4.0. 
But as I understand arduino-esp32 is based on IDF 3.3 and I can't use it as a component for idf 4.0. It is then recommended to roll back the idf version to one compatible with the arduino component so 3.3.

Last arduino-esp32 release is from octobre and the gap with 3.3 is increasing... 
Will there be an arduino-esp32 release  soon?