espressif/arduino-esp32 #3352: com

rahul2244, 2019-10-13:
Make your question, not a Statement, inclusive. Include all pertinent information:
Multiple libraries were found for "WiFi.h"
In file included from C:\Desktop\New folder (3)\sketch_oct13a\sketch_oct13a\sketch_oct13a.ino:5:0:

 Used: C:\Users\rp787\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
D:\libraries\NtpClientLib-3.0.2-beta\src/NtpClientLib.h:56:21: fatal error: TimeLib.h: No such file or directory

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
compilation terminated.

Using library WiFi at version 1.0 in folder: C:\Users\rp787\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi 
Using library ESPmDNS at version 1.0 in folder: C:\Users\rp787\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\ESPmDNS 
Using library NtpClientLib-3.0.2-beta at version 3.0.2-beta in folder: D:\libraries\NtpClientLib-3.0.2-beta 
exit status 1
Error compiling for board ESP32 Dev Module.

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

atanisoft, 2019-10-13:
It appears you have an issue with a user installed library called "NtpClient" which is not maintained/supported by this repository. With that said, it appears you don't have the correct dependencies for that library installed. You should open an issue with the correct maintainer for that library or follow the readme/wiki for that library to install the correct dependencies.

lbernstone, 2019-10-13:
NTP is available without any additional libraries.
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Time/SimpleTime/SimpleTime.ino

closed 2019-12-26 by stale[bot]