espressif/arduino-esp32 #3355: ESP32-CAM bad hostname  (IDFGH-1922)

rserroyen, 2019-10-14:
Hello
When I tried this code

`
#include "esp_camera.h"
#include <WiFi.h> 

#include "camera_pins.h" 
const char* ssid = "xxxxxxxxxx";
const char* password = "xxxxxx";
const char* hostname = "Sillage";

// Set your Static IP address
IPAddress local_IP(192, 168, 1, 208);
// Set your Gateway IP address
IPAddress gateway(192, 168, 1, 1);

IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(192, 168, 1, 1);   //optional
IPAddress secondaryDNS(192, 168, 1, 1);   //optional

void setup() {
........

WiFi.disconnect(); //Prevent connecting to wifi based on previous configuration
  
  WiFi.setHostname(hostname);
  
// Configures static IP address
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
  }
  
  WiFi.begin(ssid, password);
  ...........```
`
I have IP address 192.168.1.208 but the hostname as 192.168.1.208

When i tried with a dynamic hostname is wake Sillage

What is the problem

Thank you so much

me-no-dev, 2019-10-14:
let's discuss this here :) it's not an IDF question.

Please clarify what exactly is happening and what you expect to happen. I was not able to understand your question.

rserroyen, 2019-10-15:
thank you for spending time for me

the problem is and simple, when i run the code with a Static IP address, the hostname as : "192.168.1.208" and not the hostname "Sillage"

because of my avc I can not explain

thank you for your help

me-no-dev, 2019-10-15:
See [this](https://github.com/espressif/arduino-esp32/blob/a59eafbc9dfa3ce818c110f996eebf68d755be24/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino#L77-L80) example :) You need to attach to the events and set it at the proper time :)

me-no-dev, 2019-10-15:
I edited your sketch a bit up there :) check it out, it should now work.

closed 2019-12-28 by stale[bot]