espressif/arduino-esp32 #3359: Please add support for TTGO T-Display

DaQue, 2019-10-15:
Getting the TTGO  T-Display demo to display anything is problematic. Compiling and uploading seems to work but display does not show anything and the back light is on.

I thought this might work but the result is the same is trying to do it manually in Arduino IDE.
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TTGO-T-Display.ino.

robcazzaro, 2019-10-21:
I have used the TTGO display following the instructions on the page you link, and it works without problems. You can try the example on this repository and see if it works for you https://github.com/robcazzaro/TetrisTFTAnimation. The only trick is to copy the file TTGO_T_Display.h into the TFT_eSPI/User_Setups/ directory  and adding #include <User_Setups/TTGO_T_Display.h> to the TFT_eSPI/User_Setup_Select.h file

DaQue, 2019-10-21:
Thank you I will try after work.

closed 2020-01-03 by stale[bot]

KyleMicrocare, 2020-02-27:
please reopen

> [STALE_DEL] This stale issue has been automatically closed. Thank you for your contributions.