espressif/arduino-esp32 #3361: Static ip and dns

anITiot, 2019-10-15:
I must use the order as shown "**WiFi.config(staticip, dns, gateway, subnet)**;" to have static ip and dns  but I can't use dns2 in the WiFi.config.

IPAddress staticip(192, 168, 1, 61); //ESP static ip
IPAddress gateway(192, 168, 1, 1);   //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0);  //Subnet mask
IPAddress dns(8, 8, 8, 8);  //DNS
IPAddress dns2(8, 8, 4, 4);  //DNS2

And the manual says WiFi.config(staticip, gateway, subnet, dns, dns2) but this dosent work for me, it gives me the DHCPaddress but if I discard the DNSaddresses in WiFiconfig.h it takes the static ip but I will not have any dns.

ESP8266 latest and Arduino 1.8.10

Suggestion:
And instead of add subnet mask is it possible to enter 192.168.1.61/24?

lbernstone, 2019-10-15:
esp32 != esp8266
https://github.com/esp8266/Arduino/issues

anITiot, 2019-10-15:
Sorry !! I will open in the ESP8266

closed 2019-10-15 by anITiot