espressif/arduino-esp32 #3362: Wifi disconnects after some time, reconnection fails

fpiernas, 2019-10-15:
### Hardware:
Board:							Wemos D1 R32
Core Installation version:			Don't know
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I have a Wemos D1 R32 to open the garage door. It has been working perfectly 24/7 for over a year and it's still working fine.

I bought another Wemos D1 R32 for a project with some relays to turn lights on and off, and this one is giving me problems. 

It's connected to the same router as the other board, using a static IP. On boot, it connects and works fine, I can tell this by doing a 1000 packets Ping, no packets lost. But after some hours, packets start to get lost, more and more, until the code detects the wifi connection is lost and tries to reconnect, but fails. 

The interesting thing is that the connection and reconnection code is the same I'm using in the first board that works perfect, but somehow it's failing on the second board.

I also noticed that my router says that the board is connected but not associated nor authorized. The wifi key is a 20 character alfanumeric WPA2 password. The board that works is shown as asociated and authorized.

imgur.com/a/02H8xl6

### Sketch: 
(This is not the full sketch since it's very long, with UDP, rotary encoder and OLED codes, I will post what I think is giving the problems)
```cpp
#include "WiFi.h"

unsigned int reconnect_count=0;
void setup() {
    IPAddress local_IP(192, 168, 1, 9);
    IPAddress gateway(192, 168, 1, 1);
    IPAddress subnet(255, 255, 255, 0);
    IPAddress primaryDNS(8, 8, 8, 8);   //optional
    IPAddress secondaryDNS(1, 1, 1, 1); //optional
    if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
      Serial.println("STA Failed to configure");
    }
    WiFi.begin(ssid, password);
    while(1) {
        delay(5000);
        WiFi.begin(ssid, password);
        if(WiFi.waitForConnectResult() == WL_CONNECTED) break;
        else Serial.println("WiFi Failed");
    }
    //Do other stuff
}

void loop() {

    delay(500);
    if(millis()-reconnect_count >= 120000)
    {
           if(WiFi.waitForConnectResult() != WL_CONNECTED)
           {
                 while(1) 
                 {
                      delay(5000);
                      WiFi.begin(ssid, password);
                      if(WiFi.waitForConnectResult() == WL_CONNECTED) break;
                      else Serial.println("WiFi Failed");
                  }
            }
           reconnect_count=millis();
    }
    //Do other stuff
}
```

lnrobby, 2019-10-20:
I am having a similar problem.  I am testing by changing the ssid on my router and it fails to reconnect after leaving for a good while (over 10 hours maybe.) The app is sending data from a queue to an MQTT server. It is running on a Lolin D32 and is repeatable on more than one device. The app is built in Arduino IDE.

I have had the problem on version V1.0.4 and before this version V1.0.2 (I upgraded to see if it went away).
This is what I get as output (set as Core Debug Level Verbose)

11:52:20.244 -> Attempting to reconnect in reconnect code..- Items in Queue 
11:52:20.244 -> 20
11:52:20.244 -> A4:CF:12:12:65:B4
11:52:20.244 -> [V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 253560
11:52:20.244 -> [V][ssl_client.cpp:58] start_ssl_client(): Starting socket
11:52:24.253 -> [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for m24.cloudmqtt.com
11:52:24.253 -> [E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1
11:52:24.253 -> [V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
11:52:25.244 -> Attempting to reconnect in reconnect code..- Items in Queue 20

Pressing the reset button works and the device starts correctly.

maxdd, 2019-11-03:
Well known issue on Arduino. The only solution as of today is to move to esp-idf

fpiernas, 2019-11-03:
> 
> 
> Well known issue on Arduino. The only solution as of today is to move to esp-idf

I forgot about this thread. The issue kind of self solved somehow and the Wemos D1 R32 with its ESP32 now works fine and reconnections work after a reset or blackout.

I still don't know what the problem was. The distance to the router is the same, I also erased the flash before updating sketches as usual, but it self solved.

fpiernas, 2019-11-11:
So I changed the sketch and now I call this function to connect on boot and reconnect on connection lost:
`void recon()
{
  long int a=millis();
  WiFi.disconnect(true);
  WiFi.mode(WIFI_STA);
  IPAddress local_IP(192, 168, 1, 10);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress primaryDNS(8, 8, 8, 8);   //optional
  IPAddress secondaryDNS(1, 1, 1, 1); //optional
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)){
    Serial.println("Error WIFI.");
    delay(1000);
  }
  while(true) {
      a=millis();
      WiFi.begin(ssid, password);
      delay(10000);
      if(WiFi.status() == WL_CONNECTED) break;  
      else Serial.println("WiFi Failed");
  }
  return;
}`

Fun thing about it, the ESP32 is now completely unable to connect. It attempts reconnections when status returns not connected, but it won't connect. There are moments where that function finishes because it has managed to reconnect, only that it hasn't. On my router page I can see that it's not connected.

Some of you may say the problem is the router, but as I said in the first post, there is another ESP32 connected to that router and it has been working great for more than a year.

These last weeks I have been reading a lot about this issue. I don't know what is failing on this ESP32 that it's not on the other. Is it the rotary enconder library? The pin14 I'm using? (I read somewhere that pin14 cannot be used for ADC if wifi's on, but I'm using it as digital output.

I also doubt that's the issue because this ESP32 could stay working for several days and after each power cycle it got worse.

lnrobby, 2019-11-11:
> Well known issue on Arduino.

Hi maxdd, can you explain a bit more, I have searched and can't find much about it being a well known issue.
Thanks

maxdd, 2019-11-11:
Hello Inrobby, you just can have a look at the open issue by tiping "reconnection"

https://github.com/espressif/arduino-esp32/issues/2127
https://github.com/espressif/arduino-esp32/issues/1464
https://github.com/espressif/arduino-esp32/issues/653
https://github.com/espressif/arduino-esp32/issues/307

The underline problem here is that somehow the WiFi library is not there yet for esp32.
The only working solution i've found to make it works (and believe i've tried a lot) is to make use of the following code

```
static void configureWiFi()
{
    WiFi.persistent(false);
    WiFi.disconnect(true);
    WiFi.mode(WIFI_OFF);
    WiFi.mode(WIFI_STA);
    WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
 
    WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info) {
        Serial.print("WiFi connected. IP: ");
        Serial.println(IPAddress(info.got_ip.ip_info.ip.addr));
        WiFiConnected = true;
    },
        WiFiEvent_t::SYSTEM_EVENT_STA_GOT_IP);

    WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info) {
        Serial.print("WiFi lost connection. Reason: ");
        Serial.println(info.disconnected.reason);
        WiFi.persistent(false);
        WiFi.disconnect(true);
        //WiFiConnected = false;
        ESP.restart();
    },
        WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
}
```

In my case, ever since i used this, i never dropped once

lnrobby, 2019-11-11:
Maxdd

Thanks for this. I will try this. Shame no-one is picking up the actual problem to fix it.

fpiernas, 2019-11-11:
Hi again, I have activated the debug level verbose and I see nothing but lots of AUTH_EXPIRE callbacks, I don't know if this helps to identify the problem.

bsayiner, 2019-11-20:
Does this issue be related to the ADC2 channel? Because Wifi drivers use ADC2 pins

fpiernas, 2019-11-20:
> 
> 
> Does this issue be related to the ADC2 channel? Because Wifi drivers use ADC2 pins

But does that mean you can't use those even as digital outputs?

bsayiner, 2019-11-20:
> > Does this issue be related to the ADC2 channel? Because Wifi drivers use ADC2 pins
> 
> But does that mean you can't use those even as digital outputs?

I saw that at this thread #1829

closed 2020-02-02 by stale[bot]

deepakjos2000, 2021-02-25:
I faced similar problem, 
Saw all the posts, my problem is that my wifi.staus() gets 0 after 2 hrs and i tried all the disconnecting loops etc, like get the wifi.status() = 0 then wifi.disconnect(). Though I am not a core programmer, I was afraid to peep into the wifi libraries.
However, I enabled the debug mode and saw that there are events only returning 4 and 7 even after disconnect, that means the device was not truly disconnecting the wifi.
So I decided to look into the library src codes and found something strange written in WifiSta.cpp - line 245:
bool WiFiSTAClass::disconnect(bool wifioff, bool eraseap)
{
    wifi_config_t conf;

    if(WiFi.getMode() & WIFI_MODE_STA){
        if(eraseap){
            memset(&conf, 0, sizeof(wifi_config_t));
            if(esp_wifi_set_config(WIFI_IF_STA, &conf)){
                log_e("clear config failed!");
            }
        }
        if(esp_wifi_disconnect()){
            log_e("disconnect failed!");
            return false;
        }
        if(wifioff) {
             return WiFi.enableSTA(false);
        }
        return true;
    }

    return false;
}

That means it needs wifi.disconnect (bool , bool).
So I wrote wifi.disconnect(1,1) in the event check loop like this:
 if(wifi_status_watch != WiFi.status()){Serial.print("WiFi.status(): ");Serial.println(WiFi.status());wifi_status_watch = WiFi.status();}
        if (((WiFi.status() == 0)||(WiFi.status() == 5)) && (millis() > check_wifi && check_wifi_disconnect == 0)) 
        {
          Serial.println("disconnecting wifi...");
          WiFi.disconnect(1,1);
         //disconnect(bool wifioff, bool eraseap)
          check_wifi_disconnect = 1;
          wifi_disconnect_time = 0;
          check_wifi = millis() + 30000;
          Serial.print("WiFi.status(): ");Serial.println(WiFi.status());wifi_status_watch = WiFi.status();
        }
        if (((WiFi.status() == 0)||(WiFi.status() == 6))&& (wifi_disconnect_time >= 10) && (check_wifi_disconnect == 1)) 
        {
          Serial.println("Reconnecting to WiFi...");
        
          check_wifi_disconnect = 0;
          
          WiFi.begin(Wifi_setting.ssid, Wifi_setting.password); 
          
        }
so when the wifi.status() == 0 (or wifi.status() == 5, just to check right now if truly the device disconnects or not) when i shut down router, the status goes to == 5 )
Then first the wifi.status return 255 (means no wifi shield, I guess here it clears all the difficult bits)
The it goes to status == 1, then scans, like this i get almost all the required status.

I am still not sure if it will work like this even after 2 hours or not, but I am hoping that with all this clear status modes there has to be no reason for device to connect properly.
Will post the results.

NOUSHADO, 2021-03-24:
hi sir  how are you
i am facing the issue like wifi disconnects after 1 or 2 hours

can you help me for this
i am not much familier with c+c+  language
can you suggest me how to solve this issue

skarpushin, 2023-11-16:
Happening to me too. 
Board ESP32-D0WD-V3 (revision v3.0).
After ~5h of working disconnects and cannot connect anymore.

voidzero, 2024-07-06:
Same problem here as long as I've got my 433 MHz receiver connected on GPIO 26 (I have a Wemos D1 R32). As soon as I disconnect the pin: WiFi works.

parzibyte, 2025-11-02:
Hi. Maybe it's a little obvious but I tried every solution listed here and at the end it was solved by restarting the router. I am connected through a repeater, so I restarted both the main router and the repeater, then everything worked well:
[  1090][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED

By the way, this is the code I am using, but the esp32 only connected after tried this code and restarting the router. I am using ESP32 S3 WROOM N16R8 with camera (maybe it's a clone because it does not have any psram).

```cpp
    WiFi.mode(WIFI_STA);
    WiFi.disconnect(true, true);
    delay(100);
    wl_status_t resultado = WiFi.begin("ssid", "password");
    Serial.printf("Result code %d", resultado);
    WiFi.setTxPower(WIFI_POWER_8_5dBm);

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(300);
        Serial.print(".");
    }
```

My **platformio.ini** looks like this:

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32s3]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
board_build.flash_size = 16MB
board_build.psram = enabled
lib_deps = 
	esp32async/AsyncTCP@^3.3.6
	esp32async/ESPAsyncWebServer@^3.7.2
	;witnessmenow/UniversalTelegramBot@^1.3.0

build_flags = -DCORE_DEBUG_LEVEL=5 -DDEBUG_ESP_SSL

```