espressif/arduino-esp32 #3367: Getting BLERemoteService from BLERemoteCharacteristic

coogle, 2019-10-15:
Sorry for not following the standard bug report template here..

What I'm trying to do is use an ESP32 to connect/get some notification data / disconnect / go on to next device for N number of identical BLE servers and I'm having problems because of accessors in the API being set to private. 

What I want to do is this:

1) Connect to a BLE Server, register for notifications for a characteristic provided
2) Listen to the characteristic notifications (it's a data stream) until I've captured the data I need
2a) Parse out the relevant data from the capture, store in a struct for that device ID
3) Disconnect from the BLE Server
4) Connect to the next, identical BLE server, and repeat.

The problem I'm running into is that the `BLERemoteCharacteristic::registerForNotify` callback is unnecessarily limited because not only does it not make room to pass in any sort of context `void *`, but what it does pass in doesn't let you figure it out which device this notification came from.

One easy solution here would be to make `BLERemoteCharacteristic::getRemoteService()` `public` instead of `private`. This way, from within the callback you could call `characteristic->getRemoteService()->getClient()->getPeerAddress()` and get the address of the device sending the notification. 

At the end of the day what I'd like to do is be able to tell from within a characteristic callback what device sent the notification.

coogle, 2019-10-16:
Note on this bug the real issue is that when you call `registerForNotify()` in `BLERemoteCharacteristic` you can't pass any context variables in. This could also be solved by allowing for a `void *` context variable to be passed.

wakwak-koba, 2019-10-17:
try my repo 
https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE

example
https://github.com/wakwak-koba/arduino-esp32/blob/master/libraries/BLE/examples/BLE_clients_catch_notifies/BLE_clients_catch_notifies.ino

coogle, 2019-10-17:
I was able to fix this by first updating the library I was using to the most recent build from this repo (the ESP bundled one is out of date) and making some meaningful changes to the way I tried to jump from BLE device to BLE device. I'll update the repo I pointed to in this ticket with my new code.

closed 2019-10-17 by coogle

kind3r, 2021-01-20:
I'm having the same issue, trying to build a BLE gateway, any strategic reason I'm missing as to why `BLERemoteCharacteristic::getRemoteService()` is private ?
Maybe @me-no-dev or @chegewara can shed some light, if not I can try my luck with a PR.