espressif/arduino-esp32 #3369: esp_wifi_stop crashes if wifi isn't initialised

dirk1492, 2019-10-16:
### Hardware:
Board:							M5Stack
Core Installation version:			1.0.3 and 1.0.4
IDE name:						Arduino IDE 1.8.9 and Platform.io 1.9.0
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
esp_wifi_stop should return ESP_ERR_WIFI_NOT_INIT if WiFi is not initialized by esp_wifi_init.
Beginning with version 1.03 it throws an exception and causes a reboot.
Version 1.02 is working correctly.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d7ee8  PS      : 0x00060f30  A0      : 0x800d0c37  A1      : 0x3ffb1f50  
A2      : 0x00000000  A3      : 0x3ffbf684  A4      : 0x0800001c  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d0da9  A9      : 0x3ffb1f00  
A10     : 0x00000014  A11     : 0x00000001  A12     : 0x3ffbebe8  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000134  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d7ee8:0x3ffb1f50 0x400d0c34:0x3ffb1f80 0x400d1c0f:0x3ffb1fb0 0x40088945:0x3ffb1fd0

PC: 0x400d7ee8
EXCVADDR: 0x00000134

Decoding stack results
0x400d0c34: setup() at /home/dil/Arduino/Power/Power.ino line 5
0x400d1c0f: loopTask(void*) at /home/dil/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x40088945: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

### Sketch:
```cpp
#include <esp_wifi.h>

void setup(){
  Serial.begin(115200);
  esp_wifi_stop();
}

void loop() {
  Serial.println("ok.");
}

me-no-dev, 2019-10-16:
cool, but that is esp-idf api and you should complain there :)

closed 2019-10-16 by me-no-dev

dirk1492, 2019-10-17:
The call ist working correctly in the current esp-idf version. 
I can't complain about working functions.

me-no-dev, 2019-10-17:
it is still an issue you have with IDF api in particular IDF version that we track (currently 3.2). If it gets fixed in IDF, it will land here as well. I can not change the behaviour of the function from within Arduino. 3.2 is stable and whatever you call current in idf is not :) Makes sense?

me-no-dev, 2019-10-17:
if you use this repo through GIT, you can switch to the `idf-release/v3.3` branch and see if you get what you expected. We are soon to move to 3.3