espressif/arduino-esp32 #3371: ESP32 entering into boot mode randomly

yogeshwaran774, 2019-10-16:
I am using ESP-32 WROOM in my project and some strange thing is happening when device restarts. Esp enters into boot mode sometime when device restarts and my gpio 0 is floating in my board. Basically most of the times it is not entering into the boot mode but if it enters into boot programming mode the voltage across gpio 0 and GND is about 400-500mV.

I assumed that because of floating voltage and i did pull up to 3.3v but actually this pin is pulled up when booting then also problem persist, the voltage drop across the resistor is about 1.5V. I don't know why the drop across the resistor is 1.5v since i connected to 3.3V but then also esp is going into power boot programming mode.

Gpio 12 which is used to select the LDO voltage is floating in my pcb and i don,t this will cause an issue because default state is '0' for 3.3V which is related to flash VDD_SDIO.

The only solution i found was shorting **gpio 0 to 3.3V** then esp is not subjecting any issue.
I have following questions.
1) why esp32 is going to  boot randomly even though internal pull is there in the chip?

2)Is any issue with floating of GPIO 12?

3)why it is working when i connect to 3.3v and why not if do pullup to 3.3v at measured voltage across the resistor when entering boot was 1.5 - 1.7V?

lbernstone, 2019-10-16:
Are you positive it is entering boot mode- you see the "waiting for download" message on serial?  The esp32 has an internal pull up on gpio0 ([pinmap](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap)), so if it is going low at boot, or even 1V5, something else is affecting the pin.
For GPIO12, look [here](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12)

closed 2019-12-29 by stale[bot]

vishnumaiea, 2021-12-07:
I had the same issue where an ESP32-WROOM-32E on my custom board would enter into boot mode (shows "waiting for download" on serial) randomly. I was using CH340G and the auto-programming circuit. The auto-programming was working fine, but my application code was not running. The GPIO0 and GPIO12 were floating on my design and EN pin had 10K pull-up and 100nF capacitance. The cap was placed little further from the EN pin.

Why it happens because the pins states matches the boot sequence somehow randomly due to stray effects in PCB design and assembly. The GPIO0 pins goes low for short time during reset of power on. I solved the issue by increasing the capacitance to 1uF. This causes the boot sequence to be out of phase and thus prevents unintentional boot mode.

vikramnain20, 2021-12-15:
I have same issue with my custom board.
I have also changed the 10k & 100nf for RC circuit on EN pin. 47uf for AMS1117 3.3v converter. 
I found the waiting for download randomly. And it has to keep it without power for few hours then connect again and starts working.
Iam using standard schematics for V4 module and have take care of all of points mentioned on Expressif website. 
I dont use usb & auto reset circuit. 
And one more strange issue that when it is powered theough 5v and nothing is connected then whole board and all pins have 3.27v except the IO0 pin & Uart2 - TX pin which has 0.40 or 1.20v. 
If then uart0 with gnd is connected to board to upload the code then it uploads and reset perfectly. Voltage at IO0 & uart2- TX pins  measures 3.18v. 

But if again removed the uart0 then it shows low voltage on both the pins.
Can anyone please suggest on this?
Nothing is connected to es32 module.
Whole board has 3.27v on all expected pins except the 2 explained above.
Tried to connect 10k resistor on 3.3v & IO0 but it shows 1.75v or 2.25v on the resistor. And sometime works.

Dominikzx, 2025-04-08:
I had a similar problem with my board - esp32-S3 randomly entered boot mode after a power reset. Then pin 0 was in LOW state, even when nothing was connected to it so the pull-up resistor should work (but didnt). I also had a MAX809 voltage supervisor circuit connected to the EN pin. To my surprise, the solution was to remove this circuit.

This was the first website I was directed to, so maybe this information will help someone.

ISbitPro, 2025-07-02:
I have same problem with my custom board for esp32-s3-wroom
I also suspect the EN pin is enabled before DTR pin. I use 1uF and 10k resistor for the EN 
Should I increase the capacitor to slow the boot?