espressif/arduino-esp32 #3373: [BLE+WiFI] reboot when WiFiClient report errno: 11, "No more processes"

ZorkHuang, 2019-10-17:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			ESP32:v3.2.3
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
when WiFI and Ble write data per senconds,  the wificlient report error:
  [E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
  [E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
after 7 seconds, wifi lost connect and system reboot:
  Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

### Debug Messages:
DEBUG,t:174605.59,m:30.49,Multi BLE write;
DEBUG,t:174605.60,m:31.06,Multi BLE notify;
DEBUG,t:174605.61,m:31.84,Mqtt task report states:{"K1":0,"K2":0,"K3":0,"K4":0,"K5":0,"K6":0},len:43;
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
DEBUG,t:174606.03,m:30.43,update switches:
    K1[32]=>0-->1
    K2[15]=>0-->1
    K3[2]=>0-->1
    K4[12]=>0-->1
    K5[13]=>0-->1
    K6[16]=>0-->1
DEBUG,t:DEBUG,t:174606.09,m:174606.093,m:0.36,Multi BLE write;
30.36,Multi BLE notify;
DEBUG,t:174606.10,m:31.84,Mqtt task report states:{"K1":1,"K2":1,"K3":1,"K4":1,"K5":1,"K6":1},len:43;
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
.....
DEBUG,t:174611.05,m:DEBUG,t:30.12,Multi BLE write;
174611.05,m:30.36,Multi BLE notify;
DEBUG,t:174611.06,m:31.61,Mqtt task report states:{"K1":0,"K2":0,"K3":0,"K4":0,"K5":0,"K6":0},len:43;
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 57, errno: 11, "No more processes"
,result:0
DEBUG,t:174611.43,m:37.38,WiFi disconnection;
DEBUG,t:174611.52,m:35.98,update switches:
    K1[32]=>0-->1
    K2[15]=>0-->1
    K3[2]=>0-->1
    K4[12]=>0-->1
    K5[13]=>0-->1
    K6[16]=>0-->1
DEBUG,t:174611.53,m:35.DEBUG,t:91,Multi BLE write;
174611.53,m:36.10,Multi BLE notify;
DEBUG,t:174612.43,m:36.63,WiFi connected,IP:192.168.1.127
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401d3b92  PS      : 0x00060930  A0      : 0x800d3548  A1      : 0x3ffce4e0  
A2      : 0x00000000  A3      : 0x3fffcc78  A4      : 0x00000050  A5      : 0x00000070  
A6      : 0xa6fff273  A7      : 0x00000028  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x3fffcb28  A11     : 0x3ffba54c  A12     : 0x00000050  A13     : 0x00000001  
A14     : 0xa6fff273  A15     : 0x00000028  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

Backtrace: 0x401d3b92:0x3ffce4e0 0x400d3545:0x3ffce500 0x400d3561:0x3ffce520 0x400d65c0:0x3ffce540 0x400d38c9:0x3ffce560 0x400d6d15:0x3ffce580 0x400d23b8:0x3ffce5a0 0x400e3c04:0x3ffce5c0 0x4008e009:0x3ffce5e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5800
entry 0x4008069c

reboot:
DEBUG,t:0.04,m:228.47,[CONTACTOR_V1.1.20191011.1],ID:E4927D12CFA4,ESP32:v3.2.3,cfg:352B;

me-no-dev, 2019-10-17:
can you show us the code that produced this error?

ZorkHuang, 2019-10-18:
Please send me your email. I will send you all of them.

n9wxu, 2019-12-10:
I am also seeing something similar.
Every 10 seconds a HTTP POST is executed and the response inspected.
After a few minutes I get a read failure and no more processes as described above.

closed 2020-02-22 by stale[bot]

brad-colbert, 2021-07-07:
I have the same problem.  Streaming music via BT to the device and then I access a hosted web server.

rohith-kazam, 2023-03-29:
Iam facing the same issue, connecting to mqtt server