espressif/arduino-esp32 #3375: Not working VSPI after deep sleep and wakeup.

miracle0419, 2019-10-17:
I trying the VSPI function fallowing step.
Work fine that Power on and SPI access. but, the problem is Power on deep sleep and wakeup after VSPI access.

Thanks

miracle0419, 2019-10-18:
stickbreaker, could you please let me know that how can i peripheral reset.

stickbreaker, 2019-10-18:
What ever you have connected to the ESP32 needs to be rest.

When the ESP32 wakes up from deep sleep all of it's internal peripherals are reset and configured with default values.  

I think whatever you have connected is not being properly reconfigured on wake up.  Either it is not shutdown properly before the ESP32 goes to sleep, or the floating SPI bus is causing the external device problems.

* How are you resetting your external device?
* How are you powering the external device during deep sleep?
* Are you power cycling the external device after wake up?
* How are you controlling the SS signal?  Do you have a pullup to vdd?

Chuck.

closed 2019-10-29 by miracle0419