espressif/arduino-esp32 #3381: 4WAY_HANDSHAKE_TIMEOUT

slowsquirrel69, 2019-10-18:
When I load a sketch that does a network scan I see my network

========================
WiFi scan start
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
WiFi Scan done
29 networks found
1: ARLO_VMB_1792897479 (-51)*
2: CAKE-2G (-53)*
========================

When I load a sketch to connect to my network I get

=======================
Attempting to connect to WPA SSID: CAKE-2G
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT

All other network devices can connect iPhone, Android, laptops, etc..

Why can't the Adafruit HUZZAH32 – ESP32 Feather Board connect?

TheNetStriker, 2019-11-22:
@slowsquirrel69 I have the exact same problem with the HUZZAH32, but my code worked before. I guess the problem occurs because I updated to the latest platform version. Did you find a solution for the problem?

TheNetStriker, 2019-11-22:
@slowsquirrel69 I found out that my code works again if I use the platform version 1.9.0:

platform = espressif32@1.9.0

jayrawatrj, 2020-02-01:
i am getting this error.. please help me if anyone have the solution for it.. thank you

closed 2020-04-15 by stale[bot]

happyme531, 2020-07-11:
+1 same problem at version 1.0.4 on wemos d1 r32

HamzaHajeir, 2020-09-15:
+1 same @ Version 1.0.4 , DevkitV1

cagney, 2020-11-07:
try 1.0.2, see #3225

thomai-d, 2021-05-26:
Please try this to connect:

```
WiFi.mode(WIFI_STA);
WiFi.begin("...", "...");
WiFi.persistent(false);
WiFi.setAutoConnect(false);
WiFi.setAutoReconnect(true);
WiFi.setTxPower(WIFI_POWER_2dBm);
```

The important part is the `setTxPower` at the end.

Related: https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-846588070 and https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-849013717

mike-ady, 2021-12-30:
@thomai-d...  Thanks for posting.   I recently upgraded from Espressif core 1.0.4 to 2.0.2.  I have a TTGO TWatch 2020, which is essentially a ESP32-WROVER-32 and I also have a few ESP32-WROOM-32 Devkit boards.   The TWatch used to work.  After loading with the new core, I have been unable to connect the TWatch to an access point.  The Devkit boards were still working fine.  Your fix seems to have solved my problem.  (Verified on both the TWatch and the Devkits.)

frickc3, 2022-10-19:
I just found this post and the "WiFi.setTxPower(WIFI_POWER_2dBm);" the issue I was having with a Heltec Dev Kit 32 connecting to Wifi.  Thanks to thomai-d for posting!

newace150, 2024-01-15:
In my case it was a wrong password with fancy chars (I am getting credentials via serial) connecting to the right ssid, debug what you're connecting to.