espressif/arduino-esp32 #3383: Primary I2C fail on WROVER-B on custom board

GiovanniClimacoSilitronic, 2019-10-18:
Hardware: Custom board based on ESP32-WROVER-B
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: ?yes?
Upload Speed: 921600
Computer OS: Windows 10

I just completed developing my first board using ESP32, everything is running correctly except the I2C.

The board is using first I2C channel 
Wire.begin(21, 22); 
on this pins are provided the required pull-up.

I'm experiencing a strange issue, once configured the ESP32 once the code reach and execute the "Wire.endTransmission();" istruction, I have the ESP32 continuosly producing a square waveform on both SDA and SCL as you can see here.

[image]

The code that I'm using to debug the issue is the following

[Test_I2C.ino.txt](https://github.com/espressif/arduino-esp32/files/3743375/Test_I2C.ino.txt)

While the log taken from the device core debug is the following 

[SchedaMia.txt](https://github.com/espressif/arduino-esp32/files/3743370/SchedaMia.txt)

This log file if recording the communication with an existing device on the boad, but the issue is happening also if the device is non existent, so the problem seems to be ESP32 related.

As far as I can see there are no errors given by the I2C HAL, is there any issue related to the Core or maybe there is a problem in my board ?
Does anyone know more about this issue ?

asetyde, 2019-10-18:
This problem comes from many times Giovanni, I hope it's corrected .. but I not understand if come from idf

GiovanniClimacoSilitronic, 2019-10-18:
It's a strange problem because the I2C is running outside its protocol specification, there is no communication attempt in the signals that I see on the oscilloscope.
Hope that someone will solve this issue or give us more information.

atanisoft, 2019-10-18:
@lbernstone might be able to assist, but I'm sure he will also ask the following:
1. schematic of your custom board
2. what pull-ups are on SDA and SCL
3. what devices are on the I2C bus
4. why your sketch has zero error checking...

Each of the Wire APIs you are invoking (except beginTransmission) return a status code of sorts. From the debug logs you can see:
```
[I][esp32-hal-i2c.c:437] i2cTriggerDumps(): after ProcQueue
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbebe8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb81cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8250 bits=10
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8280
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb822c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0xFFFFFFFF
```
Which indicates that before your call to Wire.write(0x10) there was an error condition on the bus (error=1). Based on this it would appear that the I2C bus was not initialized correctly or failed to reset correctly. @lbernstone would know more.

stickbreaker, 2019-10-18:
@GiovanniClimacoSilitronic Check your wiring.  It looks like you have SCL and SDA connected together.

from your log:
>[I][esp32-hal-i2c.c:437] i2cTriggerDumps(): after ProcQueue
...
[I][esp32-hal-i2c.c:288] i2cDumpDqData(): [0] 7bit 78 W STOP buf@=0x3ffc0302, len=1, pos=1, ctrl=11111
[I][esp32-hal-i2c.c:306] i2cDumpDqData(): 0x0000: .                                10 

you are sending a Write to 0x78, data 0x10 with Stop.

>[I][esp32-hal-i2c.c:346] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [01]	0x0001	0x0002	0x0002	0x0000	0x0000068e
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [02]	0x0001	0x0200	0x0000	0x0000	0x0000068e
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [03]	0x0001	0x0080	0x0000	0x0000	0x00000693

interrupt sequence:
[01] : the i2c peripheral loaded two bytes into the txFifo at tick 0x68e,
[02]: the peripheral was started in master mode.
[03]: the peripheral saw a stop on the bus. which aborted the transaction at tick 0x693

The following is the command list the i2c peripheral was told to execute.
>[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 0]	Y	RSTART	val[0]	exp[0]	en[0]	bytes[0]
[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 1]	N	WRITE	val[0]	exp[0]	en[1]	bytes[1]
[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 2]	N	WRITE	val[0]	exp[0]	en[1]	bytes[1]
[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 3]	N	STOP	val[0]	exp[0]	en[0]	bytes[0]

[0]  RSTART was executed successfully,  The START signal.
[1] WRITE 1 byte, send the slave address did not execute.
[2] WRITE 1 byte, send data did not execute.
[3] STOP, did not execute.
 
Status register:
>[I][esp32-hal-i2c.c:385] i2cDumpStatus(): ack(0) sl_rw(0) to(0) arb(0) busy(0) sl(0) trans(0) rx(0) tx(2) sclMain(0) scl(5)

no ACK, Slave Read Mode, no Timeout error, no Arbitration error, not busy, not Slave Mode, 0 bytes in rxFifo, 2 Bytes in TxFifo(unsent slave address, data), peripheral Idle(0), last SCL TX state Low.

Chuck.

stickbreaker, 2019-10-18:
@GiovanniClimacoSilitronic The debug logging is explained in a [i2c_debugging.md](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/doc/i2c_debugging.md) in the docs  folder of the Wire library.

Chuck.

GiovanniClimacoSilitronic, 2019-10-18:
@stickbreaker Thanks for your information I'll check the board for short circuit on Monday, hope this solve the issue, I'll let you know.

lbernstone, 2019-10-19:
> @lbernstone would know more.
Nope, @stickbreaker is the authority on this.  I certainly have lots of experience with bad wiring, but I'm not sure all my burns will help.

atanisoft, 2019-10-19:
My mistake on tagging @lbernstone, I had meant @stickbreaker...

GiovanniClimacoSilitronic, 2019-10-21:
@stickbreaker I was checking my wiring, there was an almost invisible short under an eeprom that was creating the issue, now everything is running correctly.
Thanks for your help

closed 2019-10-21 by GiovanniClimacoSilitronic