espressif/arduino-esp32 #3384: Question: Sending data over TCP using Ethernet ?

GeorgeFlorian, 2019-10-18:
Hello !

**TL;DR:** I would like to know how to send data over TCP from ESP32 to a client.

I am planning to connect a Radar to an ESP32, take the output of the radar through serial connection and send it to a client that has the IP and Port of the ESP32. I found something about WiFiClient and WiFiServer being able to do that. The thing is that I need to use ETH, not WiFi. Also, is WiFiClient the only library that has TCP capabilities ?

closed 2019-10-23 by GeorgeFlorian

QE-Nicholas, 2023-06-13:
Hello, I met the same issue, did you found your solution? I think ETH.h is required. 

example:

/* 
   * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
   * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   16

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

// MAC address of Device 1
uint8_t mac[] = {0xDE, 0xAD, 0xBE, 0xEC, 0xFE, 0xEC};

// Select the IP address for Device 1
IPAddress localIP(192, 168, 1, 22); // IP address of Device 1

// Select the subnet mask for Device 1
IPAddress subnet(255, 255, 255, 0); // Subnet mask for Device 1

// Select the port number for communication
uint16_t port = 1111;

WiFiClient client;
WiFiServer server;
void setup()
{
  Serial.begin(115200);

  // Initialize Ethernet and set the MAC address
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
  ETH.macAddress(mac);

  // Set the local IP address, subnet mask, and start Ethernet connection
  ETH.config(localIP, IPAddress(), subnet);

  // Start the TCP server
  server.begin();
  Serial.print("TCP server started at ");
  Serial.print(localIP);
  Serial.print(":");
  Serial.println(port);
    

}

GeorgeFlorian, 2023-06-13:
This was forever ago. I don’t remember how or if I managed to fix it. 
I am sorry.

QE-Nicholas, 2023-06-14:
> This was forever ago. I don’t remember how or if I managed to fix it. I am sorry.

No worries mate, thanks!