espressif/arduino-esp32 #3385: Poor analog readings after ESP.restart();

joeltbennett, 2019-10-18:
**What you are trying to do?**

Read a hall sensor using 3.3v / ground / IO2 pins after restarting the board programmatically.

**Describe your system( Hardware, computer, O/S, core version, environment).**

The board is Wemos D1 R32, just an Arduino-type shield. I'm using the Arduino IDE for programming. The board is connected to a Nextion screen that controls all of the functions - most of the pins on the ESP32 board are being utilized by various components. Only one analog pin is being used -- GPIO 2 reads the signal from an [SS49E hall sensor](https://sensing.honeywell.com/SS49E-linear-and-angle-sensor-ics).

**Describe what is failing.**

The hall sensor works fine after the board has been turned off and on manually. When the board is flashed (HTTP connection sending over a compiled bin), a restart is triggered (ESP.restart) - when I navigate back to the hall sensor reading page on the Nextion, the signal is just reading as a single number. If the hall sensor is unplugged, the signal reading does not change.

Show the shortest possible code that will duplicate the error.

```
ESP.restart(); // restarts the Wemos R32
Serial1.print(F("rest")); // restarts the Nextion screen simultaneously (via serial command)

--------------
// the following is the hall sensor reading - it's sending updates to the Nextion screen
// it's displaying signals when the reading is outside of a certain envelope, but the hall sensor value is still displayed every time a reading is taken - it's fixing on a single number every time and does not correct itself after running a calibration (which sets the points for the envelope based on magnetic fields present). only fully restarting by cutting the power to the board corrects the analog pin readings.

void polaritySensor() {
  hall_sensor_value = analogRead(hallSensor);
  Serial.print("Hall sensor value = ");
  Serial.println(hall_sensor_value);

  // 3.3v = 1700 - 2020
  // 5v = 2300 - 2600
  //Lower
  int poleReader0 = polaritySets[0] + 1;
  //Higher
  int poleReader1 = polaritySets[1] - 1;

  if ((hall_sensor_value > polaritySets[0]) && (hall_sensor_value < polaritySets[1])) {

    Serial1.print(F("vis p3,0"));
    HMICommandEnd();
    Serial1.print(F("vis p4,0"));
    HMICommandEnd();
    Serial1.print(F("fill 213,46,435,200,8452"));
    HMICommandEnd();
    changeTxtField(0, "No poles detected");
    changeTxtField(1, "");
  }
  if (hall_sensor_value > poleReader1) {
    Serial1.print(F("vis p4,0"));
    HMICommandEnd();
    Serial1.print(F("vis p3,1"));
    HMICommandEnd();
    Serial1.print(F("ref p3"));
    HMICommandEnd();
    changeTxtField(0, "S | Strength:");
    int newHall = hall_sensor_value - poleReader1;
    changeTxtFieldInt(1, newHall);
  }
  if (hall_sensor_value < poleReader0) {
    Serial1.print(F("vis p3,0"));
    HMICommandEnd();
    Serial1.print(F("vis p4,1"));
    HMICommandEnd();
    Serial1.print(F("ref p4"));
    HMICommandEnd();
    changeTxtField(0, "N | Strength:");
    int newHall = abs(hall_sensor_value - poleReader0);
    changeTxtFieldInt(1, newHall);
  }
  delay(500);
};
```

joeltbennett, 2019-10-18:
After reading this thread, I moved the hall sensor read pin from 2 to 36 and it's perfect now. It works after an update and there's much less EM interference too. I suppose this can be closed, sorry to annoy.

closed 2019-10-18 by joeltbennett

atanisoft, 2019-10-18:
@joeltbennett All pins on ADC2 (those less than 30) will not be usable for analogRead() when WiFi is active. Bluetooth is also a problem in the current releases but there are changes coming in a future release which will change this.

joeltbennett, 2019-10-18:
Didn't realize this is a known thing...I'll have to google harder next time :|