espressif/arduino-esp32 #3386: Cannot use dual core multitasking to read I2C and send Bluetooth data at 1000Hz

robcazzaro, 2019-10-18:
### Hardware:
Board:							Chinese board based on ESP-WROOM 32
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.9
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10?

### Description:
I need to read two MAX30102 sensors, each on its one I2C bus, and send the results via Bluetooth serial (SPP) to a PC, at ~1000Hz

I tested separately that I can read the MAX sensors at 1000Hz fast enough, with plenty of free processor time. I could also send enough Bluetooth packets with the data I need to send.

When I try to read the sensor and send the data via Bth in a loop, though, the maximum rate I can achieve is ~400 Hz. What happens is that the processor waits for the I2C data, reads it, starts the Bth transfer, but Bth takes too long to return, blocking the next I2C read. There isn't enough processor cycles to run I2C and Bth serially

So I tried to use xTaskCreatePinnedToCore() to have the I2C code run on one core, the Bth code on another. I run into 2 separate problems:

- The FreeRTOS tick is 1ms, so if I add a vTaskDelay(1); in each of my tasks, I cannot achieve 1000Hz. Without the vTaskDelay(1);, after a few cycles the watchdog reboots the ESP32 if the task on Core 0 doesn't have a vTaskDelay(). I assumed that 2 tasks could run independently on 2 cores, but clearly there are other things going on in the background that need Core 0. I'm enclosing a stack trace of what happens when the task on Core 0 doesn't release processor time. I'm using the FreeRTOS example with vTaskDelay commented out
- Even assuming I could prevent the watchdog from triggering, I realized that probably all the esp-idf core libraries run on Core 0 (hence the watchdog problem when Core 0 doesn't relinquish control), so there's no way for me to run the I2C code on one core and the Bluetooth core on another. I can see that moving tasks between cores slightly improve the situation (I can send up to 600 Hz before the watchdog trigger), but not enough

Is there any way to do what I'm trying to do (i.e. read a relatively slow I2C and send Bth serial) at a 1000Hz with Arduino, or should I consider moving directly to esp-idf? I'm uaing quite a few Arduino libraries, so my preference would be not to migrate, if at all possible

thanks in advance for any help

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
8E (10166) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10166) task_wdt:  - IDLE0 (CPU 0)
E (10166) task_wdt: Tasks currently running:
E (10166) task_wdt: CPU 0: AnalogReadA3
E (10166) task_wdt: CPU 1: TaskBlink
E (10166) task_wdt: Aborting.
abort() was called at PC 0x400d3c33 on core 0

Backtrace: 0x4008b908:0x3ffbe170 0x4008bb35:0x3ffbe190 0x400d3c33:0x3ffbe1b0 0x40084995:0x3ffbe1d0 0x400d1c61:0x3ffb8c50 0x400d0f45:0x3ffb8c70 0x400d113a:0x3ffb8c90 0x400d1353:0x3ffb8cb0 0x400d13b7:0x3ffb8cd0 0x400d13ce:0x3ffb8cf0 0x400d0cb9:0x3ffb8d10 0x400885bd:0x3ffb8d30

Rebooting...

Decoding stack results
0x4008b908: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008bb35: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400d3c33: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x400d1c61: uartWriteBuf at C:\arduino-1.8.9\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-uart.c line 322
0x400d0f45: HardwareSerial::write(unsigned char const*, unsigned int) at C:\arduino-1.8.9\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\HardwareSerial.cpp line 147
0x400d113a: Print::write(char const*) at C:\arduino-1.8.9\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h line 66
0x400d1353: Print::printFloat(double, unsigned char) at C:\arduino-1.8.9\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 89
0x400d13b7: Print::print(double, int) at C:\arduino-1.8.9\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 139
0x400d13ce: Print::println(double, int) at C:\arduino-1.8.9\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 231
0x400d0cb9: TaskAnalogReadA3(void*) at C:\arduino-1.8.9\portable\sketchbook\FreeRTOS/FreeRTOS.ino line 105
0x400885bd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

stickbreaker, 2019-10-18:
@robcazzaro post your code for `TaskAnalogReadA3()`

As I understand your description, you have two MAX30102 sensors each connected to their own I2C bus.  You are using `Wire()` for one sensor and `Wire1()` for the other sensor.  You need to read each sensor at 1KHz, so a total of 2,000 readings per second. 

After reading the datasheet for that sensor(MAX30102) it seems the sensor has an internal 32 sample buffer I would code the sampling routine using the fifo, this would allow you to reduce the read frequency by a factor of 32 and still capture all the data.: (note this is untested and probably contains errors)
```c++
//sampling using fifo
/*

// External Wire.h equivalent error Codes from esp32_hal_i2c.h
typedef enum {
    I2C_ERROR_OK=0,
    I2C_ERROR_DEV,
    I2C_ERROR_ACK,
    I2C_ERROR_TIMEOUT,
    I2C_ERROR_BUS,
    I2C_ERROR_BUSY,
    I2C_ERROR_MEMORY,
    I2C_ERROR_CONTINUE,
    I2C_ERROR_NO_BEGIN
} i2c_err_t;

// from Wire.h
    i2c_err_t writeTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true);
    i2c_err_t readTransmission(uint16_t address, uint8_t* buff, uint16_t size, bool sendStop=true, uint32_t *readCount=NULL);

*/
#define SENSORADDR 0x57
#DEFINE FIFODEPTH 32
struct FIFO_BLOCK_t { // ints are used for modulus calculations
	int8_t wrPtr;
	int8_t ovfCnt;
	int8_t rdPtr;
	uint8_t data[FIFODEPTH*6];
} FIFO_BLOCK_t;

struct SAMPLE_t {
	uint32_t redSample;
	uint32_t irSample;
} SAMPLE_t;

stuct SAMPLEBLOCK_t {
	uint8_t count;
	SAMPLE_t samples[FIFODEPTH];
} SAMPLEBLOCK_t;

/* return a pointer to collected samples, caller responsible to free buffer
*/
SAMPLEBLOCK_t* getSample(){
uint8_t fifoWrPtr = 0x04; // register address for start of fifo block
FIFO_BLOCK_t fifo;
uint16_t sensorId = SENSORADDR;
i2c_err_t error = Wire.writeTransmission(sensorId,&fifoWrPtr,1,false); // setup register to start read
if( error == I2C_ERROR_OK) { 
	error = Wire.readTransmission(sensorId,&fifo,sizeof(fifo),true); // grab max possible samples
	if( error == I2C_ERROR_OK) { // now figure out howmany samples were in fifo

		int8_t count = (fifo.wrPtr - fifo.rdPtr) % FIFODEPTH; 
// something is nagging me about using % (modulus) on negative values, 
//  I vaguely remember some difference between the c++ implementation 
// and standard math modulus, this needs to be resolved. 

		if(count > 0){// actually have samples
			SAMPLEBLOCK_t *block = (SAMPLEBLOCK_t*)malloc( count*sizeof(SAMPLE_t)+1);
			if(!block){ // memory allocation failure
				Serial.printf("Memory Allocation Failure for sample on sensor 0x%02x\n",sensorId);
				return NULL;
			}
			else { // convert to 32bit ints
				block.count = count;
				uint8_t index = 0;
				for(uint8_t i = 0; i<count; i++){
					block->samples[i].redSample = (fifo.data[index++] & 0x3);
					block->samples[i].redSample = (block->samples[i].redSample << 8)+fifo.data[index++];
					block->samples[i].redSample = (block->samples[i].redSample << 8)+fifo.data[index++];
					block->samples[i].irSample = (fifo.data[index++] & 0x3);
					block->samples[i].irSample = (block->samples[i].irSample << 8)+fifo.data[index++];
					block->samples[i].irSample = (block->samples[i].irSample << 8)+fifo.data[index++];
				}
				// reset rdptr so that no samples are missed
				struct PTRRESET_t {
					uint8_t rdPtrReg;
					uint8_t rdPtr;
				} PRTRESET_t;
				PTRRESET_t rst;
				rst.rdPtrReg = fifoWrPtr + 2;
				rst.rdPtr = fifo.wrPtr;
				error = Wire.writeTransmission(sensorId,&rst,2,true);
				if(error != I2C_ERROR_OK){
					Serial.printf("failed resetting rdptr =%d, error=%d(%s)\n",rst.rdPtr,Wire.lastError(),Wire.getErrorText(Wire.lastError()));
				}
				return block;
			}
		}
		else { // no samples available, reset read pointer
			struct PTRRESET_t {
				uint8_t rdPtrReg;
				uint8_t rdPtr;
			} PRTRESET_t;
			PTRRESET_t rst;
			rst.rdPtrReg = fifoWrPtr + 2;
			rst.rdPtr = fifo.wrPtr;
			error = Wire.writeTransmission(sensorId,&rst,2,true);
			if(error != I2C_ERROR_OK){
				Serial.printf("failed resetting rdptr =%d, error=%d(%s)\n",rst.rdPtr,Wire.lastError(),Wire.getErrorText(Wire.lastError()));
			}
			return NULL;
		}
	}
	else { // error reading samples
		Serial.printf("failed reading samples, error=%d(%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
		return NULL;
	}
}
else { // failed setting up sample reading
	Serial.printf("failed accessing sensor 0x%02x, error=%d(%s)\n",sensorId,Wire.lastError(),Wire.getErrorText(Wire.lastError()));
	return NULL;
}
}
```

`Wire()` operations stall the task while the peripheral does the work, but the cpu core is switched to next task ready task. so, this code should not cause a wdt timeout.

Chuck.

robcazzaro, 2019-10-18:
Thanks, but the I2C portion is not the problem. I'm using the Sparkfun MAX30102 library and that is indeed using the FIFO. I start both MAX, they sample at 1000Hz, filling the FIFO as needed. I run a loop checking for new data, and as soon as new data is available I read into a set of working variable, ready to send. If for whatever reason I miss one reading, the next one is always ready on the FIFO. Both devices are read at once when both have data, so the loop is running at 1000Hz, not 2000Hz. At this point I call the Bth write() function. If there is no client connected, the function returns immediately and I have plenty of spare cycles until the next data set is ready to be fetched. If I used, say, the plain old serial port at high enough data rates, I could send data at 1000Hz, no problems

If there is a client connected to the BTh SPP, the Bth write() function takes a variable time, dependent on the receiving end speed. The slower the receiving end, the slower the Bth function. With particularly slow clients, I can miss 2-4 cycles, lowering my frequency to ~250Hz. With fast clients, I can get up to 500Hz. I know that Bth SPP could be fast enough to send my packet if it were operating at 1000Hz by itself and had a full msec to execute (I can get 4kHz and more if all I do is sending Bth packets of the same size of my actual data to a fast client). Hence my desire to run Bth on a core (together with whatever housekeeping Arduino core needs), while I run my other code including I2C on the other core. Assuming a completely parallel execution. And clearly I expect Bth to be non-deterministic, give it's a wireless protocol talking to a wireless client with a complex stack

I simply cannot capture 32 samples then send, as you suggest, because my ESP32 is part of a real time setup where the PC must receive data at 1000Hz, and each packet must contain data read in real time in the previous 1msec slot. Put it another way, the PC on the other end of a Bth link must receive real time data no older than 1msec. So I need to read a sensor and send data all within a 1msec slot. I could afford a few missed packets here and there, but not half the data rate with wild fluctuations (I get from 430 to 550 packets per seconds at the moment). Since I need a deterministic data rate, I would need to drop to ~350Hz to be on the safe side, much too low for the requirements. As I said, missing very few packets won't be a huge problem (each packet is timestamped, so the PC side can handle the loss) and I know that I cannot expect BTh to be completely deterministic at high data rates. But right now I'm too far away from the goal

So I need to be able to independently run 2 set of tasks, each on its core, with no artificial vTaskDelay, one reading data from I2C and sending it to a queue, the second waiting for the queue data to be available, then sending it via Bth, and being ready by the time the next data from the I2C task is put on the queue. Either task has enough spare processor cycle to allow other small tasks to be run in parallel, but not enough time for a vTaskDelay that takes 1msec minimum

Currently the code executes serially, so i'm trying to cram 2 tasks that independently take less than 1msec, but sequentially take much longer, missing slots roughly every other time. I would not expect to be able to do what I want on a multitasking single core, but 2 almost-independent cores should be able to handle it

stickbreaker, 2019-10-18:
It sounds to me that the bottle neck is bluetooth. I don't have one of these MAX30102, but I use SSD1306 128x64 OLEDS and update the screen at 74hz. about 75kB/s.  the SSD1306 is the limiting factor. if you only need 1k 6byte samples, that is only roughly 7kB/s. I would say you are going to have to find the optimum bluetooth packet size to increase your through put. 

  The code I wrote minimizes the i2c overhead to reduce the cpu cycle cost for each read.  I would recommend you review the Sparkfun library and see what limitations exist.  The standard Arduino `Wire()` is limited to 32byte transactions, Arduino_Esp32 increased the transaction limit to 128, both still utilize single byte reading and writing philosophy. So using `Wire.read()` is an _n*p_, where my code is _n+p_ .

Chuck.

robcazzaro, 2019-10-18:
Thanks again but, as I said, there's no bottleneck in the I2C, not Bth per se. And no throughput issue, my Bth packets are ~20 bytes. So I'm sending 20 bytes every 1 msec via Bth, and Bth has plenty of capacity to handle that (as a matter of fact, as I said, I can get 4k packets sending just random data the same size of my actual packets, if the system only uses Bth and nothing else)

And the MAX30102 data rate is driven by the MAX chip, itself running at 1000Hz (one or two MAX chips are irrelevant, at them moment I'm using only one anyway to minimize the variables I'm dealing with). Even if the I2C is slow, the amount of data on the I2C bus is negligible anyway, mostly waiting for the MAX to finish its conversion. So there isn't much to optimize when there isn't much data to start with

I simply have a problem doing the 2 things serially, and there isn't enough optimization possible anywhere to do it on a single core system (which is that the ESP32 is when running Arduino when both I2C and Bth run on the same core, de facto). I need to be able to use both cores in parallel

stickbreaker, 2019-10-18:
When you issue the first `Wire.begin()` the i2c interrupts are tied to core executing the `Wire.begin()` call. so, whenever an I2C interrupt is triggered, it has to process that interrupt on that specific core. So you can set the core affinity by creating a task on a specific core and initializing i2c from that core.  You can submit i2c transaction from either core, but the ISR only executes on initial core.

I don't understand your problem, if i2c works, and bluetooth works then it works.  I think you need to review you performance numbers,  If you run the i2c bus at 400KHz that means 40Kbytes/s. so, that is 40 bytes on the bus per sample max.  If you each transaction needs write(slaveid, address),read(slaveid,6 databytes) you have at a min used 9.  that is without any status checking, or control operations. 

I think you are going to have to use some block operations and gather multiple samples to reduce the per sample overhead. 

The I2C hardware can read/write 32 bytes with one interrupt. a single 6byte sample at a minimum takes 3 interrupts, load txFifo,startMaster,Stop, the 32 sample operation only needs 9, load txFifo, StartMaster, 6 empty rxFifo, stop. 

I do not know how the bluetooth hardware operates, But, I believe the radio hardware uses a much higher cpu cycle count.  if the bluetooth operation uses critical section, it could pause the i2c interrupts.  You can try juggling the core affiliation, I would just block the transactions to reduce the interrupts.

Good luck.

Chuck.

robcazzaro, 2019-10-19:
> When you issue the first Wire.begin() the i2c interrupts are tied to core executing the Wire.begin() call. so, whenever an I2C interrupt is triggered, it has to process that interrupt on that [...]

This helps a lot, thanks. I reorganized my code as per below

My code now is running 2 tasks, each on its own core. By reorganizing code I got rid of the watchdog reset even if the I2C task has no vTaskDelay. which is good. And, yes, I know I should be using queue or other ways to wake up the send task, using a global variable was a quick hack, but I doubt it makes much difference. As soon as I have time I'll also try to suspend the Bth task and wake it up from the I2C task.

For a moment, let's forget everything about I2C and its limitations, I'm building a generic remote sensor and I could use for example a fast SPI accelerometer instead. 

I'm calling a function that reads Red and IR data from a FIFO buffer as fast as I can. I added a kludge to measure how many samples I can read in 1000msec. If there is no Bth SPP client attached, I can read ~1370 samples (the MAX is set to sample 1600 times per second, so I'm losing a few packets, but that's not relevant for now). I also might get some duplicates, once again not relevant. The relevant thing is that with no Bth client, I get 1370 samples per second

My Bth code is looping, waiting for a character to be sent from a client. As soon as a Bth client is connected, the I2C task rate drops to 1160 samples. Clearly the 2 cores are not independent. As soon as I send a handshake character, the task enters the for{;;} loop, and at this point it starts sending Bth data  (roughly 20 bytes per packet, depends on the Red and IR values... and, yes, later on I will send better encoded data, but since I need to send more info, ~20 bytes per packet are to be expected)

As soon as the for{;;} loop starts, the GetI2C task can only process between 380 and 410 packets. Clearly the 2 cores interact a lot more than I would expect in a dual core system (not knowing anything about how the Arduino core works behind the scenes). 

For comparison, I'm also adding a program with just a Bth send loop by itself, sending dummy data identical to my case. In that case I get  ~5000 packets per second, so clearly the Bth SPP link has more than enough capacity to handle my data stream. Even more interesting, if I add vTaskDelay() to the Bth-only loop, I still get exactly 1000 samples, since there seems to be enough time left to complete the 

So, to recap: I2C by itself can handle ~1300 events per second. Bth by itself can handle 5000 events per second, or 1000 events when using vTaskDelay() if nothing else is running. Assuming each ran on its own core, I should be able to capture 1000 events per second and send them in a separate Bth task in the next millisecond or so. Clearly they do not run independently

That is the problem I'm trying to solve, independently of I2C or SPI...

```
#include "BluetoothSerial.h"
#include <Wire.h>
#include "MAX30105.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

long ctr, count;
char charbuffer[30];
uint32_t Red, IR;
bool dataReady = false;

MAX30105 particleSensor;
BluetoothSerial SerialBT;

void SendData(void * parameter)
{
  Serial.begin(115200);
  Serial.println("Initializing...");

  Serial.println(" Done");
  SerialBT.begin("ESP32MAX");
  SerialBT.flush();
  SerialBT.println("Initializing...");
  Serial.print("Initializing BT Serial, waiting for handshake");
  while (SerialBT.available() <=0){
    Serial.print(".");
    vTaskDelay(250);
  }
  Serial.println(" Done");
  for(;;) {
    if (dataReady == true)
    {
      int n=sprintf(charbuffer, "%i - %i,%i\r\n", ctr, Red, IR);
      SerialBT.write((uint8_t *)charbuffer, n);
      dataReady = false;
    }
    vTaskDelay(1);
  }
}

void GetI2C(void * parameter)
{
  // Initialize sensor
  if (!particleSensor.begin(Wire, I2C_SPEED_FAST)) //Use default I2C port, 400kHz speed
  {
    Serial.println("MAX30105 was not found. Please check wiring/power. ");
    while (1);
  }
  //Setup MAX30102
[... 91 lines cut ...]
  dataReady = true;
  Red = 123456;
  IR = 789012;
  ctr = 0;
  count = millis();
}

void loop() {
  if (dataReady == true)    // keeping this only to simulate the same loop as a task
  {
    int n=sprintf(charbuffer, "%i - %i,%i\r\n", ctr, Red, IR);
    SerialBT.write((uint8_t *)charbuffer, n);
  }
  if ((millis() - count) >= 1000)
  {
    ctr = 0;
    count = millis();
  }
  else
  {
    ctr++;
  }
  vTaskDelay(1);   // <--- exactly 1000Hz with this. 5kHz when the vTaskDelay(1) is commented out
}
```

stickbreaker, 2019-10-19:
I don't know how bluetooth Serial uses cores. Under Arduino core 0 is used by most system tasks and the Arduino code is tasked to core 1.  You might try affiliating the I2C to core 0. because the OS gets the interrupt then dispatches it.  Normal code that has no core affinity can be dispatched to either core 

I would create a task to init i2c that is pinned to a core, then end that task, then create a new task that has NO affinity to do the reading. let it bounce to the next available core.

I'd be interested in seeing a oscilloscope image of the i2c bus with your current code.  I wonder how much idle time is available. With the numbers you state it looks like less than 30%.

on your global variable, the ones you use between task need to be marked as volatile
```c++
volatile ebool dataReady;
volatile uint32_t Red, IR;
```
Chuck.

stickbreaker, 2019-10-19:
another thing you think about is the tight loop of:
```c++
  if (particleSensor.check() > 0){
  }
  else { // tight
  }
```
Is there any way this loop can go idle? instead of continuous polling if ready?
If this loop could go idle(`vtaskDelay()`) then I believe you would see an improvement.  
that's why I advocate blocking.  grab a few samples in a block , send them at a standard pace, grab another block. if you grabbed 10 samples (10ms of data) then handed that block to `btsend()`
which paced them out, the `geti2c()` could mark the start time, grab block, go idle for `vtaskDelay((millis()-startTime)+block.count);` 
All of these continuous checks are wasting cpu cycles. 

Chuck.

robcazzaro, 2019-10-19:
Thanks for keeping up the suggestions. (And good catch on the volatile variable, fixed it) 

As I mentioned before, grabbing 10ms of I2C data is not an option, it would be the same as having a 100Hz sampling frequency, which is something I could achieve much more easily (out of those 10 samples, 9 are out of date by the time the PC processes them). I need the PC to receive a sample every millisecond. I can easily do it using plain old serial, so I know that the I2C is not an issue (I'm not sure how else to point out that I2C, by itself, is not the limiting factor). And, no, there's no way for that loop to go idle, since the MAX sensor will spew out its data when ready, and I don't know when. I could add an extra wire and use the interrupt function of the MAX, but it would take an extra wire and even in that case I'm almost positive that the Bth part would still be the limiting factor when run together with other tasks

I swapped the tasks, as follows:

```
  xTaskCreatePinnedToCore(
    SendData,
    "TaskSend",   // A name just for humans
    40000,  // This stack size can be checked & adjusted by reading the Stack Highwater
    NULL,   // task input parameter
    1,  // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
    NULL, // Task handle
    1);  // Core where the task should run, 0=core, 1=Arduino 

  xTaskCreatePinnedToCore(
    GetI2C,
    "TaskGet",   // A name just for humans
    40000,  // This stack size can be checked & adjusted by reading the Stack Highwater
    NULL,   // task input parameter
    2,  // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
    NULL, // Task handle
    0);  // Core where the task should run, 0=core, 1=Arduino
```

Doing this, I get ~500 packets per second, a slight improvement (which is good), but still half as much as I need

And, yes, I understand that the I2C task spends a lot of time being busy, but that's why I want to run it in a separate core :). I know I sound like a broken record, but I expected a dual core system to be able to run 2 separate tasks with at least 70% core utilization on each core. I understand that 2 core is not equal twice as much, but at least 70% on each core is not an unreasonable expectation

I'm inclined to swap out the ESP32 for a STM32+HC-06 module. From a quick test, I can sample the sensor on I2C and send the data out via serial from a STM32 at 115200 bps well within the 1msec time slot, and the HC-06 module will take care of sending it to the PC. 

But I was hoping to use the ESP32, instead of 2 separate modules...

stickbreaker, 2019-10-19:
I disagree with your out dated data conclusion.  by your definition, everything is out dated. because you are measuring/reporting something that happened in the past.  Base on a quick read through that sparkfun library, it is using blocking.  so, your data is already outdated by your definition.  you are chasing instantaneous that does not exit.

I'm out.
Chuck.

robcazzaro, 2019-10-19:
It might be outdated by 1 or 2 msec, but that still provides relevant data to the PC to work with. And 1 or 2 msec delay is acceptable, if I can get a reliable stream with a packet every msec.

And as I said, I can use different sensors that are faster. But I'm still limited by the Bth task interacting with the data reading. 

I really appreciate your input, but in the end it is becoming clear that the FreeRTOS running on a msec tick does not allow the Esp32 sampling data and sending via Bth at 1000hz

robcazzaro, 2019-10-19:
Additional info: running on a basic Arduino loop (i.e. no FreeRTOS tasks), and using the serial port at 115200 instead of Bth SPP, I can get all the I2C data packets at 1000Hz, with no loss of packets, as long as the packets are small enough. Longer packets need 230400bps to run at full speed. 

That means that the Serial library must be doing some level of real dual core performance, given that at 115200 a serial port can only send 11520 characters in a second, and I'm sending 11 bytes per packet at 1000Hz (I actually can get 1050 packets thru) while the I2C code is running and updating in real time. This is what I would expect from a dual core system (actually best case scenario). Basically the Serial port is running at full speed (the 11520 theoretical max) while other code(I2C) is running. On the other hand, the Bth SPP in the same conditions can only do less than 1/10th of the speed while running the I2C bus compared to running on its own

At this point, the clear bottleneck is the way Bth SPP works. Any suggestions on how to improve or at least look into the Bluetooth performance is welcome

beegee-tokyo, 2019-10-19:
I have a similar project where I read 2 MAX30102 sensors over 2 I2C bus and send data over Bluetooth.
Bluetooth serial was way to slow, so I switched to BLE with a custom characteristic which had a data field exactly the size of the data.
My data rate is slower, only 200SPS == 200Hz, but even then Bluetooth serial was no way to go.
If possible for you, you might think about using BLE instead of Bluetooth serial. With BLE I had no problems to get the data out.

And at the end, I switched to a Nordic nRF52832, because I couldn't get the ESP32 to run on battery for 3-4 days. The Nordic nRF52832 has only BLE, but has a very low power consumption. Even with the 2 MAX sensors running and transfering data over BLE with 200Hz I could get 4 days from a 500mAh LiIon .

robcazzaro, 2019-10-19:
Thanks for the additional info @beegee-tokyo. As a matter of fact, I'm experimenting in parallel both with SPP and BLE, to have a comparison. Unfortunately it's likely that the person working on the PC side code will prefer SPP, because BLE support in Windows 10 from Python is rather non-existent.

But even a lowly HC-06 module (Bth 2.1) can handle ~30k characters per second over Bluetooth SPP, assuming a good link. That would mean 1000 packets per second, each of 30 characters, well above my needed speeds. There seems to be a pretty fundamental speed/multicore issue with Bth SPP in the ESP32  under Arduino. By itself Bth SPP is pretty fast, but doesn't use the dual core like plain old serial does

robcazzaro, 2019-10-28:
Adding some information in case anyone reads this as part of a Google search.

I did a lot more test on Bluetooth SPP and how it interacts with the rest of the system. I have 2 tasks, one reading I2C data as soon as an interrupt is detected, and the other to send data. I communicate between tasks and from the interrupts using freeRTOS task notifications, much faster than semaphores, according to the docs

```
  xTaskCreatePinnedToCore(
      GetI2C1,
      "TaskGet",    // A name just for humans
      40000,        // This stack size can be checked & adjusted by reading the Stack Highwater
      NULL,         // task input parameter
      3,            // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
      &xNotifyI2C1, // Task handle
      1);           // Core where the task should run, 0=core, 1=Arduino

  xTaskCreatePinnedToCore(
      SendData,
      "TaskSend",   // A name just for humans
      40000,        // This stack size can be checked & adjusted by reading the Stack Highwater
      NULL,         // task input parameter
      2,            // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
      &xNotifySend, // Task handle
      0);           // Core where the task should run, 0=core, 1=Arduino
```
The I2C reading task runs on the Arduino core (and I initialize the I2C bus in that task), then waiting with a ulNotificationValue = ulTaskNotifyTake(pdFALSE, xMaxBlockTime) for the interrupt from the I2C sensor (the ISR issues a vTaskNotifyGiveFromISR(xNotifyI2C1, &xHigherPriorityTaskWokenByPost). The other task initializes the SerialBluetooth object and then waits for a notification from the I2C task with the same ulNotificationValue = ulTaskNotifyTake(pdFALSE, xMaxBlockTime). The I2C task notifies it's time to send data with xTaskNotifyGive(xNotifySend).

If I use the serial port to send data, the I2C data reading takes around 440 usec, and a notification is issued to the Send task within 14 usec. The Serial port has plenty of time to send a short packet, when loop back and wait for the next.

If I instead of the Serial port I instantiate a BuetoothSerial port and use Bth, the timing of pretty much everything becomes highly unpredictable. The I2C reading from the moment an interrupt is received, takes anywhere from 440 to 1100usec. That is just by issuing a SerialBT.begin(). without even sending Bth data yet. In theory the I2C code is running on Core1, and SerialBT on Core 0, so even assuming that the Bth code needs a lot more code/interrupts to establish a connection, it should not have such an oversized impact on the code running on the other code.

In the end, as stupid as this sound, my prototype will be built using an ESP32 and a separate Bth module (like HC-06 or HM-11) connected to a serial port running at 230400. Serial behaves like I would expect, with the sending not influencing the timing of I2C. For the final device, I will need to see if using the ESP32 is still the best choice or not. We choose the ESP32 because of the integrated Bluetooth, but the Serial BTh code is either poorly optimized or written to use both cores

closed 2019-11-11 by robcazzaro