espressif/arduino-esp32 #3387: esp32 http request works only the first time

ashrafkamel5, 2019-10-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			the problem is present with both 1.0.1 and 1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600
Computer OS:                                        Windows 8.1

### Description:
http requests work fine only the first time 
if i want to make any further requests the httpcode respond is always -1
and  i have to reconnect to my wifi
using 
```cpp
void disable_wifi(void)
{
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  delay(1);
}
```
and then connect again as usual
```cpp
{
  WiFi.mode(WIFI_STA);
  WiFi.begin(wifiName, wifiPass);
  while (WiFi.status() != WL_CONNECTED);
}
```
only then the httpclient works fine and i get the httpcode 200 and the payload
but 
it takes alot of time
need help....
### Sketch:
is just simple http get request
```cpp
WIFI_STATE wifi_http_get(char* host)
{
  int count = 0;
  char data[50];
  WIFI_STATE return_value;
  WiFi.mode(WIFI_STA);
  Serial.print("Connecting to ");
  Serial.println(wifiName);
   
  WiFi.begin(wifiName, wifiPass);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Wi-Fi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP()); //You can get IP address assigned to ES

  count = 0;
  
  http.begin(host);

LABEL:
  int httpCode = http.GET(); //Make the request
   Serial.print("httpCode :");
   Serial.println(httpCode);
  if (httpCode > 0)
  { //Check for the returning code

    payload = http.getString();
  }
  else
  {
     Serial.println("Error on HTTP request");
  }
  http.end(); //Free the resources
  Serial.print("payload :");
  Serial.println(payload);
  if (httpCode == 200)
  {
    return_value = W_HTTP_OK;
  }
  else if (httpCode == -1 && count < 2)
  {
    count++;
    delay(500);
    goto LABEL;
  }
  else if(httpCode == -1)
  {
    return_value =  INTERNET_LIMITED;
  }
  else
  {
    return_value = WIFI_CONNECTED_HTTP_ERROR;
  }
  disable_wifi(); 
  return return_value;
}
```

ashrafkamel5, 2019-10-20:
using multiWiFi class instead of WiFi for connection solved the issue

closed 2019-10-20 by ashrafkamel5

vt-d-developer, 2021-01-08:
I faced same situation. After lots of R&Ds I found, Its problem of connection reuse. In httpclient connection reuse is true by default.
I was using python anywhere free account which allows only one thread so no 2 or more connection at a time. But due to connection reuse client keeps connection open.  So server does not allow new connection until last connection close or timeout.
The solution is for this problem had to set **httpClient.setReuse(false);** . Due to this client close connection forcefully and make server to accept new connection.  Check if you server is single threaded. Usually most of development servers are single threaded.

ashrafkamel5, 2021-01-10:
thanks for your update,@vt-d-developer

ArnieO, 2024-01-10:
> The solution is for this problem had to set **httpClient.setReuse(false);**

Does not work for me.