espressif/arduino-esp32 #3388: Strange issue with FFat multiple begin() and end()

mouridis, 2019-10-19:
### Hardware:
Board:							Wemos LOLIN32
Core Installation version:			1.04+ (commit 91e095f5a76ec35f06c86689a87892425395b8db)
IDE name:						Arduino IDE 1.8.6
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10 x64

### Description:
Individual file operations on FFat work when there is one global FFat mount in the sketch but don't work when each file operation mounts and unmounts the file system. Let me explain with examples below.

The following sketch formats the FFat partition, mounts it, and then performs a file write operation, a directory listing operation and a file read operation to confirm that the file was indeed written successfully.

```cpp
#include "FS.h"
#include "FFat.h"

void setup(){
  Serial.begin(115200);
  FFat.format();
  if (!FFat.begin()){
    Serial.println("FFat Mount Failed");
    return;
  }
  writeFile("/hello.txt", "Hello Worlds!!!");
  listRootDir();
  readFile("/hello.txt");
  FFat.end();
}

void loop() {
}

void listRootDir() {
  Serial.println("Listing root directory:");
  File root = FFat.open("/");
  if (!root){
    Serial.println("- failed to open root directory");
    return;
  }
  File file = root.openNextFile();
  while (file) {
    if (file.isDirectory()) {
      Serial.printf("  DIR : %s\r\n", file.name());
    } else {
      Serial.printf("  FILE: %s\tSIZE: %u\r\n", file.name(), file.size());
    }
    file = root.openNextFile();
  }
}

void readFile(const char * path) {
  Serial.printf("Reading file: %s\r\n", path);
  File file = FFat.open(path);
  if (!file || file.isDirectory()) {
    Serial.println("- failed to open file for reading");
    return;
  }
  while (file.available()) {
    Serial.write(file.read());
  }
}

void writeFile(const char * path, const char * message) {
  Serial.printf("Writing file: %s", path);
  File file = FFat.open(path, FILE_WRITE);
  if (!file) {
    Serial.println("- failed to open file for writing");
    return;
  }
  if (file.print(message)) {
    Serial.println(" - file written");
  } else {
    Serial.println(" - write failed");
  }
}
```

This sketch works well and produces the following output:
```
Writing file: /hello.txt - file written
Listing root directory:
  FILE: /hello.txt	SIZE: 15
Reading file: /hello.txt
Hello Worlds!!!
```

Now, suppose we slightly modify the script so that the FFat.begin() and FFat.end() do not take place on setup(), but on each of the writeFile(), readFile() and listRootDir() functions. The rest of the code remains exactly the same:
```cpp
#include "FS.h"
#include "FFat.h"

void setup(){
  Serial.begin(115200);
  FFat.format();
  writeFile("/hello.txt", "Hello Worlds!!!");
  listRootDir();
  readFile("/hello.txt");
}

void loop() {
}

void listRootDir() {
  Serial.println("Listing root directory:");
  if (!FFat.begin()){
    Serial.println("FFat Mount Failed");
    return;
  }
  File root = FFat.open("/");
  if (!root){
    Serial.println("- failed to open root directory");
    FFat.end();
    return;
  }
  File file = root.openNextFile();
  while (file) {
    if (file.isDirectory()) {
      Serial.printf("  DIR : %s\r\n", file.name());
    } else {
      Serial.printf("  FILE: %s\tSIZE: %u\r\n", file.name(), file.size());
    }
    file = root.openNextFile();
  }
  FFat.end();
}

void readFile(const char * path) {
  Serial.printf("Reading file: %s\r\n", path);
  if (!FFat.begin()){
    Serial.println("FFat Mount Failed");
    return;
  }
  File file = FFat.open(path);
  if (!file || file.isDirectory()) {
    Serial.println("- failed to open file for reading");
    FFat.end();
    return;
  }
  while (file.available()) {
    Serial.write(file.read());
  }
  FFat.end();
}

void writeFile(const char * path, const char * message) {
  Serial.printf("Writing file: %s", path);
  if (!FFat.begin()){
    Serial.println("FFat Mount Failed");
    return;
  }
  File file = FFat.open(path, FILE_WRITE);
  if (!file) {
    Serial.println("- failed to open file for writing");
    FFat.end();
    return;
  }
  if (file.print(message)) {
    Serial.println(" - file written");
  } else {
    Serial.println(" - write failed");
  }
  FFat.end();
}
```

This sketch doesn't work. This is the output:
```
Writing file: /hello.txt - file written
Listing root directory:
Reading file: /hello.txt
- failed to open file for reading
```

Despite the fact that the individual FFat.begin() return true in all situations (otherwise a "FFat Mount Failed" message would appear) and despite the fact that the file write operation also returns true (as indicated by "-file written" in the output), no file is written in the partition.

Am I doing something wrong or is there an issue with FFat library?

lbernstone, 2019-10-20:
You must close the file before ending FFat.

mouridis, 2019-10-20:
Damn! So simple. Lost many hours looking in the wrong places about what's wrong! Thank you!

closed 2019-10-20 by mouridis