espressif/arduino-esp32 #3389: Possible Memory leak of 3kB in BLEDevice and possible Bug: Not setting initialized to false in BLEDevice.cpp

jubueche, 2019-10-20:
Hi,

I am running a program that initializes a BLE connection using this code:

```
void Chaze_ble::initialize_connection(){

// Create the BLE Device
  if(!BLEDevice::getInitialized())
  {
    BLEDevice::init("UART Service");
  }

  //Set the MTU of the packets sent, maximum is 500, Apple needs 23 apparently.
  BLEDevice::setMTU(25);
  config.MTU_BLE = 23;

  // Create the BLE Server
  pServer = BLEDevice::createServer(); 

  ESP_LOGI(TAG_BLE, "Created server");
  if(callbacks == NULL)
  {
    callbacks = new MyServerCallbacks();
  }
  pServer->setCallbacks(callbacks);
  // Create the BLE Service
  pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
										CHARACTERISTIC_UUID_TX,
										BLECharacteristic::PROPERTY_NOTIFY
									);
                      
  if(ble_2902 == NULL)
  {
    ble_2902 = new BLE2902();
  }
  pTxCharacteristic->addDescriptor(ble_2902);

  pRxCharacteristic = pService->createCharacteristic(
											 CHARACTERISTIC_UUID_RX,
											BLECharacteristic::PROPERTY_WRITE
										);

  // Start the service
  pService->start();

}
``` 

And then I (try) to clean up the resources used using this function:

```
my_ble::~my_ble(void){
  
  BLEDevice::deinit();

  pServer->removeService(pService);

  free(pServer);
  free(pService);
  
  pTxCharacteristic->~BLECharacteristic();
  pRxCharacteristic->~BLECharacteristic();

  free(pTxCharacteristic);
  free(pRxCharacteristic);

  free(ble_2902);
  ble_2902 = NULL;

  //Release the wifi_synch_semaphore
  ESP_LOGE(TAG_BLE, "Free heap space is %d", esp_get_free_heap_size());
}
```

Am I missing something?

Also something else that I have noticed:

When I deinitialize the BLEDevice using ```BLEDevice::deinit()``` it seems like this code:

```
void BLEDevice::deinit(bool release_memory) {
    if (!initialized) return;

	ESP_LOGI("BLEDevice", "Called deinit of BLE Device.");
    esp_bluedroid_disable();
    esp_bluedroid_deinit();
    esp_bt_controller_disable();
    esp_bt_controller_deinit();
	// initialized = false; // Shouldn't this be added?
#ifndef ARDUINO_ARCH_ESP32
    if (release_memory) {
        esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);  // <-- require tests because we released classic BT memory and this can cause crash (most likely not, esp-idf takes care of it)
		ESP_LOGE("BLEDevice", "Releasing memory!!");
                // Why is initialized = false not set here? Because we cannot reinitialize anyway (bc of release_mem=true)?
	} else {
		ESP_LOGE("BLEDevice", "Set initialized to false");
        initialized = false;   
    } 
#endif
}
```
Should set initialized to false regardless of whether the flag ```ARDUINO_ARCH_ESP32``` is set.

jubueche, 2019-10-20:
Apparently the memory leak disappears when omitting the call to ```esp_bluedroid_deinit()```. The problem then becomes that ```btStart()``` gets stuck in a while loop:

```
if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
        esp_bt_controller_init(&cfg);
        while(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){}
  }
```

closed 2020-01-02 by stale[bot]

peter1a, 2020-10-02:
Any news with that?
Anybody found solution for deinit / init ble?