espressif/arduino-esp32 #3390: Reading from sensor via I2C shows weird behavior

VanKurt, 2019-10-21:
### Description:
With my ESP32 I want to read data from a MPU6050 sensor via I2C. Most of the time this works as expected. But sporadically, the I2C communication freaks out.

This is a valid communication frame. The ESP32 reads 14 bytes from the sensor:
[image]

And this is how it looks like when things go wrong:
[image]

As you can see, the ESP32 sets a NACK in the middle of the communication. After that it just continues reading. Of course the slave does not answer anymore (since it has seen the NACK). After the 14 bytes, the communication does not to end. The bus is stuck and the ESP32 runs in a timeout (50ms), then restarts the bus.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     Dunno
Upload Speed:					115200
Computer OS:                                         Windows 10

### Sketch: 
I use the MPU6050 library from here: https://github.com/TKJElectronics/KalmanFilter/tree/master/examples/MPU6050
This contains the following I2C code used to read 14 bytes from the slave:
```cpp
uint8_t i2cRead(uint8_t registerAddress, uint8_t *data, uint8_t nbytes) {
  uint32_t timeOutTimer;
  Wire.beginTransmission(IMUAddress);
  Wire.write(registerAddress);
  uint8_t rcode = Wire.endTransmission(false); // Don't release the bus
  if (rcode) {
    Serial.print(F("i2cRead failed: "));
    Serial.println(rcode);
    return rcode; // See: http://arduino.cc/en/Reference/WireEndTransmission
  }
  Wire.requestFrom(IMUAddress, nbytes, (uint8_t)true); // Send a repeated start and then release the bus after reading
  for (uint8_t i = 0; i < nbytes; i++) {
    if (Wire.available())
      data[i] = Wire.read();
    else {
      timeOutTimer = micros();
      while (((micros() - timeOutTimer) < I2C_TIMEOUT) && !Wire.available());
      if (Wire.available())
        data[i] = Wire.read();
      else {
        Serial.println(F("i2cRead timeout"));
        return 5; // This error value is not already taken by endTransmission
      }
    }
  }
  return 0; // Success
}
```

### Debug Messages:
I have enabled debugging, which gives me the following output when the problem occurs:
```
[D][esp32-hal-i2c.c:1345] i2cProcQueue():  Gross Timeout Dead start=0x84aa8, end=0x84adb, =51, max=51 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbec00
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb843c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=200
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb9078
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=2
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=1
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=2
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:288] i2cDumpDqData(): [0] 7bit 68 W  buf@=0x3ffc0482, len=01, pos=01, ctrl=11101
[I][esp32-hal-i2c.c:306] i2cDumpDqData(): 0x0000: ;                                3b 
[I][esp32-hal-i2c.c:288] i2cDumpDqData(): [1] 7bit 68 R STOP buf@=0x3ffc03fc, len=14, pos=14, ctrl=11111
[I][esp32-hal-i2c.c:306] i2cDumpDqData(): 0x0000: .......;......                   00 c8 04 9a bc 12 f0 3b ff a9 80 ff 7f 7f 
[I][esp32-hal-i2c.c:346] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [01]	0x0001	0x0002	0x0003	0x0000	0x00084aa8
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [02]	0x0001	0x0200	0x0000	0x0000	0x00084aa8
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [03]	0x0003	0x0100	0x0000	0x000e	0x00084ad1
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[V][esp32-hal-i2c.c:1484] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1680] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1696] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1704] i2cSetFrequency(): Fifo delta=1
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=1
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles
```

stickbreaker, 2019-10-21:
@VanKurt you are going to have to capture the log output when the error happens.  The log you posted is after the error. you can force logging with the `Wire.setDebugFlags()` as explained in the [i2c_debugging.md](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/doc/i2c_debugging.md).

the interesting bits would be 0x07. The master is generating the clock. the slave does not know the length of the read transmission.  it will keep sending until it see a NAK from the master.  Your logic analyzer capture show the master generating all the 14 bytes worth of clock. the Slave stops sending after it sees the NAK.  Something is driving the SDA after the ESP32 completes.  One of the devices on the bus is out of sequence.

I have never seen any report of this failure.

An explanation could be that that the slave missed a clock cycle(a glitch) during that last successful byte.

Increase your bandwidth on your capture, see if there are any signal glitches on the SCL line.

Chuck.

VanKurt, 2019-10-22:
@stickbreaker 
Thank's for your input. It seems there is actually some sort of noise on the I2C lines. This might trigger various errors, depending on where/when it occurs.

So for now it seems safe to assume there is nothing wrong with the I2C implementation.

closed 2019-10-22 by VanKurt