espressif/arduino-esp32 #3392: Cannot locate Additional Board Manager URLs

franpog859, 2019-10-21:
### Hardware:
Board:							NodeMCU ESP-32S
Core Installation version:			Arduino 2:1.0.5+dfsg2-4.1
IDE name:						Arduino IDE
Flash Frequency:					
PSRAM enabled:                                    
Upload Speed:					
Computer OS:                                        Ubuntu

### Description:

I try to install arduino-esp32 with Boards Manager which is preferred but I cannot locate the field mentioned [here](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) 
```
Enter one of the release links above into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
```
How could I install it then with Boards Manager?

Bmooij, 2019-10-21:
Check out: https://www.arduino.cc/en/guide/cores
`How to install a third party core`

lbernstone, 2019-10-21:
Use arduino >1.6.4 .  The version you are using is several years old.

franpog859, 2019-10-21:
You are right! I installed it using `apt-get install arduino` and this version appeared on my computer. I installed the newest version on Windows and everything looks fine. Probably if I install it correctly it will work also on Ubuntu. Thanks for the fast answers! 🎉 
I think it's ok to close this issue 😉

closed 2019-12-22 by franpog859