espressif/arduino-esp32 #3393: MDNS crash when appending txt record

Bmooij, 2019-10-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
I'm using the app [Service Browser](https://play.google.com/store/apps/details?id=com.druk.servicebrowser&hl=en_US) [(source code)](https://github.com/andriydruk/BonjourBrowser)

And when I request the details of my mDNS service, the ESP crashes:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```
Core 0 register dump:
PC      : 0x4019a968  PS      : 0x00060830  A0      : 0x8019b528  A1      : 0x3ffdc030  
A2      : 0x3ffc6028  A3      : 0x3ffdc0ce  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8019a95d  A9      : 0x3ffdc010  
A10     : 0x3ffd9920  A11     : 0x00000003  A12     : 0x00060620  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4019a968:0x3ffdc030 0x4019b525:0x3ffdc080 0x4019b74e:0x3ffdc0c0 0x4019bccb:0x3ffdc0f0 0x4019cd5b:0x3ffdc120 0x4019d2e7:0x3ffdc1b0 0x4019d34b:0x3ffdc1f0 0x4008a991:0x3ffdc220
```
Decoded stack results:
```
0x4019a968: _mdns_append_txt_record at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 534
0x4019b525: _mdns_append_answer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 850
0x4019b74e: _mdns_dispatch_tx_packet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 933
0x4019bccb: _mdns_create_answer_from_parsed_packet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 1266
0x4019cd5b: mdns_parse_packet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 3001
0x4019d2e7: _mdns_execute_action at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 3877
0x4019d34b: _mdns_service_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c line 3993
0x4008a991: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

I used wireshark to log the requested query:
```
source						destination		Protocol	length	Info
fe80::c2ee:fbff:fedc:8694	ff02::fb		MDNS		91		Standard query 0x0000 TXT myhost.local, "QM" question
``` 
### Sketch
```cpp
MDNS.begin("myHost");
MDNS.setInstanceName("myHost_45213548795212");

MDNS.addService("myService", "_tcp", 80);
MDNS.addServiceTxt("myService", "_tcp", "identifier", "45213548795212");
MDNS.addServiceTxt("myService", "_tcp", "application", "my.web.server");

char buffer[100];

sprintf(buffer, "%d", 1);
MDNS.addServiceTxt(_serviceName, "_tcp", "versionCode", buffer);

sprintf(buffer, "%d", 5);
MDNS.addServiceTxt(_serviceName, "_tcp", "coreVersionCode", buffer);
```

lbernstone, 2019-10-22:
Does it crash if you use "http" instead of "myService"/_serviceName?

PoojaG20, 2019-10-26:
Hi i am using Arduino IDE, i am also getting this error...any luck on why it comes

chegewara, 2019-10-26:
https://github.com/espressif/esp-idf/issues/4224

Bmooij, 2019-10-26:
If that is true, according to the [fix](https://github.com/espressif/esp-idf/commit/3d1170031b340a231949fdc0d9c46d87af0d1b5d), the possible work around would be to set the mDNS instance name.

If the instance name was not set, the hostname is used by default [link](https://docs.espressif.com/projects/esp-idf/en/v3.2.3/api-reference/protocols/mdns.html))

closed 2020-01-08 by stale[bot]