espressif/arduino-esp32 #3395: Unable to install, esp32-1.0.4.zip cannot be downloaded

jbrepogmailcom, 2019-10-22:
When trying to download https://github.com/espressif/arduino-esp32/releases/download/1.0.4/esp32-1.0.4.zip, I get error

The webpage at https://github-production-release-asset-2e65be.s3.amazonaws.com/70127218/8cbf5700-e513-11e9-9a02-f0cf91d23417?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191022%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191022T095409Z&X-Amz-Expires=300&X-Amz-Signature=b0ec6dba5f81635300e21d6684b1b5d99d73c8ad0ca69ee37c1216699ee79117&X-Amz-SignedHeaders=host&actor_id=17232670&response-content-disposition=attachment%3B%20filename%3Desp32-1.0.4.zip&response-content-type=application%2Foctet-stream might be temporarily down or it may have moved permanently to a new web address.

Bmooij, 2019-10-22:
it looks like it was `temporarily down`, it should work now

tyrrr-aj, 2019-10-22:
I was unable to download the file too, it redirects to:
https://github-production-release-asset-2e65be.s3.amazonaws.com/70127218/8cbf5700-e513-11e9-9a02-f0cf91d23417?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20191022%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191022T123548Z&X-Amz-Expires=300&X-Amz-Signature=9401af441415b8ea5d51947faebd094e29f1e377f9bd3e6117a8e2bbe0b6f76f&X-Amz-SignedHeaders=host&actor_id=38702585&response-content-disposition=attachment%3B%20filename%3Desp32-1.0.4.zip&response-content-type=application%2Foctet-stream

and ends with DNS_PROBE_FINISHED_NXDOMAIN

jbrepogmailcom, 2019-10-22:
Hello, it appears to be problem with some wifi routers. I have created wifi apn from my mobile, connected to that and download worked

closed 2019-10-22 by jbrepogmailcom

aastha3107, 2020-05-20:
when i try to install ESP32 to Ardiuno IDE i keep getting this error 
 Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.4/esp32-1.0.4.zip

Can someone please help ?