espressif/arduino-esp32 #3396: missing: code examples to read string inputs via html GET and POST

dsyleixa, 2019-10-22:
missing: code examples to read string inputs via html GET and POST
can you please provide some in your standard repo examples?

Bmooij, 2019-10-22:
Look at the [HelloServer](https://github.com/espressif/arduino-esp32/blob/91e095f5a76ec35f06c86689a87892425395b8db/libraries/WebServer/examples/HelloServer/HelloServer.ino#L30) example

You can use the following function:
[arg](https://github.com/espressif/arduino-esp32/blob/91e095f5a76ec35f06c86689a87892425395b8db/libraries/WebServer/src/WebServer.h#L101) // get request argument value by name

atanisoft, 2019-10-22:
@dsyleixa those are basic tasks that are not specific to esp32 and should not be posted as an issue. Instead it should be posted to the forum on www.esp32.com

dsyleixa, 2019-10-22:
sorry, no, they should be available in the examples in 
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
just like other basic examples!

atanisoft, 2019-10-22:
@dsyleixa Feel free to submit a PR for examples you feel are missing.

dsyleixa, 2019-10-22:
that is what I just am doing with this ssue report.
they would be needed urgently, but I can't write them though.

atanisoft, 2019-10-22:
@dsyleixa Re-reading your issue here, it seems you are needing HTML help and not something specific to the ESP32. I'd suggest learning how to do what you are after in the HTML code itself.

dsyleixa, 2019-10-22:
I actually do not mid about what you are thinking, this issue is about code examples to read keyboard string inputs via html GET and POST in the examples of https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples

atanisoft, 2019-10-22:
HTTP GET/POST would only be *AFTER* the user has entered whatever text they are entering in the text input box in the HTML form. The server side does not care or know about any keyboard input, once the request reaches the server the data is complete and no user input possible.

You will need to do any special keyboard input handling on the HTML side only.

Bmooij, 2019-10-22:
You will never find HTTP examples in `libraries/WiFi/examples`. WiFi examples are TCP/UDP connections over WiFi.

For HTTP examples you can look here:
Server: https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples
Client: https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPClient/examples

Bmooij, 2019-10-22:
@dsyleixa do you have an example of an other Arduino library that supplies such example?

atanisoft, 2019-10-22:
@dsyleixa you will need to provide a valid use case, as I've mentioned above what you are after is not feasible. You are mixing up client side and server side. GET/POST are server side and the keyboard would be client side.

This is also not the first time you have blown off valid criticism and requests for you to write some snippets of code to solve your issues.

dsyleixa, 2019-10-22:
which part of 

> 
> this issue is about additional code examples to read keyboard string inputs via html GET and POST in the examples of https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
> or perhaps also in https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples

didn't you understand?

atanisoft, 2019-10-22:
@dsyleixa which part did you not understand of my post? By the time the GET/POST request reaches the ESP32 there is no keyboard input possible. The only time a keyboard can be involved is *BEFORE* the GET/POST request is sent and that would all be handled inside the HTML code that doesn't run on the ESP32, it runs inside the browser.

lbernstone, 2019-10-22:
SimpleAuthentication shows a pretty straightforward form POST transaction.

dsyleixa, 2019-10-22:
SimpleAuthentication is an example, but it is not doing it via html.
but anyway, those examples ready to compile+run are missing, and that is what this issue is about.

lbernstone, 2019-10-22:
You need to clarify what your issue is.  You are being quite hostile here, and not accepting our help.  We would like to help you.

dsyleixa, 2019-10-22:
I'm not hostile but I am annoyed about off-topc answers.
My issue report is just about (missing) additional code examples to read keyboard string inputs via html GET and POST in the examples of https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
or perhaps also in https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples

lbernstone, 2019-10-22:
read keyboard string inputs where?  In the browser?  @atanisoft has already explained that esp32 has nothing to do with the browser keyboard.  The webserver provides a page with a form in it, and the client sends a POST request with the form results.  The server can then process the results.  SimpleAuthentication does this.  Examples are to provide reference for the methods, not to solve your exact problem.

Bmooij, 2019-10-22:
I think what @dsyleixa means is that the available examples don't ship with the required HTML forms for testing

dsyleixa, 2019-10-22:
you or anyone else should write such examples an upload them in https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
or perhaps also in https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples

lbernstone, 2019-10-22:
```
  String content = "<html><body><form action='/login' method='POST'>To log in, please use : admin/admin<br>";
  content += "User:<input type='text' name='USERNAME' placeholder='user name'><br>";
  content += "Password:<input type='password' name='PASSWORD' placeholder='password'><br>";
  content += "<input type='submit' name='SUBMIT' value='Submit'></form>" + msg + "<br>";
  content += "You also can go <a href='/inline'>here</a></body></html>";
  server.send(200, "text/html", content);
```

dsyleixa, 2019-10-22:
am  really speking Chinese?

I mean  I mean comlete (!) examples how to retrieve a string input from keyboard (until Enter is pressed) from a keyboard text input field on the website:
a) using the GET method to read it
b) using the POST method to read it.
then e.g. display the string which has been read in the Serial console.

and all those uploaded  in https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
or perhaps also in https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples

atanisoft, 2019-10-22:
@dsyleixa nobody is going to do the work *FOR* you.. you will need to invest some time to write what you are after.

dsyleixa, 2019-10-22:
there are already code exaples provided by esp devs, and    this issue is about additional code examples to read keyboard string inputs via html GET and POST in the examples of https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples
    or perhaps also in https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer/examples

do you finally understand?

me21, 2019-10-23:
But what do you need it for? What problem are you trying to solve?

dsyleixa, 2019-10-23:
Arduino Users are mostly non-professional hobby programmers and need examples to see the correct way to handle keyboard-text input on a html website just like how to handle a button-press or to show up analog or digital values  on a html website, all are basic operations e.g. to develop an IoT smart home program.

atanisoft, 2019-10-23:
There is no need to duplicate something like https://www.w3schools.com/ throughout every project that provides a webserver like example.

dsyleixa, 2019-10-23:
atanisoft, I am really annoyed about your snooty and ignorant attitude.
Again, my topic is about working additional examples suitable for beginners, not about plain html code or a plain wifiserver example.

atanisoft, 2019-10-23:
@dsyleixa Many of us have tried repeatedly to assist you and yet you continue to disregard any of our advice and continue to copy/paste the same request over and over again. I will assert again that *YOU* need to take an active part in resolving *YOUR* perceived lack of examples, as we have said there are examples available (many of which are not esp32 specific) that can be used to accomplish what you are after, not all of those examples within this repository (and nor should they be).

dsyleixa, 2019-10-23:
I didn't ask YOU, and I never would, tbh...

stickbreaker, 2019-10-23:
@atanisoft he is still trolling. Wah, Wah, Do it for me, I can't do it for myself. You must do it for me because I don't want to learn how to do it for myself. Wah, Wah! Like Barbie said "programming is hard, I'm just a girl!"

dsyleixa, 2019-10-23:
stickbreaker,
I think YOU are trolling, in my thread-opening post I just requested additional examples.

stickbreaker, 2019-10-23:
"Do it for me, I don't know how"  is the same a "I just requested additional examples"

dsyleixa, 2019-10-23:
atanisoft, stickbreaker:
is here a moderator around  to remove your trolling posts?

stickbreaker, 2019-10-23:
Thumbs down just show my accuracy.  If I was incorrect It would not hurt.  I am one, @atanisoft and @lbernstone are other.

me21, 2019-10-23:
In my opinion, there's enough information in documentation and examples at least to get started. One can always write more and more examples, it can never end.

Show us some effort, post some code you've written which should work from your point of view, but it doesn't. Search ESP32 forums, there are some on the internet.

Currently, you don't want to lift a finger, you said you can't write the examples yourself. Why do you expect someone to do it instead of you? Looks like you're the only one who needs more examples.

Anyway, the devs have more urgent matters at the moment than writing more examples and teaching someone who doesn't want to learn.

I think this issue should be closed, this is not a support forum.

closed 2019-10-23 by dsyleixa

atanisoft, 2019-10-23:
@dsyleixa 

> removed because of trolling posts

The only "trolling" was by you, your unwillingness to contribute to a meaningful conversation about examples which you have a desire to see included. Unfortunately you have deleted all of your posts to this issue and closed it.  So I'm guessing you don't really want assistance afterall but instead wanted someone to do all of the work for you.