espressif/arduino-esp32 #3397: Can't connect to ESP32 BLE via Service UUID

mrdc, 2019-10-22:
Hello,

I'm trying to connect to ESP32-Wroom from my iOS app via BLE using Service UUID, but I get an error that no device is connected. I can connect via BLE DeviceName or via DeviceUUID.

### Hardware:
Board:							ESP32 DevKit v4 (38pin)
Core Installation/update date:			ESP32 by Espressif v1.0.4
IDE name:					        Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:						921600
Computer OS:                                           Win

```cpp

#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLECharacteristic.h>
// See the following for generating UUIDs:
// https://www.uuidgenerator.net/
#include "WiFi.h"

#define WIFI_SSID ""
#define WIFI_PASSWORD ""

#define SERVICE_UUID         "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC1_UUID "aeb5483e-36e1-4688-b7f5-ea07361b26a8"
#define CHARACTERISTIC2_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class Characteristic1_Callbacks: public BLECharacteristicCallbacks {
  void onRead(BLECharacteristic *pCharacteristic){
    Serial.println("Characteristic1 Read");
  }
  void onWrite(BLECharacteristic *pCharacteristic){
    Serial.print("Characteristic1 Written: ");
    Serial.println((char *) pCharacteristic->getData());
  }
};

class Characteristic2_Callbacks: public BLECharacteristicCallbacks {
  void onRead(BLECharacteristic *pCharacteristic){
    Serial.println("Characteristic2 Read");
  }
  void onWrite(BLECharacteristic *pCharacteristic){
    Serial.print("Characteristic2 Written: ");
    Serial.println((char *) pCharacteristic->getData());
  }
};
static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    
    BLEDevice::setCustomGapHandler(my_gap_event_handler);
    BLEDevice::setCustomGattsHandler(my_gatts_event_handler);
    BLEDevice::setCustomGattcHandler(my_gattc_event_handler);

  BLEDevice::init("Long");
  Serial.println("Starting BLE work!");
  BLEServer *pServer = BLEDevice::createServer();
  Serial.println("Starting BLE work!");
  BLEService *pService = pServer->createService(SERVICE_UUID);

  Serial.println("Starting BLE work!");
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC1_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new Characteristic1_Callbacks());
  pCharacteristic->setValue("I am Characteristic #1");

  Serial.println("Starting BLE work!");
  pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC2_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new Characteristic2_Callbacks());
  pCharacteristic->setValue("I am Characteristic #2");

  Serial.println("Starting BLE work!");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}

```

chegewara, 2019-10-25:
This looks like issue with iPhone app.

lyusupov, 2019-10-26:
@chegewara 

Should that:
```
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
```
be actually coded like this:
```
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMaxPreferred(0x12);
```
?

mrdc, 2019-10-26:
> be actually coded like this:

Probably yes, I've taken this code (marked as working) from one of issues: https://github.com/espressif/arduino-esp32/issues/2401
The sketch is from @chegewara

chegewara, 2019-10-26:
> @chegewara
> 
> Shouldn't that:
> 
> ```
>   pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
>   pAdvertising->setMinPreferred(0x12);
> ```
> 
> be actually coded like this:
> 
> ```
>   pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
>   pAdvertising->setMaxPreferred(0x12);
> ```
> 
> ?

Yes, its typo.

mrdc, 2019-10-26:
Makes no difference - iOS app can connect via DeviceName but via Service UUID.

chegewara, 2019-10-27:
Check service UUID, maybe you have typo in iOS app.

mrdc, 2019-10-27:
I've checked - it's copy paste from ESP32 sources...

closed 2020-01-22 by stale[bot]