espressif/arduino-esp32 #3398: How to set DHCP Client Name for ESP ?

GeorgeFlorian, 2019-10-22:
In the DHCP Client List of my Access Point, the ESP's Client Name is `unknown`.

How can I change that ?

I am using an ETH connection so I have tried using `ETH.setHostname(host_name);` but this doesn't work and I have also added `WiFi.setHostname("host_name");` which also seems to not resolve my issue.

How can I fix this ? Thank you !

atanisoft, 2019-10-22:
I believe this is a know issue and is tracked in esp-idf as this functionality is provided by the LwIP layer and not the arduino-esp32 side.

GeorgeFlorian, 2019-10-22:
> I believe this is a know issue and is tracked in esp-idf as this functionality is provided by the LwIP layer and not the arduino-esp32 side.

This roughly translates to "whoever uses arduino-esp32 can't set hostname", correct ?

atanisoft, 2019-10-22:
Worse than that, it is not sent via the DHCP packets at all for esp-idf either.

You can however set the hostname via mdns which will work for other usecases.

atanisoft, 2019-10-22:
https://github.com/espressif/arduino-esp32/issues/2537

GeorgeFlorian, 2019-10-22:
Can I use `mDNS` with `ETH` and `ESPAsyncWebServer` and `UDP` and `TCP` ? It looks like `mDNS` is some sort of server.

atanisoft, 2019-10-22:
Yes, it should work fine with all interfaces. mDNS uses a background task to listen and respond to UDP broadcast packets from other devices. Take a look at the examples [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESPmDNS).

Shuptuu, 2019-11-19:
Got a workaround based on the one used for WiFi.setHostname.
If you're using WiFi.onEvent, in case SYSTEM_EVENT_ETH_CONNECTED, add:
`ETH.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);
ETH.setHostname("myhostname");`
it works for me, no more "unknown" on my router.

GeorgeFlorian, 2019-11-19:
> Got a workaround based on the one used for WiFi.setHostname.
> If you're using WiFi.onEvent, add in case SYSTEM_EVENT_ETH_CONNECTED:
> `ETH.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);`
> `ETH.setHostname("myhostname");`
> it works for me, no more "unknown" on my router.

Isn't `ETH.setHostname("myhostname");` used for Ethernet connection ? Why would it work on a WiFi connection ?

closed 2020-02-01 by stale[bot]

mepster, 2020-11-03:
> Got a workaround based on the one used for WiFi.setHostname.
> If you're using WiFi.onEvent, add in case SYSTEM_EVENT_ETH_CONNECTED:
> `ETH.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);`
> `ETH.setHostname("myhostname");`
> it works for me, no more "unknown" on my router.

@Shuptuu thanks for the tip... from your suggestion I found that this simpler version works for me:

WiFi.begin(ssid, pass);
// doesn't work without the following line
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname("FOOBAR");
 
I found that I did not have to set those event handlers.