espressif/arduino-esp32 #3399: ESP32-WROOM-32 i2c IntegerDivideByZero error

Speedy2k, 2019-10-22:
I have a Adafruit HUZZAH32 that as the ESP32-WROOM-32 chip on it.
I try to use a HD44.. LCD with the liquid_crystal_i2c library modified for ESP32.

The much i write to the LCD from the program the more frequently i get this error.

At first i have been obligated in my lcd library to set the Wire.setClock(10000); because if i don't, nothing is working. Now it work, but if i try to write to the LCD very frequently, i always got this guru meditation error:

PC: 0x400e78bf: i2cProcQueue at C:\Users\i7\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-i2c.c line 1279
EXCVADDR: 0x00000000

Decoding stack results
0x400e78bf: i2cProcQueue at C:\Users\i7\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-i2c.c line 1279
0x400e7c7b: i2cWrite at C:\Users\i7\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-i2c.c line 1624
0x400e6d39: TwoWire::writeTransmission(unsigned short, unsigned char*, unsigned short, bool) at C:\Users\i7\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src\Wire.cpp line 133
0x400e6d6d: TwoWire::endTransmission(bool) at C:\Users\i7\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src\Wire.cpp line 158
0x400e6dc9: TwoWire::endTransmission() at C:\Users\i7\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src\Wire.cpp line 316
0x400dc4df: LiquidCrystal_I2C::expanderWrite(unsigned char) at C:\Users\i7\Documents\Arduino\libraries\esp32-lcd\LiquidCrystal_I2C.cpp line 239
0x400dc52c: LiquidCrystal_I2C::write4bits(unsigned char) at C:\Users\i7\Documents\Arduino\libraries\esp32-lcd\LiquidCrystal_I2C.cpp line 232
0x400dc54d: LiquidCrystal_I2C::send(unsigned char, unsigned char) at C:\Users\i7\Documents\Arduino\libraries\esp32-lcd\LiquidCrystal_I2C.cpp line 227
0x400dc638: LiquidCrystal_I2C::write(unsigned char) at C:\Users\i7\Documents\Arduino\libraries\esp32-lcd\LiquidCrystal_I2C.cpp line 216
0x400d1ccb: printRSSItoLcd() at Z:\ErabliTEK\ErabliTEK\Martin\Projet Relacheur 3.0\0403/0403.ino line 647
0x400d1e17: UpdateLcd() at Z:\ErabliTEK\ErabliTEK\Martin\Projet Relacheur 3.0\0403/0403.ino line 257
0x400ee53c: timer_process_alarm at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer.c line 290
0x400ee5a3: timer_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer.c line 318
0x400896d9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32-WROOM-32
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     Don't know
Upload Speed:					921600
Computer OS:                                        Windows

stickbreaker, 2019-10-22:
@Speedy2k are you calling I2C functions from inside an interrupt? (timer?)  The `Wire()` object is not thread safe, nor is it re-entrant. Only one task at a time can use `Wire()`  The errors you are seeing lead me to believe your code is accessing `Wire()` then interrupting the access with another usage of `Wire()`.  

Post the shortest piece of code that will duplicate you errors.

Chuck.

Speedy2k, 2019-10-22:
You seems right stickbreaker!!!!

My bad not yet used with timer!!! It work amazingly but lead to unkown behaviour!!

Thanx a lot!

robcazzaro, 2019-10-23:
@stickbreaker a question related to this "The Wire() object is not thread safe, nor is it re-entrant.", if I may. Please let me know if I should open a separate issue instead, but I thought it would be good to capture related questions on this issue for future reference

I plan to use two separate I2C buses for a project, each with only one device. Each device will generate its own interrupt, on a separate GPIO pin. I will instantiate two separate Wire() instances, one for each bus, using different SCL and SDA pins. Can independent tasks use each instance at the same time, or should I plan to use semaphores to prevent the problems @Speedy2k ran into?

stickbreaker, 2019-10-23:
@robcazzaro Don't run `Wire()` from an interrupt context.  the i2c hardware driver is interrupt driven.  Have your interrupt set a flag/semaphore that is monitored in a task. Have that task to all `Wire()` operations. Each `Wire()` object is independent, so if you have two task, and each of these task only uses its own `Wire()` object and each `Wire()` object is using dedicated GPIO pins.  It will work fine.

I.E.
```c++
   SemaphoreHandle_t  int1Semaphore = NULL;

   SemaphoreHandle_t  int2Semaphore = NULL;

void IRAM_ATTR pinInt1( ){

     signed BaseType_t xHigherPriorityTaskWoken = pdFalse;

     SemaphoreGiveFromISR  ( int1Semaphore,&xHigherPriorityTaskWoken);
     portYIELD_FROM_ISR( xHigherPriorityTaskWoken );
}

void IRAM_ATTR pinInt2( ){

     signed BaseType_t xHigherPriorityTaskWoken = pdFalse;

     SemaphoreGiveFromISR  ( int2Semaphore,&xHigherPriorityTaskWoken);
     portYIELD_FROM_ISR( xHigherPriorityTaskWoken );
}

// A task that creates a semaphore.
void vATask( void * pvParameters )
{
   // Create the semaphore to guard a shared resource.
   vSemaphoreCreateBinary( int1Semaphore );

   if( int1Semaphore != NULL )
   {
      if(Wire.begin(SDA,SCL,100000)) {  //use board default SDA, Scl
        //attach interrupt function to specific pin and pass semaphore handle
        attachInterrupt(digitalPinToInterrupt(interruptPin), pinInt1, FALLING);
        //
        for (;;){
            // See if we can obtain the semaphore.  If the semaphore is not available
           // wait 1000 ticks to see if it becomes free.
           if( xSemaphoreTake( xSemaphore, ( TickType_t ) 1000 ) == pdTRUE )
            {
               // We were able to obtain the semaphore and can now access the
               // shared resource.
          
              // use Wire()
          } 
          else   {
           // is interrupt dead?
          }
   }
}

// duplicate task for int 2
```

chuck.

robcazzaro, 2019-10-23:
Thanks @stickbreaker! And, yes, sorry if I was not clear, I was not planning to run the I2C calls from the interrupt servicing, but always good for that to be pointed out.

I plan to use vTaskNotify(), which according to the freeRTOS documentation can be used as a faster version of a semaphore, and in my case has the added benefit of blocking the  waiting I2C reading task, speeding up the interrupt servicing cycle overall. As far as I can tell, the implementation below should be functionally equivalent to the code you kindly provided. But I'll compare semaphore and task notify side by side and see what works best

BTW: I could not find info on Wire() not being thread safe and not re-entrant anywhere else (which makes sense, in hindsight), so I think that it might be helpful to mark this issue as "for reference" for the future

```
static TaskHandle_t xI2C1Notify = NULL;

void IRAM_ATTR I2C1_Int()
{
  BaseType_t xHigherPriorityTaskWokenByPost = pdFALSE;
  vTaskNotifyGiveFromISR(xI2C1Notify , &xHigherPriorityTaskWokenByPost );
  if( xHigherPriorityTaskWokenByPost )
  {
    flag = true;
    portYIELD_FROM_ISR();
} 

void GetI2C1(void * parameter)
{
  // Initialize sensor I2C_1 sensor once

  for(;;) {

    ulNotificationValue = ulTaskNotifyTake( pdFALSE, xMaxBlockTime );

       // read I2C_1 sensor
       // send data

  }
}

void setup()
{
  xTaskCreatePinnedToCore ( GetI2C1, "GetI2C1", TaskStack15K, NULL, Priority5, &xI2C1Notify , 1 ); 
}

```

stickbreaker, 2019-10-23:
should work, add some code for error recovery.  all of the `Wire()` calls support `Wire.lastError()` which returns success of prior operation. `Wire.begin()` will recycle the bus using existing parameters and return a boolean (true) if the bus is recovered. 
from [esp32_hal_i2c.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.h)
```c++
// External Wire.h equivalent error Codes
typedef enum {
    I2C_ERROR_OK=0,
    I2C_ERROR_DEV,
    I2C_ERROR_ACK,
    I2C_ERROR_TIMEOUT,
    I2C_ERROR_BUS,
    I2C_ERROR_BUSY,
    I2C_ERROR_MEMORY,
    I2C_ERROR_CONTINUE,
    I2C_ERROR_NO_BEGIN
} i2c_err_t;
```

closed 2020-01-05 by stale[bot]