espressif/arduino-esp32 #3400: ESP.waitForConnectResult() return 0 when ESP is connected to iPhone's hotspot

OmarAlkassab, 2019-10-23:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
when I try to connect the ESP32 module to my iPhone's hotspot, the iPhone show that a new device is connected, but the WiFi.waitForConnectResult() (also WiFi.status()) return WL_IDLE_STATUS (return 0).

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>
void setup() {
Serial.begin(9600);
WiFi.begin("iPhone", "password");
int result = WiFi.waitForConnectResult();
if(result == WL_CONNECTED)
  Serial.print("connected");
else
  Serial.print("result = " + String(result));
}

void loop() {
}
```

### Debug Messages:
```
result = 0
```

Jeroen88, 2019-10-23:
If you enable debugging you may find more information on why it fails

OmarAlkassab, 2019-12-24:
Closing as the problem solved when updating to the latest firmware always (I don’t remember which version has solved the issue)

closed 2019-12-24 by OmarAlkassab