espressif/arduino-esp32 #3401: WiFiClient::write hangs sometimes

MrMabulous, 2019-10-23:
I'm using the ESP32 with the latest Arduino build.

sometimes WiFiClient::write() never returns, it usually happens while a lot of processing happens in other tasks.
The other tasks do not stall - only the task in which WiFiClient::write is called never gets to the subsequent instruction.

lbernstone, 2019-10-23:
You mention other tasks.  Make sure there are delays/yields in them to allow other tasks some cycles.  Also, check your priority levels.  You generally want WiFi (typically 19) to be a higher priority than less time sensitive activities.

MrMabulous, 2019-10-24:
The other tasks don't block, they continue and properly delay/yield. The task calling WifiClient::write() get's stuck sometimes though somewhere inside WifiClient::write() and I have to restart the device

closed 2020-01-06 by stale[bot]

alternadom, 2020-01-08:
Hi,
Did you find any solution to your problem ?
I'm having some issues too with WiFiClient::write() :(