espressif/arduino-esp32 #3404: ESP32 pin to GND results in Guru Mediation Error

GeorgeFlorian, 2019-10-24:
I might be doing something wrong, but isn't this a way of simulating an input on an ESP32 pin ?

```
#define TRIGGER_PIN 9
pinMode(TRIGGER_PIN, INPUT_PULLUP);
digitalWrite(TRIGGER_PIN, HIGH);
Serial.println(digitalRead(TRIGGER_PIN));
```

So while this PIN is not connected it returns 1. If I put it to Ground it should return 0, shouldn't it ?

GeorgeFlorian, 2019-10-24:
GPIO 9 worked for 2 values then it crashed.
```
1
0
0
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40140ba8: 00007015 04000136 72010100
Core 1 register dump:
PC      : 0x40140baf  PS      : 0x00060230  A0      : 0x8012f2ab  A1      : 0x3ffcded0  
A2      : 0x3ffd2894  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffb60b4  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffd2894  A9      : 0x3ffd2897  
A10     : 0x3ffd2894  A11     : 0x40085090  A12     : 0x3ffb643c  A13     : 0x00000000  
A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40140baf:0x3ffcded0 0x4012f2a8:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0 0x4012f311:0x3ffcded0
```

For whatever reason I cannot decode this backtrace. I've decoded all of the backtraces until this one using the tool "ESP Exception Decoder" that's installed on Arduino.

lbernstone, 2019-10-24:
That pinmap is incorrect.  gpio6-11 are used for the SPI flash, and are not available for other uses.  Use the [pinmap](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap) on the repository homepage.

GeorgeFlorian, 2019-10-24:
> That pinmap is incorrect. gpio6-11 are used for the SPI flash, and are not available for other uses. Use the [pinmap](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap) on the repository homepage.

And I've specified that I have also tried pins 12,21 and 23 and they also behave the same.

GeorgeFlorian, 2019-10-24:
@lbernstone 

I have used GPIO 21 and 23 with this code:
```
#define TRIGGER_PIN 21
// #define TRIGGER_PIN 23
pinMode(TRIGGER_PIN, INPUT_PULLUP);
digitalWrite(TRIGGER_PIN, HIGH);
Serial.println(digitalRead(TRIGGER_PIN));
```

GPIO 21 is always 0, doesn't matter if it's connected to GND or up in the air.
Also, booting the ESP with GPIO 21 to GND disables ETH, and I need ETH.

GPIO 23 behaves exactly the same.

Does it have something to do with `INPUT_PULLUP` ? Do I need to put a resistor ? 

If I instead use `INPUT_PULLDOWN` I can't simulate using GND, correct ?

closed 2019-10-24 by GeorgeFlorian

GeorgeFlorian, 2019-10-24:
I believe that the `Guru Meditation Errors` were a result of not a correct/usable GPIO.

And the "usable" pins always returning 0 issue is coming from the fact that I don't have a resistor to the input.