espressif/arduino-esp32 #3406: Receiving nothing on Serial2

GeorgeFlorian, 2019-10-24:
Board: ESP32-EVB

I am trying to read data from a Radar's RS232 Serial Interface through the UEXT connector on the ESP.

[image]

As per the schematics, I am using GPIO 36 as the rxPin and GPIO 4 as the txPin.

```
#define RXD2 36
#define TXD2 4

void setup() {
  delay(100);
  Serial.begin(115200);
  delay(1000);
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
}

void loop() {
  if(Serial2.available() >0) {
    Serial.println("Serial2 is available.");
    while (Serial2.available()) {
      char c = char(Serial2.read());
      Radar_String += c;
      Serial.print(char(Serial2.read()));
    }
    Serial.println("Radar_String = " + Radar_String);
}
```

I receive no data. 

Any thoughts ?

lbernstone, 2019-10-24:
Make sure you connect the TX to the RX of your radar module and vice-versa with the RX.  You can put an led/resistor on the RX pin to see if there is any data actually coming through.

GeorgeFlorian, 2019-10-24:
> Make sure you connect the TX to the RX of your radar module and vice-versa with the RX. You can put an led/resistor on the RX pin to see if there is any data actually coming through.

The TX of the ESP to the RX of the Radar ? But that's done by a UEXT connector... how can I modify that ?

lbernstone, 2019-10-24:
?? These are just wires.  I have no idea how you are connecting this, so I can't answer that.  The TX from the radar side needs to connect to the RX side of the ESP32 and vice-versa.  How that happens is up to you.

GeorgeFlorian, 2019-10-24:
> ?? These are just wires. I have no idea how you are connecting this, so I can't answer that. The TX from the radar side needs to connect to the RX side of the ESP32 and vice-versa. How that happens is up to you.

[image]

I am connecting the Radar to the ESP through an UART/RS232 converter. So I can't change the wires.

Will it work if I reverse them from code ?

On the UART Connector the GPIO 36 is the rxPin and GPIO 4 is the txPin. So in code I should write:
```
#define RXD2 4
#define TXD2 36
```

stickbreaker, 2019-10-24:
@GeorgeFlorian Here are some questions.
* how is your "Radar" wired?
You are using a ribbon cable, and show the ESP32 connections, but what are the "Radar" connections?
* you say "Radar RS232".
RS232 is a voltage and format standard. It is negative logic with a high level represented as a voltage from -3 to -12V, a Low a +3 to +12V
* ESP32 `Serial()` is not RS232  it is TTL serial with a high as positive 3.3v and a low a 0 V.
ESP32 `Serial()` Is postive logic at 3.3v.  Unless you have a level shifter you cannot directly connect to RS232 signals.

Verify the voltages, verify signal pins.  @lbernstone is just asking you to look at your physical connections. `Serial1()`  and `Serial2()` work, your code looks ok, so it must be a hardware failure. Either you wired it backwards (Tx->TX, Rx->Rx), or physically not connected, or incorrect voltage levels.

Chuck.

atanisoft, 2019-10-24:
you can try:
```
Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2, true);
```
The extra parameter inverts the logic the ESP32 uses for high/low signals. But the rest of the TTL/RS232 details that @stickbreaker references would still apply. You will likely need a level shifter unless the board with the RS232 connector has it onboard.

GeorgeFlorian, 2019-10-25:
This is the UEXT / RS232 module.
https://www.olimex.com/Products/Modules/Interface/MOD-RS232/open-source-hardware

I have connected an ESP32 DevKitV4 to the ribbon of the UEXT/RS232.

ESP's GND to Ribbon's GND (Pin 2)
ESP's RX (GPIO 16) to Ribbon's TX (Pin 3)
ESP's TX (GPIO 17) to Ribbon's RX (Pin 4)

The simplest code:
```
#include <Arduino.h>

#define RX2 16
#define TX2 17

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial2.begin(9600, SERIAL_8N1, RX2, TX2);

}

String Radar_String = "";

void loop() {
  if(Serial2.available() >0) {
    Serial.println("Serial2 is available.");
    while (Serial2.available()) {
      char c = char(Serial2.read());
      Radar_String += c;
      Serial.print(char(Serial2.read()));
    }
    Serial.println("Radar_String = " + Radar_String);
  }
}
```

and nothing...

I will next check the RS232 pins.

atanisoft, 2019-10-25:
@GeorgeFlorian you can't switch the pins in code. Pin 36 will not work for TX as it is input only.

GeorgeFlorian, 2019-10-25:
I don't understand the RS232 connectors...

I have a FEMALE RS232 connector on the board and a MALE RS232 from the Radar.

[image]

The wires that come from the Radar are soldered to the MALE RS232 connector as follows:

PIN1 (upper-left) = SG
PIN3 = RX
PIN4 = TX
PIN6 (bottom-left) = nothing
PIN7 = RTS
PIN8 = CTS

And I have found these pinmaps online:

[image]
[image]

From these schematics you can see that:
Male PIN2 (RXD) will connect to the Female PIN2 (TXD)
Male PIN3 (TXD) will connect to the Female PIN3 (RXD)

But, the schematics of the UEXT / RS232 board thingy has the Female pinmap different:

[image]

Here the FEMALE connector has PC_RXD on PIN2 and PC_TXD on PIN3.

**I have no idea how to connect to what...** 

From the first two schematics the Male RX is connected to the Female TX and the Male TX is connected to Female RX. Does this mean that I don't have to connect the TX from the radar to the RX of the ESP32 and the RX from the radar to the TX of the ESP, because the switch is happening here ?

From this Olimex board' schematics the Male RX is connected to the Female RX and the Male TX is connected to the Female TX. So, in this case, will I still have to connect the TX from the radar to the RX of the ESP32 and the RX from the radar to the TX of the ESP ?

GeorgeFlorian, 2019-10-25:
I have soldered the wires from the Radar to the MALE RS232 connector using a FEMALE pinmap.

I will re-solder the wires and come back with an update.

In the meantime I still have curiosities about the FEMALE RS232 that's on the Olimex UEXT / RS232 adapter.

As you can see in the post above the FEMALE connector has PC_RXD on PIN2 and PC_TXD on PIN3, this means that the MALE RX will connect to the FEMALE RX and the MALE TX will connect to the FEMALE RX.

Will I have to solder the Radars's TX to the MALE's RX and the Radar's RX to the MALE's TX ?

GeorgeFlorian, 2019-10-25:
Resoldering the wires as explained in the post above fixed my issue.

closed 2019-10-25 by GeorgeFlorian