espressif/arduino-esp32 #3408: ESP32 filesystem error (PLEASE KINDLY HELP ME!)

DmitryBorisenko33, 2019-10-25:
I have problem with spiffs file system. I am using standard example "FSBrowser"
But files in "data" folder (html, js, css) i am using not like in example. This is my files. This files perfectly running on esp8266 with the same sketch. But when i start using esp32 i have errors during web page refresh. 
Sometimes page stopped like this:
[image]
Sometimes going more:
[image]

I have now big problem with this. My project is stopped... Please kindly help me. I have read many information and forums but cant find how to resolve this problem. I found many people who have now same problem.

lbernstone, 2019-10-26:
You need to post logs from the esp32 side.  Turn on verbose debug and upload your sketch again.  Post the serial monitor logging during the file request.  Also please include all the hardware information requested in the [ISSUE_TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).
File locations in esp32 must always start with a /

DmitryBorisenko33, 2019-10-26:
I am author of ESP8266 + IOTmanager application modules firmware 
[This is my project](https://github.com/DmitryBorisenko33/esp8266_iot-manager_modules_firmware)
[This is our application](http://iotmanager.ru) now we working on ver 2 this application
Now i try to transfer my project from esp8266 to esp32
I am using arduino ide 1.8.9 and ESP32 arduino core 1.0.4

----------------------------------------------------------

  Hardware:
  Board: TTGO MINI32 V2.0.13 ESP32 rev1 [(this board)](https://ru.aliexpress.com/item/32845357819.html?spm=a2g0s.9042311.0.0.274233edrSkaLH) 
  Core Installation/update date: 02/10/2019
  IDE name: Arduino IDE 

  Flash Frequency:  80Mhz 
  PSRAM enabled: no
  Upload Speed: 921600

  Computer OS: Windows 10

  Description:
I have problem with spiffs file system. I am using standard example "FSBrowser"
But files in "data" folder (html, js, css) i am using not like in example. This is my files. This files perfectly running on esp8266 with the same sketch. But when i start using esp32 i have errors during web page refresh.

  Debug Messages (enable verbose debug) during page refresh:

```
[V][WebServer.cpp:286] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.5
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: Cache-Control
[V][Parsing.cpp:220] _parseRequest(): headerValue: max-age=0
[V][Parsing.cpp:219] _parseRequest(): headerName: Upgrade-Insecure-Requests
[V][Parsing.cpp:220] _parseRequest(): headerValue: 1
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,ru;q=0.8,mt;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
[V][WebServer.cpp:286] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /js/build.chart.js search: v07.04.2018
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.5
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: */*
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.5/
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,ru;q=0.8,mt;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: v07.04.2018
[V][Parsing.cpp:265] _parseArguments(): args count: 1
[V][Parsing.cpp:273] _parseArguments(): pos 0 =@-1 &@-1
[E][Parsing.cpp:275] _parseArguments(): arg missing value: 0
[V][Parsing.cpp:291] _parseArguments(): args count: 0
[... 11 lines cut ...]
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: Pragma
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: Cache-Control
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: image/webp,image/apng,image/*,*/*;q=0.8
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.5/
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,ru;q=0.8,mt;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /favicon.ico
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:617] _handleRequest(): request handler not found

[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
```
For your information my file system is  [here](https://github.com/DmitryBorisenko33/esp8266_iot-manager_modules_firmware/tree/master/data) and i have some folders in it: css js lang
[image]

DmitryBorisenko33, 2019-10-26:
[This is youtube video with my problem demonstration](https://youtu.be/hV1L-lzEonw)

Please help me.

DmitryBorisenko33, 2019-10-27:
Please look at my issue. Im sorry, i really need your help!

stickbreaker, 2019-10-27:
@DmitryBorisenko33 
* you haven't posted your code
How do you expect up to identify problems in your code if we can't see it?
* you have not registered any request handlers: `server.on()`
The error log you posted show:
>[E][WebServer.cpp:617] _handleRequest(): request handler not found
handleFileRead: /js/build.chart.js
[V][Parsing.cpp:230] _parseRequest(): Request: /
[V][Parsing.cpp:231] _parseRequest():  Arguments: 

Chuck.
[E][WebServer.cpp:617] _handleRequest(): request handler not found

lbernstone, 2019-10-27:
The WebServer is unpacking the index.htm.gz properly.  The contents of that html/js are not correct.  The server has no handler for the parameters that are going along with the GET requests for those files (css/build.js?v07.04.2018 vs css/build.js) so it fails.  Either set up a handler for them in the esp32 code, or request without the parameter.

DmitryBorisenko33, 2019-10-27:
> @DmitryBorisenko33
> 
> * you haven't posted your code
>   How do you expect up to identify problems in your code if we can't see it?
> * you have not registered any request handlers: `server.on()`
>   The error log you posted show:
> 
> > [E][WebServer.cpp:617] _handleRequest(): request handler not found
> > handleFileRead: /js/build.chart.js
> > [V][Parsing.cpp:230] _parseRequest(): Request: /
> > [V][Parsing.cpp:231] _parseRequest():  Arguments:
> > [E][WebServer.cpp:617] _handleRequest(): request handler not found
> 
> Chuck.

Hi thanks for reply. 
Im sorry, but my code is standard example for esp 32 web server: examples/web server/FSBrowser

code:

```

/*
  FSWebServer - Example WebServer with FS backend for esp8266/esp32
  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the WebServer library for Arduino environment.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.
  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.
  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

  upload the contents of the data folder with MkSPIFFS Tool ("ESP32 Sketch Data Upload" in Tools menu in Arduino IDE)
  or you can upload the contents of a folder if you CD in that folder and run the following command:
  for file in `ls -A1`; do curl -F "file=@$PWD/$file" esp32fs.local/edit; done

  access the sample web page at http://esp32fs.local
  edit the page by going to http://esp32fs.local/edit
*/
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

#define FILESYSTEM SPIFFS
// You only need to format the filesystem once
#define FORMAT_FILESYSTEM false
#define DBG_OUTPUT_PORT Serial

#if FILESYSTEM == FFat
#include <FFat.h>
#endif
#if FILESYSTEM == SPIFFS
#include <SPIFFS.h>
#endif

const char* ssid = "wifi-ssid";
const char* password = "wifi-password";
const char* host = "esp32fs";
WebServer server(80);
//holds the current upload
File fsUploadFile;

[... 232 lines cut ...]
  });

  //get heap status, analog input value and all GPIO statuses in one json call
  server.on("/all", HTTP_GET, []() {
    String json = "{";
    json += "\"heap\":" + String(ESP.getFreeHeap());
    json += ", \"analog\":" + String(analogRead(A0));
    json += ", \"gpio\":" + String((uint32_t)(0));
    json += "}";
    server.send(200, "text/json", json);
    json = String();
  });
  server.begin();
  DBG_OUTPUT_PORT.println("HTTP server started");

}

void loop(void) {
  server.handleClient();
}

```

When i use this the same code with the same file system on esp8266 everything working well. 
But with esp32 i have problems.

lbernstone, 2019-10-27:
esp32 != esp8266.  If you want to ensure compatibility, I would recommend using https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/examples/ESP_AsyncFSBrowser/ESP_AsyncFSBrowser.ino, which is supported from the same code base on both platforms.

DmitryBorisenko33, 2019-10-27:
I understood your recommendation about ESP_AsyncFSBrowser. It is really good and i will try it. But anyway i want to resolve this problem. I make new code and video for you.

This is my new code:

```
// Инициализация FFS
void FS_init(void) {
  if (FORMAT_FILESYSTEM) FILESYSTEM.format();
  FILESYSTEM.begin();
  {
      File root = FILESYSTEM.open("/");
      File file = root.openNextFile();
      while(file){
          String fileName = file.name();
          size_t fileSize = file.size();
          file = root.openNextFile();
      }
  }

  //SERVER INIT
  //list directory
  HTTP.on("/list", HTTP_GET, handleFileList);
  //load editor
  HTTP.on("/edit", HTTP_GET, []() {
    if (!handleFileRead("/edit.htm")) {
      HTTP.send(404, "text/plain", "FileNotFound");
    }
  });
  //create file
  HTTP.on("/edit", HTTP_PUT, handleFileCreate);
  //delete file
  HTTP.on("/edit", HTTP_DELETE, handleFileDelete);
  //first callback is called after the request has ended with all parsed arguments
  //second callback handles file uploads at that location
  HTTP.on("/edit", HTTP_POST, []() {
    HTTP.send(200, "text/plain", "");
  }, handleFileUpload);

  //called when the url is not defined here
  //use it to load content from FILESYSTEM
  HTTP.onNotFound([]() {
    if (!handleFileRead(HTTP.uri())) {
      HTTP.send(404, "text/plain", "FileNotFound");
    }
  });

  //get heap status, analog input value and all GPIO statuses in one json call
  HTTP.on("/all", HTTP_GET, []() {
    String json = "{";
    json += "\"heap\":" + String(ESP.getFreeHeap());
    json += ", \"analog\":" + String(analogRead(A0));
    json += ", \"gpio\":" + String((uint32_t)(0));
    json += "}";
    HTTP.send(200, "text/json", json);
    json = String();
  });

}
//format bytes
String formatBytes(size_t bytes) {
  if (bytes < 1024) {
    return String(bytes) + "B";
  } else if (bytes < (1024 * 1024)) {
    return String(bytes / 1024.0) + "KB";
  } else if (bytes < (1024 * 1024 * 1024)) {
    return String(bytes / 1024.0 / 1024.0) + "MB";
  } else {
    return String(bytes / 1024.0 / 1024.0 / 1024.0) + "GB";
  }
}
[... 268 lines cut ...]
[V][Parsing.cpp:219] _parseRequest(): headerName: Pragma
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: Cache-Control
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.120 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: image/webp,image/apng,image/*,*/*;q=0.8
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.5/
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,ru;q=0.8,mt;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /favicon.ico
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
```

This is two case. 

one - refresh page by arrow - file system working fully
two - refresh page by refresh button or first start - file system not working

DmitryBorisenko33, 2019-10-28:
Please dont forget look my video! its very important to understand.
[link to youtube](https://youtu.be/2rD6sQke5Jw)

DmitryBorisenko33, 2019-10-28:
Some people [here](https://github.com/espressif/arduino-esp32/issues/2560) had a same problem. But they were not resolve it.

me-no-dev, 2019-10-29:
in your code you do not have handlers for those files. it's quite simple :) pasting your setup() would be enough in that case. how you expect the server to find your files? because the FSBrowser does?
there is plenty info in the readme of the async web server for you to solve this :)

DmitryBorisenko33, 2019-10-29:
I have handlers for this file in my code: 

```
 void handleFileList() {
  if (!HTTP.hasArg("dir")) {
    HTTP.send(500, "text/plain", "BAD ARGS");
    return;
  }

  String path = HTTP.arg("dir");

  File root = FILESYSTEM.open(path);
  path = String();

  String output = "[";
  if(root.isDirectory()){
      File file = root.openNextFile();
      while(file){
          if (output != "[") {
            output += ',';
          }
          output += "{\"type\":\"";
          output += (file.isDirectory()) ? "dir" : "file";
          output += "\",\"name\":\"";
          output += String(file.name()).substring(1);
          output += "\"}";
          file = root.openNextFile();
      }
  }
  output += "]";
  HTTP.send(200, "text/json", output);
}
```

But anyway as per recommendation above, i already start AsyncWebServer. And my file system working properly and fully. Thank you so match for this recommendation.

AsyncWebServer more good and have login and password identification.

closed 2019-10-29 by DmitryBorisenko33